owfs/owfs #10: Invalid argument MAX31850

misisnik, 2018-07-11:
I have communication throw I2C with my Raspberry Pi  where is installed latest version of owfs.
Its using Alpine linux in docker (root system under the docker is Hypriot OS - debian). 

I want to read temperature of MAX31850 which is connected into the DS2482 i2c to 1w converter.
I have 5 1w devices (each is MAX31850). Mounted directory via owfs is ok (/mnt/1wire/) and i see all 6 devices (3B.3E0B5909FC4C/ , 3B.740B5909FC4C/, 3B.490B5909FC4C/ , 3B.EB418803DC4C/,  3B.5F0B5909FC4C/).

If i read temperature on chip (cat /mnt/1wire/3B.EB418803DC4C/temperature) it shows correct temperature. But If I read temperature on thermocouple (cat /mnt/1wire/3B.EB418803DC4C/thermocouple) it gives me error "read error: Invalid argument".

Is it bug? Or something with thermocouple? I using K type thermocouple and also MAX31850 model for K type.

Thanks Mike

ianka, 2018-07-11:
Please tell which version of owfs you are using: `owfs --version`. "Latest" is unclear.

Also, the FUSE way of accessing owfs has unfixable race conditions, don't build any production software on it. Use the owserver binary and the owread/owwrite/owdir tools instead.

For debugging, shutdown all the owfs based services, then run `owserver --debug` in one terminal window and do `owread /uncached/3B.EB418803DC4C/thermocouple` in another.

misisnik, 2018-07-12:
/opt/owfs/bin/owfs version: 3.2p2
libow version:  3.2p2
(latest commit ba66022f115909396a4b939f37e8c3ee8d293862)

When i run owserver and get thermocouple:

`DEBUG: ow_1820.c:(972) Error flag on thermocouple read of /uncached/3B.EB418803DC4C/thermocouple
  DEBUG: ow_read.c:(620) Read /uncached/3B.EB418803DC4C/thermocouple Extension 0 Gives result -22
  DEBUG: ow_read.c:(253) return=-22
  DEBUG: ow_read.c:(263) After read is performed (bytes or error -22)
OWQ OneWireQuery structure of /uncached/3B.EB418803DC4C/thermocouple
    OneWireQuery size=12 offset=0, extension=0
Byte buffer OneWireQuery buffer, length=12
--000: 00 00 00 00 00 00 00 00 00 00 00 00
   <............>
    Cleanup = 0006    OneWireQuery I=0 U=0 F=0 Y=0 D=Thu Jan  1 00:00:00 1970

--- OneWireQuery done
  DEBUG: ow_read.c:(226) /uncached/3B.EB418803DC4C/thermocouple returns -22
  DEBUG: ow_read.c:(103) /uncached/3B.EB418803DC4C/thermocouple return -22
  DEBUG: read.c:(80) ReadHandler: FS_read_postparse read on /uncached/3B.EB418803DC4C/thermocouple return = -22`

ianka, 2018-07-12:
> ow_1820.c:(972) Error flag on thermocouple read of …

If you look into `owfs/module/owlib/src/c/ow_1820.c`, line 970, you can see its testing bit 0 of the scratchpad byte 0 and bits 0-2 of the scratchpad byte 2. In page 13 of the MAX31850 datasheet, it's saying these are fault bits, if set. It doesn't tell you whether the chip thinks its open or shorted to GND or VDD but we could add that if you like.

**Please check the wiring of your thermocouple.**

misisnik, 2018-07-17:
I was change thermocouple. Reading is now possible.
owread /uncached/3B.EB418803DC4C/thermocouple gives me 4.4375. On reference thermometer is 70°C. Is any calculation needed for correct temperature value (i except 70°C but it returns 4.375) ?

ianka, 2018-07-17:
That could actually be a bug. Please file it again, as a bug and with a proper heading, so I can close this issue as "solved by user" and work on the new one.

closed 2018-07-17 by ianka