owfs/owfs #11: MAX31850 thermocouple read BUG

misisnik, 2018-07-17:
I am probably found a bug in thermocouple reading.

owread /uncached/3B.EB418803DC4C/thermocouple it returns 4.5625 but on reference thermocouple in multimeter is around 63°C.

Its connected like - K thermocouple to MAX31851 to DS2482 to I2C  (RPI)

Debug data from reading:
```
--- OneWireQuery done
  DEBUG: ow_read.c:(226) /uncached/3B.EB418803DC4C/thermocouple returns 12
  DEBUG: ow_read.c:(103) /uncached/3B.EB418803DC4C/thermocouple return 12
  DEBUG: read.c:(81) ReadHandler: FS_read_postparse read on /uncached/3B.EB418803DC4C/thermocouple return = 12
OWQ OneWireQuery structure of /uncached/3B.EB418803DC4C/thermocouple
    OneWireQuery size=12 offset=0, extension=0
Byte buffer OneWireQuery buffer, length=12
--000: 20 20 20 20 20 20 20 34 2E 33 37 35
   <       4.375>
    Cleanup = 0006    OneWireQuery I=12 U=12 F=4.375 Y=12 D=Thu Jan  1 00:00:12 1970

--- OneWireQuery done
  DEBUG: read.c:(89) ReadHandler: FS_read_postparse ok size=12
  DEBUG: read.c:(100) ReadHandler: To Client cm->payload=12 cm->size=12 cm->offset=0
Byte buffer Data returned to client, length=12
-- NULL buffer
  DEBUG: data.c:(146) Read message done value=0x76d9e740
  DEBUG: ow_parsename.c:(63) /uncached/3B.EB418803DC4C/thermocouple
  DEBUG: data.c:(193) DataHandler: FS_ParsedName_destroy done
  DEBUG: data.c:(207) DataHandler: cm.ret=12
  DEBUG: to_client.c:(76) payload=12 size=12, ret=12, sg=0x10A offset=0
  DEBUG: data.c:(226) Finished with client request
  DEBUG: handler.c:(135) OWSERVER handler done
  DEBUG: ow_net_server.c:(259) Normal completion.
```

ianka, 2018-07-17:
It uses the same temperature report function for the thermocouple and the cold junction temperature, but it seems someone has missed there's a factor of 16 between the two on the chip. Please multiply with 16 until it's fixed.

misisnik, 2018-07-18:
ok thanks

Switching-Power, 2020-07-28:
When multiplying with 16 as the current work around you only get whole numbers.

The MAX31850 and MAX31851 have 14bit resolution with 2 fraction bits for a 0.25°C accuracy in the thermocouple register, in the current code it is only 12 bits.
https://github.com/owfs/owfs/blob/7475794534a627fbb3ab6b74ade97fa74b102da3/module/owlib/src/c/ow_1820.c#L241-L242

The correct register sizes should be this if understand the code and [datasheet at page 13](https://datasheets.maximintegrated.com/en/ds/MAX31850-MAX31851.pdf) correctly.
```c
struct tempresolution ResolutionCLD= {12, 0x00, 100, 0xF0, 16.}; /* MAX31850  cold junction*/
struct tempresolution ResolutionTCP= {14, 0x00, 100, 0xFC, 4.};  /* MAX31850  thermocouple*/
```