owfs/owfs #12: DS9490 slewrate problem

anos21, 2018-07-31:
I have an [Elektor](https://www.elektormagazine.com/magazine/elektor-200604/18207) design (DS2490+DS2751) based on Maxim/Dallas [AN3874](https://pdfserv.maximintegrated.com/en/an/AN3874.pdf). It does not work with any OWFS release after 2.8p14, the DS2751 chip never gets detected. The standard linux kernel module ds2490 can detect the DS2751.

By reverting the commit [8eff440](https://github.com/owfs/owfs/commit/8ef4400e18474d83655eb77b60fd519c67987e60#diff-3313065efda23967c8a0c29d3fffc4ea) I can get the latest release 3.2p2 working. Reading the [datasheet](https://datasheets.maximintegrated.com/en/ds/DS2490.pdf) I would argue that the commit is not correct. Default slew rate at power up for flexible speed is 0.83V/us, which was used before the commit. However different 1W network might require other values depending on impedance, termination, topography etc.

As a reference, looking at linux kernel module source it seems like the slew rate setting is never changed.

ianka, 2018-07-31:
Thanks for your efforts to investigate this. I can only imagine the tedious process to pinpoint it to that one single commit.

[The DS2490 datasheet](https://pdfserv.maximintegrated.com/en/ds/DS2490.pdf) tells on page 4, the pulldown slew rate control circuit is inactive in the DS2490 at power-on and only setting another value than the default "0" activates it. To me, that means that Elektor circuit does not support a standard feature of a chip it uses rather than that commit is wrong. In addition that slewrate change has been working in many many cases over years so far.

**Does using --altUSB instead of -u or --usb help?**

I assume you know you could always use the DS2490 kernel driver by specifying --w1 on the OWFS command line.

anos21, 2018-08-01:
It was not so hard to find the one single commit, just run 2.8p14 and 2.8p15 in debug, compare logs and diff the important files.

The --altUSB does not give any change. From the code, that version seems to increase pulldown slew rate even more. Historically I've tried the --w1 but without real success, same now, but didn't spend too much time on it.

I can agree that the datasheet might not be super clear, but my interpretation is that it will be 0.83V/us (which is the power on default), if you not actively set it to something else. So reverting the commit or not touching it at all should give the same result.

From a HW near SW design perspective, and from the datasheet, I see the pulldown slew rate as an "application tuning parameter", and if you don't provide/expose that parameter to the user, you should have very good rationals to set it to anything else than the power on default. But that is just my opinion and how I would have designed the functionality. I can live with patching and building owfs locally, no problem. This was just a suggestion.

Then you never know, I might have a bad individual of either/or the DS2490 and DS2751, which makes it more sensitive. That's the problem with having just one test sample.

miccoli, 2018-08-02:
Just for reference: the changes in commit 8ef4400e18474d83655eb77b60fd519c67987e60 were proposed in <https://sourceforge.net/p/owfs/mailman/message/29078910/>.

ianka, 2018-08-02:
The reason why I'm picky about changing this back is there was a reason for adding this patch, which is as valid as your concern.

You are right it should be a tuneable setting. But there's also a reason why we have that --altUSB quirk, it's not so easy to implement correctly.

miccoli, 2018-08-03:
Obviously the one-size-fits-all approach does not hold here.

If it is note desirable to have the `master.usb.pulldownslewrate` tuneable at launch time (in order to avoid adding too many obscure owfs configuration options), may be at least a `configure` time option should be added: `configure --enable-usb --enable-usb-alt-slewrate` or something similar.