owfs/owfs #22: ds18b20, multiple temperatures

hakelm, 2018-12-01:
I just added a DS18B20 to my sensors and find multiple and different temperature files under OWFS, se below:

root@nilx:~# cat /ow/28.44F9FC050000/temperature; echo; cat /ow/28.44F9FC050000/temperature9; echo;cat /ow/28.44F9FC050000/temperature10; echo;cat /ow/28.44F9FC050000/temperature11; echo; cat /ow/28.44F9FC050000/temperature12;echo
1.4375
1.5
1.5
1.625
1.4375

what are the temperature9..temperature12 files and why are they different?
Thanks in advance
H

hakelm, 2018-12-02:
I find on
[http://owfs.org/index.php?page=ds18b20](url)

temperature9 temperature10 temperature11 temperature12
read-only, floating point Measured temperature at 9 to 12 bit resolution. There is a tradeoff of time versus accuracy in the temperature measurement.
Sorry about that.

miccoli, 2018-12-02:
@hakelm Happy you fond this answer by yourself. (Please note that you can also close the issue you opened.)

closed 2018-12-02 by miccoli

hakelm, 2018-12-02:
Thanks
can I conclude that 
/ow/28.44F9FC050000/temperature
is the same as
/ow/28.44F9FC050000/temperature12
something that would facilitate my software. 
H

miccoli, 2018-12-03:
@hakelm 

> Thanks
> can I conclude that
> /ow/28.44F9FC050000/temperature
> is the same as
> /ow/28.44F9FC050000/temperature12
> something that would facilitate my software.
> H

I've never used a DS18B20, but according to the manual the values of both endpoints should be the same.