owfs/owfs #25: Missing README file for autoreconf

vdanjean, 2018-12-11:
Hi,
The README file has been renamed into README.md.
This leads to an error when autoreconf (or just automake) is re-run:
    $ autoreconf -vif
    [...]
    autoreconf: running: automake --add-missing --copy --force-missing
    [...]
    Makefile.am: error: required file './README' not found
    [...]
    autoreconf: automake failed with exit status: 1

In the Debian build, I workaround the problem by copying README.md to README at the start of the build process.

miccoli, 2019-01-04:
In our bootstrap script
https://github.com/owfs/owfs/blob/e7b1c0fc9a3f00877847212c22a38a6dc9d474b7/bootstrap#L12
we run `automake` with the `--foreign` option and this makes `automake` a little less fussy about GNU conventions.

Since we do not use `autoreconf` I think that this issue should be fixed downstream and not here.

bahm-lgs, 2020-02-14:
Same issue happens when building for yocto. In the owfs recipe the openembedded project has added  a patch for this:
https://github.com/openembedded/meta-openembedded/blob/master/meta-filesystems/recipes-filesystems/owfs/owfs/0001-Add-build-rule-for-README.patch