owfs/owfs #34: Cannot open server at localhost:4304

Nornode, 2019-02-09:
Hi, all!

I'm running owfs 
```
/usr/lib/owfs/owfs version:
        3.1p5
libow version:
        3.1p5
```

Since yesterday I'm I got troubles accessing my Dallas 1-wire sensors read from [this](R-pi i2c 1wire expansion module) "hat" on a Raspberry 3.

This command: sudo /usr/bin/owfs --i2c=ALL:ALL --allow_other /mnt/1wire/ --debug
```
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1B
  DEBUG: ow_ds2482.c:(516) ok
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1B cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1C
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1C cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1D
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1D cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1E
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1E cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1F
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1F cannot be reset. Not a DS2482.
  DEBUG: ow_com_close.c:(43) Unimplemented!!!
CONNECT: owlib.c:(145) Cannot detect an i2c DS2482-x00 on ALL:ALL
  DEBUG: ow_net_client.c:(28) Called with 127.0.0.1:4304 default=4304
  DEBUG: ow_regex.c:(154) Not found
  DEBUG: ow_regex.c:(201) 0: 0->14 found <><127.0.0.1:4304><>
  DEBUG: ow_regex.c:(201) 1: 0->9 found <><127.0.0.1><:4304>
  DEBUG: ow_regex.c:(201) 2: 10->14 found <127.0.0.1:><4304><>
  DEBUG: ow_parse_address.c:(53) IP <127.0.0.1>
  DEBUG: ow_parse_address.c:(120) First <127.0.0.1>
  DEBUG: ow_parse_address.c:(125) Second <4304>
  DEBUG: ow_parse_address.c:(57) Num <4304> 4304
  DEBUG: ow_net_client.c:(85) Called with [127.0.0.1:4304] IP address=[127.0.0.1] port=[4304]
CONNECT: ow_net_client.c:(147) [Connection refused] Socket problem
CONNECT: owlib.c:(108) Cannot open server at 127.0.0.1:4304 -- first attempt.
  DEBUG: ow_net_client.c:(28) Called with 127.0.0.1:4304 default=4304
  DEBUG: ow_regex.c:(154) Not found
  DEBUG: ow_regex.c:(201) 0: 0->14 found <><127.0.0.1:4304><>
  DEBUG: ow_regex.c:(201) 1: 0->9 found <><127.0.0.1><:4304>
  DEBUG: ow_regex.c:(201) 2: 10->14 found <127.0.0.1:><4304><>
  DEBUG: ow_parse_address.c:(53) IP <127.0.0.1>
  DEBUG: ow_parse_address.c:(120) First <127.0.0.1>
  DEBUG: ow_parse_address.c:(125) Second <4304>
  DEBUG: ow_parse_address.c:(57) Num <4304> 4304
  DEBUG: ow_net_client.c:(85) Called with [127.0.0.1:4304] IP address=[127.0.0.1] port=[4304]
CONNECT: ow_net_client.c:(147) [Connection refused] Socket problem
CONNECT: owlib.c:(113) Cannot open server at 127.0.0.1:4304 -- second (and final) attempt.
DEFAULT: owlib.c:(52) No valid 1-wire buses found
  DEBUG: ow_exit.c:(17) Exit code = 1
```

I have rebooted and also confirmed that I have a i2c instance and restarted all ow* instanced under /usr/bin that I could guess..

What has gone wrong? - This thaw weather can't make it refuse to start (short circut)?

miccoli, 2019-02-10:
> I have rebooted and also confirmed that I have a i2c instance and restarted all ow* instanced under /usr/bin that I could guess.

Actually for debugging you should *stop* all `ow*` daemons and keep only `owserver` running. Moreover, if you are on raspbian, don't start daemon directly (`sudo /usr/bin/ow*`) but use the appropriate `sudo systemctl start ow*.service`.

My personal advice: keep it simple, uninstall all ow* packages, and keep only `owserver` and `ow-shell` and verify if the `owdir` shell command lists all your devices.

Nornode, 2019-02-20:
Thanks miccoli, 
I'm sure that's the way to go but don't seem to solve my problem? 
I did an unistall and a reinstall of OWFS (All packages) but if I understand you right, I should only install 
`owserver` and `ow-shell` ?

Here's the outpus tarting owserver --debug

> sudo /usr/bin/owserver --i2c=ALL:ALL --allow_other /mnt/1wire/ --debug
> DEBUG MODE
> libow version:
> 	3.1p5
> DEFAULT: ow_arg.c:(115) Not a "character" device /mnt/1wire/ (st_mode=41ed)
>   DEBUG: ow_daemon.c:(170) main thread id = 1996144768
>   DEBUG: ow_inotify.c:(80) No configuration files to monitor
> CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
>    CALL: ow_parsename.c:(104) path=[]
>   DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
> ]*$> compiled to 0x76f72304g Ex expression <^ *([^ ]+)[
> ]*$> compiled to 0x76f72324g Ex expression <^ *([^ ]+) *: *([^ ]+)[
> ]*$> compiled to 0x76f72344g Ex expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[
>   DEBUG: ow_regex.c:(154) Not found
>   DEBUG: ow_regex.c:(201) 0: 0->7 found <><ALL:ALL><>
>   DEBUG: ow_regex.c:(201) 1: 0->3 found <><ALL><:ALL>
>   DEBUG: ow_regex.c:(201) 2: 4->7 found <ALL:><ALL><>
>   DEBUG: ow_regex.c:(74) Reg Ex expression <^$> compiled to 0x76f72244
>   DEBUG: ow_regex.c:(74) Reg Ex expression <^all$> compiled to 0x76f72264
>   DEBUG: ow_regex.c:(74) Reg Ex expression <^scan$> compiled to 0x76f72284
>   DEBUG: ow_regex.c:(74) Reg Ex expression <^\*$> compiled to 0x76f722a4
>   DEBUG: ow_regex.c:(74) Reg Ex expression <^[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}$> compiled to 0x76f722c4
>   DEBUG: ow_regex.c:(74) Reg Ex expression <^-?[[:digit:]]+$> compiled to 0x76f722e4
>   DEBUG: ow_parse_address.c:(44) All <ALL>
>   DEBUG: ow_parse_address.c:(120) First <ALL>
>   DEBUG: ow_parse_address.c:(125) Second <ALL>
>   DEBUG: ow_parse_address.c:(44) All <ALL>
> CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/.
> CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/..
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 18
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 18 cannot be reset. Not a DS2482.
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 19
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 19 cannot be reset. Not a DS2482.
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1A
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1A cannot be reset. Not a DS2482.
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1B
>   DEBUG: ow_ds2482.c:(516) ok
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1B cannot be reset. Not a DS2482.
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1C
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1C cannot be reset. Not a DS2482.
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1D
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1D cannot be reset. Not a DS2482.
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1E
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1E cannot be reset. Not a DS2482.
> CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1F
> CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1F cannot be reset. Not a DS2482.
>   DEBUG: ow_com_close.c:(43) Unimplemented!!!
> CONNECT: owlib.c:(145) Cannot detect an i2c DS2482-x00 on ALL:ALL
> DEFAULT: owlib.c:(52) No valid 1-wire buses found
>   DEBUG: ow_exit.c:(17) Exit code = 1
>    CALL: ow_lib_close.c:(21) Starting Library cleanup
>    CALL: ow_lib_stop.c:(22) Clear Cache
>   DEBUG: ow_cache.c:(295) Flipping cache tree (purging timed-out data)
>   DEBUG: ow_cache.c:(313) flip cache. tdestroy() will be called.
>   DEBUG: ow_cache.c:(295) Flipping cache tree (purging timed-out data)
>   DEBUG: ow_cache.c:(313) flip cache. tdestroy() will be called.
>    CALL: ow_lib_stop.c:(24) Closing input devices
>    CALL: ow_lib_stop.c:(26) Closing output devices
>    CALL: ow_lib_stop.c:(28) Clearing compiled expressions
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f72244
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f72264
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f72284
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f722a4
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f722c4
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f722e4
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f72304
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f72324
>   DEBUG: ow_regex.c:(117) Regex Free 0x76f72344
>   DEBUG: ow_regex.c:(129) Regex destroy done
>    CALL: ow_lib_close.c:(42) Finished Library cleanup
>   DEBUG: ow_lib_close.c:(50) Libraries closed
> 

When installing I used this method:
https://wiki.m.nu/index.php/OWFS_with_i2c_support_on_Raspberry_Pi_(English_version)
But installed with `sudo apt install owfs``
Thanks for your help!

miccoli, 2019-02-20:
> I did an unistall and a reinstall of OWFS (All packages) but if I understand you right, I should only install
> `owserver` and `ow-shell` ?

Yes, `apt install owserver owdir`, just to be sure that there are no other daemons accessing the 1-wire bus master; however the important thing is to double check that there are no other ow* daemons running (eg with `pstree` and killing all running ow* daemons). Then the correct command to start the owserver is
```
sudo /usr/bin/owserver --i2c=ALL:ALL --debug
```
Once the owserver is up and running you should run 
```
owdir
```
from a different terminal.

Nornode, 2019-02-21:
That worked!
No ow* daemon was running and I installed the recommended packages.  I can
see my sensors running owdir!

So why are they not mounted on the place I have directed?
What's the dependencies as Domoticz I'm using to read depends on a folder
listing (mnt/1wire/).

Thanks for the troubleshooting help!

miccoli, 2019-02-21:
`/usr/bin/owfs` (the fuse filesystem access to OWFS) is now deprecated: in this regard instructions at <https://wiki.m.nu/index.php/OWFS_with_i2c_support_on_Raspberry_Pi_(English_version)> are outdated, if not wrong.

I know, docs on `ow*` are sparse and not well maintained; currently the only official docs are the man pages (`man owserver`, `man owfs.conf`) and the old home <http://owfs.org>.

If I can condense advice in a few words:

1. if possible avoid the fuse thing `/mnt/1wire/`: specific instructions depend on your application, but at least for shell programs just change `cat /mnt/1wire/26.xxxxxxxxxxxx/temperature` into `owread /26.xxxxxxxxxxxx/temperature` etc.

1. in Debian derived distros (ubuntu, raspbian, etc.) ensure the daemon is started at boot by enabling its systemd service: `sudo systemctl enable owserver.service` and do not edit files in `/etc/init.d`

1. finally take the time to properly configure the `/etc/owfs.conf` file. It's a little bit more complicated than necessary, but not that difficult.

A great place for asking for help is the mailing list, <https://sourceforge.net/p/owfs/mailman/owfs-developers/>: low traffic and usually very friendly.

Nornode, 2019-02-26:
Hey, coming back a couple of days later trying your advice (and having to write a bash script to out the values in to files) owserver stopped and don't want to start again.
Claiming that ow_com_close is not implemented and it can't detect an i2c DS2482 on ALL:ALL.
(Log output below)

However, I think it's strange as it did work excellently before and my i2cdetect still gives me a proper value on the bus.

This ain't cool to just deprecate a functioning setup!

`sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- 1b -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --`

`sudo /usr/bin/owserver --i2c=ALL:ALL --debug
DEBUG MODE
libow version:
        3.1p5
  DEBUG: ow_daemon.c:(170) main thread id = 1996193920
  DEBUG: ow_inotify.c:(80) No configuration files to monitor
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
   CALL: ow_parsename.c:(104) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
]*$> compiled to 0x76f7e304g Ex expression <^ *([^ ]+)[
]*$> compiled to 0x76f7e324g Ex expression <^ *([^ ]+) *: *([^ ]+)[
]*$> compiled to 0x76f7e344g Ex expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[
  DEBUG: ow_regex.c:(154) Not found
  DEBUG: ow_regex.c:(201) 0: 0->7 found <><ALL:ALL><>
  DEBUG: ow_regex.c:(201) 1: 0->3 found <><ALL><:ALL>
  DEBUG: ow_regex.c:(201) 2: 4->7 found <ALL:><ALL><>
  DEBUG: ow_regex.c:(74) Reg Ex expression <^$> compiled to 0x76f7e244
  DEBUG: ow_regex.c:(74) Reg Ex expression <^all$> compiled to 0x76f7e264
  DEBUG: ow_regex.c:(74) Reg Ex expression <^scan$> compiled to 0x76f7e284
  DEBUG: ow_regex.c:(74) Reg Ex expression <^\*$> compiled to 0x76f7e2a4
  DEBUG: ow_regex.c:(74) Reg Ex expression <^[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}$> compiled to 0x76f7e2c4
  DEBUG: ow_regex.c:(74) Reg Ex expression <^-?[[:digit:]]+$> compiled to 0x76f7e2e4
  DEBUG: ow_parse_address.c:(44) All <ALL>
  DEBUG: ow_parse_address.c:(120) First <ALL>
  DEBUG: ow_parse_address.c:(125) Second <ALL>
  DEBUG: ow_parse_address.c:(44) All <ALL>
CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/.
CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/..
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 18
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 18 cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 19
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 19 cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1A
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1A cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1B
  DEBUG: ow_ds2482.c:(516) ok
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1B cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1C
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1C cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1D
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1D cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1E
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1E cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1F
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1F cannot be reset. Not a DS2482.
  DEBUG: ow_com_close.c:(43) Unimplemented!!!
CONNECT: owlib.c:(145) Cannot detect an i2c DS2482-x00 on ALL:ALL
DEFAULT: owlib.c:(52) No valid 1-wire buses found
  DEBUG: ow_exit.c:(17) Exit code = 1
   CALL: ow_lib_close.c:(21) Starting Library cleanup
   CALL: ow_lib_stop.c:(22) Clear Cache
  DEBUG: ow_cache.c:(295) Flipping cache tree (purging timed-out data)
  DEBUG: ow_cache.c:(313) flip cache. tdestroy() will be called.
  DEBUG: ow_cache.c:(295) Flipping cache tree (purging timed-out data)
  DEBUG: ow_cache.c:(313) flip cache. tdestroy() will be called.
   CALL: ow_lib_stop.c:(24) Closing input devices
   CALL: ow_lib_stop.c:(26) Closing output devices
   CALL: ow_lib_stop.c:(28) Clearing compiled expressions
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e244
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e264
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e284
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e2a4
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e2c4
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e2e4
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e304
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e324
  DEBUG: ow_regex.c:(117) Regex Free 0x76f7e344
  DEBUG: ow_regex.c:(129) Regex destroy done
   CALL: ow_lib_close.c:(42) Finished Library cleanup
  DEBUG: ow_lib_close.c:(50) Libraries closed`

miccoli, 2019-08-12:
I'm closing this isse because It has become too old to be still useful.

Unfortunately we are not able to give advice on generic configuration/installation issues on this site.

Sorry for closing and feel free to ask again on the mailing list.

closed 2019-08-12 by miccoli