owfs/owfs #35: owfs.org is down

Misiu, 2019-04-08:
Hi there,
it looks like http://owfs.org/ is down. All weekend there were errors from PHP about cache folder permissions, and now (2019-04-08) site is down.

miccoli, 2019-04-08:
The dns-server is down; the domain itself is about to expire.

From whois:

```
Domain ID: D122312630-LROR
Registrar WHOIS Server: whois.enom.com
Registrar URL: http://www.enom.com
Updated Date: 2018-04-28T07:04:50Z
Creation Date: 2006-05-13T17:52:22Z
Registry Expiry Date: 2019-05-13T17:52:22Z
Registrar Registration Expiration Date:
Registrar: eNom, Inc.
Registrar IANA ID: 48
Registrar Abuse Contact Email: abuse@enom.com
Registrar Abuse Contact Phone: +1.4252982646
Reseller:
Domain Status: clientTransferProhibited 

https://icann.org/epp#clientTransferProhibited

Registrant Organization:
Registrant State/Province: SK
Registrant Country: CA
Name Server: NS1-R6-VANCOUVER.WEBSERVERSYSTEMS.COM
Name Server: NS2-R6-VANCOUVER.WEBSERVERSYSTEMS.COM
DNSSEC: unsigned
URL of the ICANN Whois Inaccuracy Complaint Form https://www.icann.org/wicf/)
```

Misiu, 2019-04-08:
@miccoli DNS seems to work, but now I get this when I open http://owfs.org:
[image]

miccoli, 2019-04-08:
At google public dns, owfs.org is not reachable:
```
$ dig @8.8.8.8 owfs.org SOA

; <<>> DiG 9.10.6 <<>> @8.8.8.8 owfs.org SOA
; (1 server found)
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 975
;; flags: qr rd ra; QUERY: 1, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;owfs.org.			IN	SOA

;; Query time: 446 msec
;; SERVER: 8.8.8.8#53(8.8.8.8)
;; WHEN: Mon Apr 08 22:41:01 CEST 2019
;; MSG SIZE  rcvd: 37
```

AFAIK no one here at the owfs GitHub repo has control over owfs.org: it seems that the server and domain have been discontinued. I will report this on the OWFS mailing list: @Misiu thx for reporting.

miccoli, 2019-04-08:
Current (2019-04-08) domain status.

<details>
<summary>WHOIS query</summary>

```
$ whois owfs.org
% IANA WHOIS server
% for more information on IANA, visit http://www.iana.org
% This query returned 1 object

refer:        whois.pir.org

domain:       ORG

organisation: Public Interest Registry (PIR)
address:      1775 Wiehle Avenue
address:      Suite 102A
address:      Reston Virginia 20190
address:      United States

contact:      administrative
name:         Director of Operations, Compliance and Customer Support
organisation: Public Interest Registry (PIR)
address:      1775 Wiehle Avenue
address:      Reston Virginia 20190
address:      United States
phone:        +1 919 369 4161
fax-no:       +1 703 889 5779
e-mail:       ops@pir.org

contact:      technical
name:         Senior Director, DNS Infrastructure Group
organisation: Afilias
address:      Building 3, Suite 105
address:      300 Welsh Road
address:      Horsham, Pennsylvania 19044
address:      United States
phone:        +1 215.706.5700
fax-no:       +1 215.706.5701
e-mail:       tld-tech-poc@afilias.info

nserver:      A0.ORG.AFILIAS-NST.INFO 199.19.56.1 2001:500:e:0:0:0:0:1
nserver:      A2.ORG.AFILIAS-NST.INFO 199.249.112.1 2001:500:40:0:0:0:0:1
nserver:      B0.ORG.AFILIAS-NST.ORG 199.19.54.1 2001:500:c:0:0:0:0:1
nserver:      B2.ORG.AFILIAS-NST.ORG 199.249.120.1 2001:500:48:0:0:0:0:1
nserver:      C0.ORG.AFILIAS-NST.INFO 199.19.53.1 2001:500:b:0:0:0:0:1
nserver:      D0.ORG.AFILIAS-NST.ORG 199.19.57.1 2001:500:f:0:0:0:0:1
ds-rdata:     9795 7 2 3922b31b6f3a4ea92b19eb7b52120f031fd8e05ff0b03bafcf9f891bfe7ff8e5
ds-rdata:     9795 7 1 364dfab3daf254cab477b5675b10766ddaa24982

whois:        whois.pir.org

status:       ACTIVE
remarks:      Registration information: http://www.pir.org

created:      1985-01-01
changed:      2016-05-10
source:       IANA

Domain Name: OWFS.ORG
Registry Domain ID: D122312630-LROR
Registrar WHOIS Server: whois.enom.com
Registrar URL: http://www.enom.com
Updated Date: 2018-04-28T07:04:50Z
Creation Date: 2006-05-13T17:52:22Z
Registry Expiry Date: 2019-05-13T17:52:22Z
Registrar Registration Expiration Date:
Registrar: eNom, Inc.
Registrar IANA ID: 48
[... 97 lines cut ...]
Admin Phone: REDACTED FOR PRIVACY
Admin Phone Ext: 
Admin Fax: REDACTED FOR PRIVACY
Admin Email: REDACTED FOR PRIVACY
Tech Name: REDACTED FOR PRIVACY
Tech Organization: REDACTED FOR PRIVACY
Tech Street: REDACTED FOR PRIVACY
Tech Street: 
Tech City: REDACTED FOR PRIVACY
Tech State/Province: REDACTED FOR PRIVACY
Tech Postal Code: REDACTED FOR PRIVACY
Tech Country: REDACTED FOR PRIVACY
Tech Phone: REDACTED FOR PRIVACY
Tech Phone Ext: 
Tech Fax: REDACTED FOR PRIVACY
Tech Email: REDACTED FOR PRIVACY
Name Server: NS1-R6-VANCOUVER.WEBSERVERSYSTEMS.COM
Name Server: NS2-R6-VANCOUVER.WEBSERVERSYSTEMS.COM
DNSSEC: UNSIGNED
Registrar Abuse Contact Email: ABUSE@ENOM.COM
Registrar Abuse Contact Phone: +1.4259744689
URL of the ICANN WHOIS Data Problem Reporting System: HTTP://WDPRS.INTERNIC.NET/
>>> Last update of WHOIS database: 2018-04-17T03:44:27.00Z <<<
```
</details>

Misiu, 2019-04-08:
@miccoli two or three days ago the website was working just fine. It looks like there is something wrong with PHP.
I'm struggling with setting up OWFS on openhabian so docs would help a lot.
Github would be an awesome place to host documentation. Updating it would be just a simple PR.

stromnet, 2019-04-09:
> At google public dns, owfs.org is not reachable:
> 
> ```
> $ dig @8.8.8.8 owfs.org SOA
> 
...> ;; ->>HEADER<<- opcode: QUERY, status: SERVFAIL, id: 975

Not sure why you got a SERVFAIL there but the above query works fine from here today:
```
$ dig owfs.org @8.8.8.8 SOA

; <<>> DiG 9.10.6 <<>> owfs.org @8.8.8.8 SOA
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 13451
;; flags: qr rd ra; QUERY: 1, ANSWER: 1, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 512
;; QUESTION SECTION:
;owfs.org.			IN	SOA

;; ANSWER SECTION:
owfs.org.		21599	IN	SOA	ns1.outofthisworldhosting.com. mike.kalist.gmail.com. 2019040103 86400 7200 3600000 86400

;; Query time: 243 msec
;; SERVER: 8.8.8.8#53(8.8.8.8)
;; WHEN: Tue Apr 09 06:51:25 CEST 2019
;; MSG SIZE  rcvd: 120
```

The server is however only serving an error page, as mentioned above (from the looks of it, some templating framework which fails to paint the page).

Unfortunately I don't know know who has/had access to either DNS, domain or server. There is a old mail thread regarding move of the site: https://sourceforge.net/p/owfs/mailman/message/35353958/

Unfortunately it never amounted to much of a move, besides moving code to Github.

miccoli, 2019-04-09:
DNS changes take long to propagate, in fact DNS is up again also from my location... however there are obvious hosting problems for <http://owfs.org>

@Misiu for now you can access the archived version on the wayback machine, e.g.:
<http://web.archive.org/web/20181229052637/http://owfs.org/>

IMHO a logical way forward would be 

1. we elect a BDFL
2. the BDFL takes control of the owfs.org domain
3. we use [github hosting with a custom domain](https://help.github.com/en/articles/using-a-custom-domain-with-github-pages) for making <https://owfs.github.io/> accessible at <https://owfs.org>
4. we all contribute to the docs at the [wiki](https://github.com/owfs/owfs-doc/wiki) and on the [site](https://owfs.github.io)

I would not try to rescue the old php site: I think it would be too hard to maintain.

Misiu, 2019-04-09:
@miccoli I agree, GitHub is a good place for docs, especially that they can be easily updated with PR.
I'm new to OWFS and I'm trying to setup it on Raspberry PI3 on openHAB using GPIO.
If I succeed I can write readme with all the steps.

The most important thing now is the domain, it will expire on 2019-05-13T17:52:22Z.
Someone knows who the current owner is and how to contact him?

stromnet, 2019-04-11:
FYI: there is ongoining discussion on mailing list with current domain holder & hoster, and also input from owfs original author Paul Alfille.

Please have your say there!

cuora, 2019-05-10:
Mike is the holder he asked for someone to take over the website and the domain.  Meanwhile others have stolen  the domain.  of i'm wrong.  What i've mentioned  in the mailing list.  I reserved the domain onewirefilesystem.org  and rebuild it from the wayback archived site.   I hope we can thave he original domain back.

miccoli, 2019-08-12:
<https://www.owfs.org> has been rescued. Thanks to all involved.

closed 2019-08-12 by miccoli