owfs/owfs #36: run owfs on raspberrz pi with GPIO

Misiu, 2019-04-10:
I'm trying to setup and run OWFS on openhabian running on Raspberry Pi 3B.
I have two DS18B20 temperature sensors connected to GPIO4.
If I understand [the docs ](http://owfs.org/index.php?page=w1-project)correctly I can use OWFS with W1 linux kernel module, without any extra hardware.
The problem is I get erros every time.
I have installed:

> ow-shell 3.1p5-1
> owserver 3.1p5-1
> usbutils 1:007-4

Below is my /etc/owfs.conf:

```
# Sample configuration file for the OWFS suite for Debian GNU/Linux.
#
#
# This is the main OWFS configuration file. You should read the
# owfs.conf(5) manual page in order to understand the options listed
# here.

######################## SOURCES ########################
#
# With this setup, any client (but owserver) uses owserver on the
# local machine...
! server: server = localhost:4304
#
# ...and owserver uses the real hardware, by default fake devices
# This part must be changed on real installation
server: FAKE = DS18S20,DS2405
#
# USB device: DS9490
#server: usb = all
#
# Serial port: DS9097
#server: device = /dev/ttyS1
#
# owserver tcp address
#server: server = 192.168.10.1:3131
#
# random simulated device
#server: FAKE = DS18S20,DS2405
#
######################### OWFS ##########################
#
#mountpoint = /mnt/1wire
#allow_other
#
####################### OWHTTPD #########################

http: port = 2121

####################### OWFTPD ##########################

ftp: port = 2120

####################### OWSERVER ########################

server: port = localhost:4304
```

And this is the result of `sudo service owserver status`

> ● owserver.service - LSB: 1-wire TCP server
>    Loaded: loaded (/etc/init.d/owserver; generated; vendor preset: enabled)
>    Active: active (running) since Sun 2019-04-07 00:17:09 CEST; 3 days ago
>      Docs: man:systemd-sysv-generator(8)
>   Process: 515 ExecStart=/etc/init.d/owserver start (code=exited, status=0/SUCCESS)
>    CGroup: /system.slice/owserver.service
>            └─645 /usr/bin/owserver -c /etc/owfs.conf --pid-file /var/run/owfs/owserver.pid
> 
> Apr 07 00:17:08 openHABianPi systemd[1]: Starting LSB: 1-wire TCP server...
> Apr 07 00:17:08 openHABianPi OWFS[645]: DEFAULT: ow_daemon.c:(144) Entered background mode, quitting.
> Apr 07 00:17:09 openHABianPi owserver[515]: Starting 1-Wire TCP Server: owserver.
> Apr 07 00:17:09 openHABianPi systemd[1]: Started LSB: 1-wire TCP server.

If I understand correctly I should have fake server because I have no hash on line `server: FAKE = DS18S20,DS2405`

I've checked that I have connected my devices correctly because I see two folders when I go to `/sys/bus/w1/devices`

I'll be gratefull for help.

renegade2k, 2019-11-21:
Same here.
Trying to get DS18b20 to work on GPIO pin 4.
Can read the temperature data manually, but owfs does not see the sensors.
There is no hint on how to config it :(

ianka, 2019-11-21:
You start it with the --w1 option or a minimal owfs.conf having

    server: w1
    ! server: server = localhost:4304

Restart owserver.

    $ owdir /uncached

should list the sensor.

Frank-Friemel, 2020-02-14:
Just for the records ... I had a problem with `localhost` for the host definition, because owserver listened on ipv6’s localhost but *not* on ipv4’s . Many clients (e.g. owjs) try to connect with ipv4 and are failing. I‘ve replaced `localhost` in owfs.conf with an `*` (Asterisk) which did the trick, because owserver listens then on all host adapters. Including the hardware adapters (so beware)

miccoli, 2021-08-19:
> Just for the records ... I had a problem with `localhost` for the host definition, because owserver listened on ipv6’s localhost but _not_ on ipv4’s . Many clients (e.g. owjs) try to connect with [...]

Yes, IPV6 should be handled better, see #90

miccoli, 2021-08-19:
Closing this issue because it's an old request for help. Please feel free to ping if you think it should be reopenend.

closed 2021-08-19 by miccoli

Misiu, 2022-01-06:
Since 1-wire is deprecated in Home Assistant many users (including me) will look, for an alternative.
I'll try this approach again (running OWFS from GPIO).
So I'm sure I'll return back with questions