owfs/owfs #38: Can't configure

rchovan, 2019-06-17:
Hi, 
I have cloned repository and run this commands: 

```
#!/bin/bash
./bootstrap
./configure --disable-owftpd --enable-owhttpd --enable-debian --enable-mt --disable-w1 --disable-usb --disable-i2c --disable-fuse --prefix=/usr
make -j4
make install
```

but it ends with error:

> checking if usb support is enabled... no
./configure: line 16918: syntax error near unexpected token `LIBUSB,'
./configure: line 16918: `      PKG_CHECK_MODULES(LIBUSB, libusb-1.0 >= 0.9.1, ENABLE_USB=true,ENABLE_USB=false)'
make: *** No targets specified and no makefile found.  Stop.
make: *** No rule to make target 'install'.  Stop.

Why, when usb is disabled  ? 

my config: 
**dpkg -l "*usb*"**
```
Desired=Unknown/Install/Remove/Purge/Hold
| Status=Not/Inst/Conf-files/Unpacked/halF-conf/Half-inst/trig-aWait/Trig-pend
|/ Err?=(none)/Reinst-required (Status,Err: uppercase=bad)
||/ Name                                         Version                     Architecture                Description
+++-============================================-===========================-===========================-=============================================================================================
ii  libusb-0.1-4:amd64                           2:0.1.12-30                 amd64                       userspace USB programming library
ii  libusb-1.0-0:amd64                           2:1.0.21-1                  amd64                       userspace USB programming library
ii  libusb-dev                                   2:0.1.12-30                 amd64                       userspace USB programming library development files
un  usb-modeswitch                               <none>                      <none>                      (no description available)
```
**uname -a**
```
Linux gitlab 4.9.0-6-amd64 #1 SMP Debian 4.9.88-1+deb9u1 (2018-05-07) x86_64 GNU/Linux
```

rchovan, 2019-06-17:
I have manually commented in configure this, 
```
#if test "${ENABLE_USB}" != "false"; then
#<----->#LIBUSB_CONFIG
#<----->PKG_CHECK_MODULES(LIBUSB, libusb-1.0 >= 0.9.1, ENABLE_USB=true,ENABLE_USB=false)
#fi
```

but it ends with same error with AVAHI, 
```
checking if Avahi support is enabled... no
./configure: line 17009: syntax error near unexpected token `LIBAVAHI,'
./configure: line 17009: `          PKG_CHECK_MODULES(LIBAVAHI, avahi-client,  dnl ok, avahi found with pkg-config'
```

I have tried --disable-avahi and --enable-avahi=no.

miccoli, 2019-06-17:
Is [`pkg-config` ](https://packages.debian.org/stretch/pkg-config) installed? Your error seems to be linked to the fact that the `PKG_CHECK_MODULES` macro is undefined.

rchovan, 2019-06-17:
thank you, pkg-config was not installed. Closed, Solved.

closed 2019-06-17 by rchovan