owfs/owfs #39: [owserver] DS18B20 can read other parameters, but not temperature

rchovan, 2019-06-19:
Hi, 
I have strange problem with owserver. I can read adress, id and other parameters, but not temperature, fasttemp, etc. I'm using passive components with parasitic power connected to rs232 and DS18B20 sensors. I have tried 1 passive with non-parasitic power, but result is same.
Also I have tried to change sensors. 

I have tried version 3.2p2 and 3.2p3 cloned from git. Same results

my owfs config: 
cat /etc/owfs.conf | grep -v '#'
```
server:passive=/dev/ttyr00
server:passive=/dev/ttyr01
server:passive=/dev/ttyr02
server:passive=/dev/ttyr03
server:pid_file=/var/run/owserver.pid
server:timeout_volatile=120
server:timeout_serial=10

```

```
root@owfs:/opt/owfs-3.2p2# owdir /bus.1/
/bus.1/interface
/bus.1/28.161896050000
/bus.1/simultaneous
/bus.1/alarm
root@owfs:/opt/owfs-3.2p2# owdir /bus.1/28.161896050000
/bus.1/28.161896050000/address
/bus.1/28.161896050000/alias
/bus.1/28.161896050000/crc8
/bus.1/28.161896050000/errata
/bus.1/28.161896050000/family
/bus.1/28.161896050000/fasttemp
/bus.1/28.161896050000/id
/bus.1/28.161896050000/latesttemp
/bus.1/28.161896050000/locator
/bus.1/28.161896050000/power
/bus.1/28.161896050000/r_address
/bus.1/28.161896050000/r_id
/bus.1/28.161896050000/r_locator
/bus.1/28.161896050000/scratchpad
/bus.1/28.161896050000/temperature
/bus.1/28.161896050000/temperature10
/bus.1/28.161896050000/temperature11
/bus.1/28.161896050000/temperature12
/bus.1/28.161896050000/temperature9
/bus.1/28.161896050000/temphigh
/bus.1/28.161896050000/templow
/bus.1/28.161896050000/tempres
/bus.1/28.161896050000/type
root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/type
DS18B20root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/fasttemp
ServerRead: Data error on /bus.1/28.161896050000/fasttemp
root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/temperature
ServerRead: Data error on /bus.1/28.161896050000/temperature
root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/tid
ServerRead: Data error on /bus.1/28.161896050000/tid
root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/id
161896050000root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/crc8
68root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/power
ServerRead: Data error on /bus.1/28.161896050000/power
root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/address
2816189605000068root@owfs:/opt/owfs-3.2p2#
root@owfs:/opt/owfs-3.2p2# owread /bus.0/
ServerRead: Data error on /bus.0/
root@owfs:/opt/owfs-3.2p2# owread /bus.0/
ServerRead: Data error on /bus.0/
root@owfs:/opt/owfs-3.2p2# owread /bus.0/
ServerRead: Data error on /bus.0/
root@owfs:/opt/owfs-3.2p2# owread /bus.1/28.161896050000/fasttemp
ServerRead: Data error on /bus.1/28.161896050000/fasttemp
root@owfs:/opt/owfs-3.2p2#
```

Here is complete debug log. 
[owserver_debug.txt](https://github.com/owfs/owfs/files/3304735/owserver_debug.txt)

Here is stripped debug log. 
```
CONNECT: ow_select.c:(229) Select error for DS18B20 on bus /dev/ttyr00
  DEBUG: ow_transaction.c:(91) select = 1
  DEBUG: ow_cache.c:(941) 28 4F 48 96 05 00 00 45 size=4
  DEBUG: ow_cache.c:(1070) Search in cache sn 28 4F 48 96 05 00 00 45 pointer=0x7f8c5423e2ec index=-2 size=4
  DEBUG: ow_cache.c:(1106) Value not found in cache
  DEBUG: ow_select.c:(70) Selecting a path (and device) path=/28.4F4896050000/temperature12 SN=28 4F 48 96 05 00 00 45 last path=00 00 00 00 00 00 00 00
  DEBUG: ow_select.c:(84) Continuing root branch
  DEBUG: ow_tcp_read.c:(63) attempt 1 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 1 - 0 = 1
  DEBUG: ow_select.c:(226) Selecting device 28 4F 48 96 05 00 00 45
  DEBUG: ow_bus_data.c:(120) Splitting byte 0 of 9 = 55
  DEBUG: ow_bus_data.c:(120) Splitting byte 1 of 9 = 28
  DEBUG: ow_bus_data.c:(120) Splitting byte 2 of 9 = 4F
  DEBUG: ow_bus_data.c:(120) Splitting byte 3 of 9 = 48
  DEBUG: ow_bus_data.c:(120) Splitting byte 4 of 9 = 96
  DEBUG: ow_bus_data.c:(120) Splitting byte 5 of 9 = 05
  DEBUG: ow_bus_data.c:(120) Splitting byte 6 of 9 = 00
  DEBUG: ow_bus_data.c:(120) Splitting byte 7 of 9 = 00
  DEBUG: ow_bus_data.c:(120) Splitting byte 8 of 9 = 45
  DEBUG: ow_tcp_read.c:(63) attempt 24 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 24 - 0 = 24
  DEBUG: ow_tcp_read.c:(63) attempt 24 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 24 - 0 = 24
  DEBUG: ow_tcp_read.c:(63) attempt 24 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 24 - 0 = 24
  DEBUG: ow_bus_data.c:(141) Consolidating byte 0 of 9 = 55
  DEBUG: ow_bus_data.c:(141) Consolidating byte 1 of 9 = 28
  DEBUG: ow_bus_data.c:(141) Consolidating byte 2 of 9 = EF
  DEBUG: ow_bus_data.c:(141) Consolidating byte 3 of 9 = 48
  DEBUG: ow_bus_data.c:(141) Consolidating byte 4 of 9 = 96
  DEBUG: ow_bus_data.c:(141) Consolidating byte 5 of 9 = 55
  DEBUG: ow_bus_data.c:(141) Consolidating byte 6 of 9 = 00
  DEBUG: ow_bus_data.c:(141) Consolidating byte 7 of 9 = 80
  DEBUG: ow_bus_data.c:(141) Consolidating byte 8 of 9 = 47
  DEBUG: ow_bus_data.c:(38) Response doesn't match data sent
  DEBUG: ow_transaction.c:(113) send = 1
CONNECT: ow_select.c:(229) Select error for DS18B20 on bus /dev/ttyr00
  DEBUG: ow_transaction.c:(91) select = 1
  DEBUG: ow_read.c:(620) Read /28.4F4896050000/temperature12 Extension 0 Gives result -22
  DEBUG: ow_parsename.c:(133) /28.4F4896050000/temperature12
  DEBUG: ow_read.c:(620) Read /28.4F4896050000/temperature Extension 0 Gives result -22
  DEBUG: ow_read.c:(253) return=-22
  DEBUG: ow_read.c:(263) After read is performed (bytes or error -22)
OWQ OneWireQuery structure of /28.4F4896050000/temperature
    OneWireQuery size=12 offset=0, extension=0
Byte buffer OneWireQuery buffer, length=12
--000: 00 00 00 00 00 00 00 00 00 00 00 00
   <............> 
```

ianka, 2019-06-19:
Could you please ```$ owdir /uncached``` instead? In the owserver log it says it doesn't see that chip on the bus so this could just be a cache issue – you are trying to access a chip you had connected before, but which isn't connected any more. The address, id and other static data don't require the chip to be actually connected.

rchovan, 2019-06-19:
```
root@owfs:/etc# owdir /uncached
/uncached/28.1EEA42030000
/uncached/28.161896050000
/uncached/28.131743030000
/uncached/bus.2
/uncached/bus.1
/uncached/bus.0
/uncached/settings
/uncached/system
/uncached/statistics
/uncached/structure
/uncached/simultaneous
/uncached/alarm

root@owfs:/etc# owdir /uncached/28.161896050000
/uncached/28.161896050000/address
/uncached/28.161896050000/alias
/uncached/28.161896050000/crc8
/uncached/28.161896050000/errata
/uncached/28.161896050000/family
/uncached/28.161896050000/fasttemp
/uncached/28.161896050000/id
/uncached/28.161896050000/latesttemp
/uncached/28.161896050000/locator
/uncached/28.161896050000/power
/uncached/28.161896050000/r_address
/uncached/28.161896050000/r_id
/uncached/28.161896050000/r_locator
/uncached/28.161896050000/scratchpad
/uncached/28.161896050000/temperature
/uncached/28.161896050000/temperature10
/uncached/28.161896050000/temperature11
/uncached/28.161896050000/temperature12
/uncached/28.161896050000/temperature9
/uncached/28.161896050000/temphigh
/uncached/28.161896050000/templow
/uncached/28.161896050000/tempres
/uncached/28.161896050000/type

root@owfs:/etc# owread /uncached/28.161896050000/temperature
ServerRead: Data error on /uncached/28.161896050000/temperature

```

and here is debug log: 
[owstr.txt](https://github.com/owfs/owfs/files/3305149/owstr.txt)

rchovan, 2019-06-19:
partial log for this test: [owserver_partial.txt](https://github.com/owfs/owfs/files/3305347/owserver_partial.txt)

```
root@owfs:/etc# owread /uncached/28.161896050000/temperature
ServerRead: Data error on /uncached/28.161896050000/temperature

root@owfs:/tmp# owget /uncached/bus.0/28.161896050000/power
ServerRead: Data error on /uncached/bus.0/28.161896050000/power

root@owfs:/tmp# owget /uncached/bus.0/28.161896050000/address
2816189605000068

root@owfs:/tmp# owget /uncached/bus.0/28.161896050000/id
161896050000

root@owfs:/tmp# owget /uncached/bus.0/28.161896050000/type
DS18B20

root@owfs:/tmp# owget /uncached/bus.0/28.161896050000/fasttemp
ServerRead: Data error on /uncached/bus.0/28.161896050000/fasttemp
```

ianka, 2019-06-19:
I cannot correlate the owserver log to your owget attempts. Could you please. 1.  Remove all the temperature sensors but e.g the 28.161896050000 one. 2. Restart owserver without the config file but use ```# owserver --debug -d /dev/ttyr00 ''' instead for a minimal log. 3. Do one single ```$ owget /uncached/28.161896050000/latesttemp```

rchovan, 2019-06-19:
When I tried --device / -d parameter, it looks for DS9097U, 

``` DEBUG: ow_ds9097U.c:(660) Failed first attempt at resetting baud rate of bus master /dev/ttyr00
  DEBUG: ow_tcp_read.c:(63) attempt 1 bytes Time: 5.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 1 - 0 = 1
  DEBUG: ow_ds9097U.c:(643) wrong response (4F not 00)
  DEBUG: ow_ds9097U.c:(665) Failed second attempt at resetting baud rate of bus master /dev/ttyr00
  DEBUG: ow_tcp_read.c:(63) attempt 1 bytes Time: 5.000000 seconds
```
but I have passive adapter, made at home from schematic for DS9097 without bus master, so I run debug with this parameters 
```
owserver --debug --6bit --baud=9600 --passive=/dev/ttyr00
```
then I have tried single read 
```
owfs@gitlab:/tmp$ owread /uncached/28.161896050000/latesttemp
ServerRead: Data error on /uncached/28.161896050000/latesttemp
```
and here is log: [owsingle.txt](https://github.com/owfs/owfs/files/3307522/owsingle.txt)

Then I have restarted owserver, and tried multiple reads 
```
owfs@gitlab:/tmp$ owdir /uncached/
/uncached/bus.0
/uncached/settings
/uncached/system
/uncached/statistics
/uncached/structure
owfs@gitlab:/tmp$ owdir /uncached/
/uncached/28.161896050000
/uncached/bus.0
/uncached/settings
/uncached/system
/uncached/statistics
/uncached/structure
/uncached/simultaneous
/uncached/alarm
owfs@gitlab:/tmp$ owread /uncached/28.161896050000/latesttemp
ServerRead: Data error on /uncached/28.161896050000/latesttemp
owfs@gitlab:/tmp$ owread /uncached/28.161896050000/fasttemp
ServerRead: Data error on /uncached/28.161896050000/fasttemp
owfs@gitlab:/tmp$ owread /uncached/28.161896050000/temperature
ServerRead: Data error on /uncached/28.161896050000/temperature
```

and here is log: [owmulti.txt](https://github.com/owfs/owfs/files/3307525/owmulti.txt)

ianka, 2019-06-19:
In line 70 of owsingle.txt, owserver complains about the slave not being on the bus. But it responds to the SEARCH ROM command otherwise you wouldn't see it in the /uncached listing. A few lines above owserver complains the response doesn't match the sent data. I'm not familiar with the passive adapter but the problem is pretty certainly located here.

ianka, 2019-06-19:
The CC 66 command in the log doesn't even address your slave chip but is a general command to all DS2409 which may be on the bus to turn off both branches. And this fails already.

rchovan, 2019-06-20:
ok, today I buy new parts, and I'm going to build new one adapter with non parasitic power. Btw, old site is down, do you have any schematic with parts for DS9097U ? I know, that illustration schematic is in datasheet, but without part list and values.

ianka, 2019-06-20:
Check the [DS2480B datasheet](https://datasheets.maximintegrated.com/en/ds/DS2480B.pdf. ). It has some example circuits.

rchovan, 2019-06-21:
At the least, the problem was not in adapter. New adapter acts like the old one. Then I have used usb<>rs232 cable converter and it started working. So something is wrong with my rs233 port. But it is strange, that it can see that sensor, read some data, but not temperature. Maybe something with voltage. 
Anyway, thank you for you time and help.

closed 2019-06-21 by rchovan