owfs/owfs #40: ds1977: how to write to memory?

MichelBr, 2019-07-30:
Hello,

I'm trying to use owfs with an ds1977 and i would like to write to its memory. The ibutton is recognized and I can read its type, address, ... but when I try to read or write to its memory, owfs logs error -22, for which I can't find a description. As far as I know there is no password set yet as an older version of owfs didn't support it. I'm now using the latest release using an rpi i2c-extensions for onewire.

It is possible to write to the memory of a ds1977 using owfs and read from it?

Thank you and kind regards,

Michel

ianka, 2019-07-31:
Errno==-22 is EINVAL. Unfortunately, this is a very common error code within owfs. We need the owserver log. Stop owserver (and check it's stopped), then restart it inm a terminal by hand with 

```# owserver --debug --w1```

(or whatever host adapter you use). Open another terminal and use 

```$ owread /uncached/<id>/memory```

Post the owserver log.

miccoli, 2021-08-19:
Closing this issue because there was no feedback. Please feel free to ping if you think it should be reopenend.

closed 2021-08-19 by miccoli