owfs/owfs #41: owhttpd 3.2p3 segfaults

spaceboy-cz, 2019-08-11:
There seems to be a bug in `owhttpd` `3.2p3` talking to `owserver` both running on RPI 3 / Raspbian Buster. `owhttpd` segfaults after few minutes under moderate load - with some 23 sensors on the bus being polled every minute or so. 
Downgrading to `3.1p5-1` makes is stable again.
Syslog: 
```
Aug  6 22:47:15 spacehouse systemd-coredump[436]: Process 30700 (owhttpd) of user 103 dumped core.
Stack trace of thread 399:
#0  0x0000000076d839e0 parse_reg_exp (libc.so.6)
#1  0x0000000076d81e18 parse_sub_exp (libc.so.6)
#2  0x0000000076d8389c parse_branch (libc.so.6)
#3  0x0000000076d83a10 parse_reg_exp (libc.so.6)
#4  0x0000000076d83ef8 parse (libc.so.6)
#5  0x0000000076d8503c __GI___regcomp (libc.so.6)
#6  0x0000000076efa10c ow_regcomp (libow-3.2.so.3)
```

What should I provide in order to help debug this?

miccoli, 2019-08-12:
@spaceboy-cz thanks for reporting.

Please provide details on your setup: 
- command used to start the server
- `/etc/owfs.conf` if relevant
- is the failing owhttpd compiled by you, or the raspbian packaged one?
- how did you downgrade to `3.1p5-1`?

Finally do you have a reproducible way of causing the segfault? (e.g. start the owhttpd server and querying a specific url via [curl](https://curl.haxx.se/docs/manpage.html)?)

miccoli, 2019-08-12:
I was able to replicate this bug on RPI model B /Raspbian.

<details><summary>Rasbian config</summary>

```
ii  ow-shell                         3.2p3+dfsg1-2               armhf
ii  owfs-common                      3.2p3+dfsg1-2               all
ii  owhttpd                          3.2p3+dfsg1-2               armhf
ii  owserver                         3.2p3+dfsg1-2               armhf
```
</details>

If I start tho `owhttpd` as follows
```
owhttpd --foreground -s 127.0.0.1 -p 8989
```
and from a different machine I run 
```
ab -s 5 -r -k -n 100 -c 8 http://myserver:8989/
```
owhttpd segfaults.

The key point here is "flooding" the owhttpd with concurrent requests (`-c 8`): if I run 
```
ab -s 5 -r -k -n 100 -c 1 http://myserver:8989/
```
(`-c 1` is no concurrency) owhttpd just works fine.

Curiously I was not able to break owhttpd running `ab` from the same host:
```
ab -r -s 5 -k -n 100 -c 8 http://127.0.0.1:8989/
This is ApacheBench, Version 2.3 <$Revision: 1843412 $>
Copyright 1996 Adam Twiss, Zeus Technology Ltd, http://www.zeustech.net/
Licensed to The Apache Software Foundation, http://www.apache.org/

Benchmarking 127.0.0.1 (be patient)...apr_pollset_poll: The timeout specified has expired (70007)
Total of 24 requests completed
```
The `ab` errors out but the owhttpd daemon remains alive.

miccoli, 2019-08-12:
Combining
<details> <summary>stack trace</summary>

```
Process 14073 (owhttpd) of user 1000 dumped core.
                                                
                                                Stack trace of thread 14331:
                                                #0  0x00000000b6d58650 re_compile_fastmap_iter (libc.so.6)
                                                #1  0x00000000b6d60980 __re_compile_fastmap (libc.so.6)
                                                #2  0x00000000b6d61058 __GI___regcomp (libc.so.6)
                                                #3  0x00000000b6ed610c ow_regcomp (libow-3.2.so.3)
```
</details>
and
<details> <summary>gdb backtrace</summary>

```
[Current thread is 1 (Thread 0xb60ff3f0 (LWP 14331))]
(gdb) bt
#0  0xb6d58650 in re_compile_fastmap_iter (bufp=bufp@entry=0x28138, 
    fastmap=fastmap@entry=0x119c960 "", init_state=<optimized out>, 
    init_state=<optimized out>) at regcomp.c:301
#1  0xb6d60980 in __re_compile_fastmap (bufp=bufp@entry=0x28138)
    at regcomp.c:274
#2  0xb6d61058 in __GI___regcomp (preg=0x28138, 
    pattern=0x1524c "host *: *([^ ]+) *\r", cflags=<optimized out>)
    at regcomp.c:508
#3  0xb6ed610c in ow_regcomp ()
   from /usr/lib/arm-linux-gnueabihf/libow-3.2.so.3
#4  0x00011f58 in ?? ()
#5  0x000118e8 in ?? ()
#6  0xb6ec447c in ?? () from /usr/lib/arm-linux-gnueabihf/libow-3.2.so.3
#7  0xb6dfc494 in start_thread (arg=0xb60ff3f0) at pthread_create.c:486
#8  0xb6d7f578 in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```
</details>

I think that the problem I'm experiencing is with the last line of this code-block
<https://github.com/owfs/owfs/blob/a3b0de7ecdea4e7d19ac764c50d060890f5c1f02/module/owhttpd/src/c/owhttpd_handler.c#L547-L556>

miccoli, 2019-08-12:
Final comment for today: I can confirm that `owhttpd` leaks memory. When I run `ab` with no concurrency  owhttpd does not crash, but DRS and RSS (as reported by `ps -v`) grow without bound.

leschNUX, 2019-09-15:
Same Problem on my raspberry 3b+.
owget works but http crashed

smrdkoster, 2019-11-26:
I have the same problem on number of RPI versions b - 3+. Only owhttp crashes and after restart it works normally with owserver.

The only solution on buster was to recompile the version 3.1p5, none of the 3.2xx versions worked.

fvonberg, 2022-08-14:
Hello,
is this issue going to be fixed soon or can someone explain me how to downgrade to 3.1p5 on a raspberry?
I tried to build from the release file but ran into errors building owhttpd.
Thanks.

miccoli, 2022-08-18:
> is this issue going to be fixed soon or can someone explain me how to downgrade to 3.1p5 on a raspberry?

Unfortunately this project is not actively supported, so I would not expect that a new release will be made soon.

My personal advise is to avoid using owhttpd: if possible try to use owserver and one of the language bindings available in `module/ownet` or other projects that interface with the ownet protocol.

As what regards the problems compiling 3.1p5, please open a new issue with mode details, if you need support.