owfs/owfs #42: 1-wire web for ds1991

patmassen, 2019-09-18:
I set up a 1-wire web for a ds1991 using a ds9490r usb.
Its giving the id of the adress type etc.
When i go to settings there are displayed the three id from the subkeys.
But when i fill in the correct password of id 0,1 or 2 the zeros stay the same from the pages (subkeys). Normally they give carbage when you fill in the wrong pass but here i only get zeros.
Has anyone a solution for this?

miccoli, 2019-09-18:
Sorry but I do not understand your issue: please provide more context. 
- which program are you using?
- which is your config file?
- etc.

patmassen, 2019-09-19:
Hi Micolli,
Thank you for responding me!

I am using Webserver
 /opt/owfs/bin/owhttpd -u -p 4444

The program does recognize my ds1991 ibutton and the three id fields.
But the subkeys give only 00000000000000 if i change the password i still get zeros.
Normally when i give a wrong password i get random data in the subkey (memory) field.
But here i got always zeros. Has this program been tested?

In fact i have the same problems as posted here: https://sourceforge.net/p/owfs/support-requests/21/
i have the correct password, but even if i didn't, the program must not give zero as data for the subkeys

I hope there is a fix for this problem. I am not a engineer, just a simple man trying to understand and learn.

Patrick

ianka, 2019-09-19:
You have to prefix the onewire path before the chip id with ```uncached/``` to have owhttpd read fresh data from the chip.

patmassen, 2019-09-20:
Hi Ianka,
Thank you for your reply and time.
I am going to find out how to prefix it.
This can take me some days because i have never done it but i am going to try it!
Its a adventure for me :)
If you have an example where to find that path for me that could help me to learn

Kind regards,

Patrick

Kind regards,

Patrick

ianka, 2019-09-20:
Just click on *uncached* in the main directory. You will find the same structure, but this time the underlying library will read from the chip on each access. (You can also see how the URL changes.)

patmassen, 2019-09-21:
Hi ianka,

Thanks again!
Well i am making steps in the good direction. When i reset password in uncached i can make a new ID. But when I open it in the normal settings i see the new ID but the pages still give only zeros (0000000000000000000000000000000000000000000000000000000000000000
00000000000000000000000000000000

It looks like it does not read the subkeys. Even when I reset passwrd in uncached and make a new ID and a new page(=subkey) 0 (or 1 or 2). In the normal setting i only get the new ID to see and the subkeys still give the zeros.....

patmassen, 2019-09-21:
This i was forgotten to tell: When i reset the password in uncached i set a new passowrd en ID and a page, the ID and Passoword are correctly changed (i can confirm that with the program ibutton viewer32) but the page(subkey) is empty andd not written with the new info. So it looks like there is something wrong with readin and writing the subkey

ianka, 2019-09-21:
Please test this with *owserver* instead of *owhttpd* and *owread/owwrite/owdir* instead of a web browser. *Owhttpd* is a clever hack and wasn't meant for such complicated stunts, neither is your web browser.

patmassen, 2019-09-21:
Okay, i will try owserver. Thanks again for your time and effort.

patmassen, 2019-09-23:
Hi Ianka,

Well i am trying to understand how to read the ds1991 but i am afraid that i am stuck here.

ls /mnt/1wire/02.F5BFE7000000
address  alias	crc8  family  id  locator  r_address  r_id  r_locator  subkey0	subkey1  subkey2  type

 ls /mnt/1wire/02.F5BFE7000000/subkey0
id.xxx	password.xxx  reset.xxx  secure_data.xxx

I found the path to the DS1991 but i dont know how to fill in the id and password so i can read the secure data. Or pherhaps i use the wrong method...

I hope you find the time to help me out on this one...

ianka, 2019-09-23:
Please don't use the owfs binary either. It has race conditions. Please set up owserver and use the owread/owwrite/owdir shell tools.

$ owwrite /uncached/02.F5BFE7000000/subkey0/password.xxx MYPASSWORD
$ owread /uncached/02.F5BFE7000000/subkey0/secure_data.xxx

patmassen, 2019-09-23:
Thanks! i am going to set up owserver and try again....

patmassen, 2019-09-27:
Well i am stuck again...when i am making a owdir i get this:

/10.67C6697351FF
/05.4AEC29CDBAAB
/bus.0
/uncached
/settings
/system
/statistics
/structure
/simultaneous
/alarm

owdir uncached:

 owdir /uncached
/uncached/10.67C6697351FF
/uncached/05.4AEC29CDBAAB
/uncached/bus.0
/uncached/settings
/uncached/system
/uncached/statistics
/uncached/structure
/uncached/simultaneous
/uncached/alarm

ls /mnt/1wire:

02.F5BFE7000000  10.67C6697351FF  bus.0  settings    structure  uncached
05.4AEC29CDBAAB  81.AD612C000000  bus.1  statistics  system

So the ds1991 (02.F5BFE70000000) is connected but not visible in owdir(uncached).

What have i done wrong?

Best regards,

Patrick

ianka, 2019-09-27:
I guess you run **both** owserver and owfs *concurrently* on the same host adapter. This is prone to fail as there is a mix of hardware control on the driver level then. Please stop the owfs binary!

patmassen, 2019-10-18:
Hi,

I have solved the problem and learn a lott :) thanks for your help.

The config of the owserver was not correct.
With the command sudo nano /etc/owfs.conf i deleted the fake server and configure the correct usb. I can now read the ID of the DS1991(its another one) with the command owread /uncached/02.FD7BA3000000/subkey0/id.xxx and get a good response. Thats great!

But when i write: owwrite /uncached/02.FD7BA3000000/subkey0/password.xxx UUUUUUUU it produces not the right data of the secure subkey (owread /uncached/02.FD7BA3000000/subkey0/secure_data.xxx). When i do that i get this as respond: W�����%patrick@patrick-AOA150:~$ F���. I filled the Subkey 0 secure data with UUUU with ibviewer32 so i know for sure that this is the secure data. My password is UUUUUUUU. So i would expect the correct data from the secure Subkey. Also when i fill it in with HEX I dont get the right response.

Any help would be appreciated, thanks in advance....

ianka, 2019-10-18:
owread and owwrite by default pass raw data.  You may like the --hex, --start and --size options. Pleaseread  the the owshell manpage.

patmassen, 2019-10-20:
Hi ianka,

Yess i did it! I can read and write in HEX. The data in the subkey had nonprintable information but was vital for me. Now i can read/write the info in HEX: owwrite --hex /02.F5BFE7000000/subkey0/secure_data...........  ......
Ibuttonviewer32 was not capable to do that, so i had to find another way.
I like to thank you for all your help and support!
It took me 3 months to understand the basics of owfs/owserver and learned a lott, it was a great adventure. The best is to install only owserver and owshell, configure owserver and you are ready to go. Owserver is the best utility and easy (when you understand what you are doing :) ).

Thanks again and i wish you the best!

Greetings 

Patrick

miccoli, 2019-12-03:
The conversation seems terminated and resolved to me, so I'm closing the issue. Please feel free to ask for reopening if I'm mistaken.

closed 2019-12-03 by miccoli