owfs/owfs #45: owserver memory leak?

bahm-lgs, 2019-11-27:
**Issue**
I am facing the issue of owserver continuously occupying more memory with each access.
- it does not matter whether accessing a temperature sensor or an eeprom
- it does not matter whether accessing via owfs or the owshell tools
- there is no difference between cached and uncached access
- even using just owpresent instead of owread results in growing memory consumption
- the bug was seen in version 3.1p1 and all newer releases, even with the latest master code from git

**Target system**
The owfs code is cross-compiled for aarch64 using x86/64 Linux gcc.
Binaries are running on a Xilinx Zynq MPSoC with Petalinux-2018.2 os.

**Compiler**
aarch64-linux-gnu-gcc (Linaro GCC 7.2-2017.11-rc1) 7.2.1 20171011

**Test log**
```
# ps ax | grep owserver
 9642 ?        Ssl    0:00 /usr/bin/owserver -c /etc/owfs.conf --pid_file /var/run/owserver.pid
 9938 pts/0    S+     0:00 grep owserver
# while true; do owread 28.2BC710060000/temperature; echo -n " -- "; grep RSS /proc/9642/status; sleep 1; done
     26.0625 -- VmRSS:      2900 kB
     26.0625 -- VmRSS:      2900 kB
     26.0625 -- VmRSS:      2904 kB
     26.0625 -- VmRSS:      2904 kB
     26.0625 -- VmRSS:      2904 kB
     26.0625 -- VmRSS:      2916 kB
     26.0625 -- VmRSS:      2920 kB
     26.0625 -- VmRSS:      2920 kB
     26.0625 -- VmRSS:      2920 kB
^C
# while true; do owread uncached/28.2BC710060000/temperature; echo -n " -- "; grep RSS /proc/9642/status; sleep 1; done
     26.0625 -- VmRSS:      2924 kB
      26.125 -- VmRSS:      2936 kB
      26.125 -- VmRSS:      2936 kB
      26.125 -- VmRSS:      2944 kB
     26.1875 -- VmRSS:      2956 kB
     26.1875 -- VmRSS:      2956 kB
     26.1875 -- VmRSS:      2956 kB
     26.1875 -- VmRSS:      2972 kB
     26.1875 -- VmRSS:      2972 kB
^C
# while true; do owexist 28.2BC710060000; echo -n " -- "; grep RSS /proc/9642/status; sleep 1; done
 -- VmRSS:          2972 kB
 -- VmRSS:          2976 kB
 -- VmRSS:          2976 kB
 -- VmRSS:          2976 kB
 -- VmRSS:          2976 kB
 -- VmRSS:          2980 kB
 -- VmRSS:          2980 kB
 -- VmRSS:          2980 kB
 -- VmRSS:          2984 kB
^C
# while true; do owpresent 28.2BC710060000; echo -n " -- "; grep RSS /proc/9642/status; sleep 1; done
1 -- VmRSS:         2988 kB
1 -- VmRSS:         2988 kB
1 -- VmRSS:         2992 kB
1 -- VmRSS:         2992 kB
1 -- VmRSS:         2992 kB
1 -- VmRSS:         2996 kB
1 -- VmRSS:         2996 kB
1 -- VmRSS:         2996 kB
1 -- VmRSS:         3000 kB
^C
#

```

ianka, 2019-11-27:
Could you also post your /etc/owfs.conf?

bahm-lgs, 2019-11-27:
```
# Sample configuration file for the OWFS suite for Debian GNU/Linux.
#
#
# This is the main OWFS configuration file. You should read the
# owfs.conf(5) manual page in order to understand the options listed
# here.

######################## SOURCES ########################
#
# With this setup, any client (but owserver) uses owserver on the
# local machine...
! server: server = localhost:4304 
#
#
############## W1 kernel module devices #################
#
server: w1 = all
#
######################### OWFS ##########################
#
mountpoint = /mnt/1wire
#
####################### OWSERVER ########################

server: port = 4304

```

bahm-lgs, 2019-11-27:
I already compiled the sources with `--enable-owmalloc`
But this did not reveal any unfreed memory allocations while issuing simple owexist commands although RSS increments by some KB with every command.

ianka, 2019-11-28:
I try to catch that problem in my own setup.

miccoli, 2019-11-28:
Just for the records: I'm running owserver 3.2p3 on a raspberry pi 24/7 without problems.

I performed some stress tests recording `size` and `resident` from `/proc/<pid>/statm` (same as `VmSize` and `VmRSS` but measured in pages).
Both quantities increase during the tests, but after a while they decrease again to the "normal" values.

Here a plot:
[image]

@bahm-lgs what about VmSize? if you wait some time after running your tests, does the memory occupation decreases again?

<details>
Matplotlib code adapted from [gallery example](https://matplotlib.org/gallery/api/two_scales.html) with data from this script:

```
OWPID=xxx
while true; do echo $(date +%s.%N) $(< /proc/$OWPID/statm); sleep 1 ; done > /tmp/data
```

</details>

bahm-lgs, 2019-11-29:
VmSize does not increase, only VmRSS. And it never decreases. It increases in my Setup by About 2MB per minute and the system stops working after about 15 Hours when no memory is left.

miccoli, 2019-11-29:
I would assume that VmRSS ≤ VmSize, so unless owserver starts with a huge VmSize, I think that also VmSize should increase over time. Please note that in my plot the scale for `size` and `resident` are different, so that it is always VmSize ≫ VmRSS. 

I failed to mention before that my instance uses the USB driver, and not the w1, so it is in that direction that we should look.

bahm-lgs, 2019-11-29:
Here is a few minutes of the /proc/pid/statm log from my system:
[data.txt](https://github.com/owfs/owfs/files/3905064/data.txt)

Unfortunately timestamp resolution is only in seconds, as the busybox date command does not support %N

bahm-lgs, 2019-11-29:
Interesting finding: I got stable RSS size for loops with owexist when not letting the server thread detach itself and letting the main thread join the created threads afterwards. I am aware this is not the functionality we want for the Server and it also does not fix the leak when using owpresent instead of owexist. But it may be a hint that we have a problem with freeing resources for detached threads...

miccoli, 2019-12-02:
> Interesting finding: ... But it may be a hint that we have a problem with freeing resources for detached threads...

Are you compiling with 'Mutexdebug' enabled or not? In fact this variable is not only about debug output, but also on how multithreading rw_locks are implemented.

bahm-lgs, 2019-12-03:
Looks like mutexdebug is enabled by default, at least my config.status contains S["OW_MUTEX_DEBUG"]="1" and my toplevel Makefile contains ENABLE_MUTEX_DEBUG = true

I also tried setting processor affinity to a single core using taskset, but this did not make a difference regarding the leaking memory.

bahm-lgs, 2019-12-03:
I think I found the issue, verification still pending, but this glibc bug seems to be exactly what I am experiencing:

https://sourceware.org/bugzilla/show_bug.cgi?id=22111

And yes, I am using glibc 2.26, because this was the default for Xilinx Petalinux-2018.2. I will try updating or at least patching the current version.

bahm-lgs, 2019-12-03:
Patching glibc 2.26 with the patch from https://sourceware.org/bugzilla/show_bug.cgi?id=22111 fixed the problem. Thanks @miccoli and @ianka for your help and sorry for reporting such an old bug. But unfortunately this is what happens when you have to stick to older software releases due to other bugs with newer releases...

closed 2019-12-03 by bahm-lgs

ianka, 2019-12-03:
Puzzling, my first guess was some unfreed ressources on thread join. But I did not want to shove the problem elsewhere before investigating it.

miccoli, 2019-12-03:
@bahm-lgs no problem. Reporting your problem here may help others with the same problem.