owfs/owfs #46: Error installing owserver on RPI4

nattila1, 2019-11-29:
Brand new RPI 4, with latest Raspbian Buster Lite. Attempted to install owserver and received an error.

```
Creating config file /etc/owfs.conf with new version
Setting up libow-3.2-3:armhf (3.2p3+dfsg1-2) ...
Setting up owserver (3.2p3+dfsg1-2) ...
Job for owserver.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status owserver.service" and "journalctl -xe" for details.
invoke-rc.d: initscript owserver, action "start" failed.
● owserver.service - Backend server for 1-wire control
   Loaded: loaded (/lib/systemd/system/owserver.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: protocol) since Thu 2019-11-28 19:37:38 CET; 59ms ago
     Docs: man:owserver(1)
  Process: 21434 ExecStart=/usr/bin/owserver -c /etc/owfs.conf (code=exited, status=0/SUCCESS)
 Main PID: 21434 (code=exited, status=0/SUCCESS)
dpkg: error processing package owserver (--configure):
 installed owserver package post-installation script subprocess returned error exit status 1
Setting up owhttpd (3.2p3+dfsg1-2) ...
Created symlink /etc/systemd/system/multi-user.target.wants/owhttpd.service → /lib/systemd/system/owhttpd.service.
Processing triggers for systemd (241-7~deb10u2+rpi1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10+rpi1) ...
Errors were encountered while processing:
 owserver
Updating FireMotD available updates count ... 
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Despite errors Owserver seemed to be active after installation according to following command
sudo systemctl status owserver.service

After reboot Owserver didn’t start. It was reported “dead”.

```
● owserver.service - Backend server for 1-wire control
Loaded: loaded (/lib/systemd/system/owserver.service; disabled; vendor preset: enabled)
Active: inactive (dead)
Docs: man:owserver(1)
```

ianka, 2019-11-29:
What's inside your /etc/owfs.conf? Which host adapter do you use?

miccoli, 2019-11-30:
This problem is clearly downstream OWFS: I mean it's not an OWFS bug, but is related to the way in which OWFS is packaged in Debian and derivates. (And we have no control of what is done downstream of us.)

You can find a brief discussion in this [post](https://marc.info/?l=owfs-developers&m=156875225614743&w=2) on the OWFS developer mailing list, under point 2.

Briefly: you can try to fix this problem as follows. First from the shell
```
sudo systemctl stop owserver
sudo systemctl edit owserver.service
```
and in the editor enter these lines
```
[Service]
ExecStart=
ExecStart=/usr/bin/owserver -c /etc/owfs.conf --foreground

[Install]
Also=
```
and then from the shell again
```
sudo systemctl daemon-reload
sudo systemctl enable owserver
sudo systemctl start owserver
```

nattila1, 2019-11-30:
> What's inside your /etc/owfs.conf? Which host adapter do you use?

Error comes up with default /etc/owfs.conf file as well. I didn't modify anything on that

nattila1, 2019-11-30:
> Briefly: you can try to fix this problem as follows.

Thank you four your detailed instructions. It looks like it solved the issue. On reboot staus of owserver nad owhttpd are active. I can see only one warning regarding to owserver. Is that important?

```
[19:26:53] openhabian@openhab:~$ sudo systemctl status owserver.service
[sudo] password for openhabian: 
Warning: The unit file, source configuration file or drop-ins of owserver.service change
● owserver.service - Backend server for 1-wire control
   Loaded: loaded (/lib/systemd/system/owserver.service; enabled; vendor preset: enabled
  Drop-In: /etc/systemd/system/owserver.service.d
           └─override.conf
   Active: active (running) since Sat 2019-11-30 19:22:04 CET; 4min 55s ago
     Docs: man:owserver(1)
 Main PID: 347 (owserver)
    Tasks: 3 (limit: 1718)
   Memory: 1.9M
   CGroup: /system.slice/owserver.service
           └─347 /usr/bin/owserver -c /etc/owfs.conf --foreground

Nov 30 19:22:04 openhab systemd[1]: Starting Backend server for 1-wire control...
Nov 30 19:22:04 openhab owserver[347]: DEFAULT: ow_opt.c:(670) systemd mode -- ignore 'p
Nov 30 19:22:04 openhab owserver[347]: DEFAULT: ow_opt.c:(780) systemd mode -- ignore fo
Nov 30 19:22:04 openhab systemd[1]: Started Backend server for 1-wire control.

[19:27:08] openhabian@openhab:~$ sudo systemctl status owhttpd.service
● owhttpd.service - Tiny webserver for 1-wire control
   Loaded: loaded (/lib/systemd/system/owhttpd.service; enabled; vendor preset: enabled)
   Active: active (running) since Sat 2019-11-30 19:22:04 CET; 5min ago
     Docs: man:owhttpd(1)
 Main PID: 407 (owhttpd)
    Tasks: 3 (limit: 1718)
   Memory: 864.0K
   CGroup: /system.slice/owhttpd.service
           └─407 /usr/bin/owhttpd --foreground -c /etc/owfs.conf

Nov 30 19:22:04 openhab systemd[1]: Starting Tiny webserver for 1-wire control...
Nov 30 19:22:04 openhab systemd[1]: Started Tiny webserver for 1-wire control.
 

```

miccoli, 2019-11-30:
@nattila1 you mean 
```
Warning: The unit file, source configuration file or drop-ins of owserver.service change...
```
If I understand it properly this should go away after `sudo systemctl daemon-reload` or reboot.

Please not that your logs are truncated at 80 columns: this is because `systemctl` pipes the output through `less`, in a mode that does not wrap lines. When copy-pasting `systemctl` output, please remember to add the `--no-pager` like this:

```
systemctl --no-pager status owserver.service
```

By the way I forgot that if you don't want the socket activation, you should also issue:
```
systemctl disable owserver.socket
```

A final comment: `systemd` support in owserver is far from perfect: the problem here is that it is not straightforward to implement a correct setup with both socket activation and `multi-user.target`. 

Maybe there is room for improvement: when I have some spare time I will investigate the problem and possibly open a different issue, or simply add some docs.

kalvdans, 2020-09-30:
Reported downstream to [Debian](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=971513) and [Ubuntu](https://bugs.launchpad.net/ubuntu/+source/owfs/+bug/1897966).

closed 2020-12-24 by miccoli