owfs/owfs #49: owserver stuck after exUnknownSensor exception while reading temperature value from sensor

puneetugru, 2019-12-03:
Hi,

I'm using the ow module of python in our project to read temperature values from different sensors connected to a Raspberry Pi.

Version details are as below,

/opt/owfs/bin/owserver version:
	3.2p2
libow version:
	3.2p2

Python 2.7.9

The initialization of ow server is done like below,

`ow.init('127.0.0.1:4304')`

After this, the list of sensors is stored using the below code,

`self.sensors = ow.Sensor("/").sensorList()`

After this, I read the temperature values of the sensors in a for loop.

**Also, the requirement is, if any sensor is not giving temperature value, I should be able to handle that case and continue reading the value from the next available sensor.**

```
for sensor in self.sensors:
    try:
        sensorTemperature = sensor.temperature
    except Exception as error:
        #PRINT_LOG
        continue
```
For some sensors, below exception is seen when trying to read the temperature,

exUnknownSensor: '/28.72297D080000/temperature'

As can be seen in the above code, exception handling is done and after this, I would like to continue with the loop reading temperature value from other sensors.

But, I'm not getting that desired result, i.e., the read loop is stuck and the program is not coming out of the loop.

I would like to know whether some API needs to be called before the continue keyword so that the read loop doesn't get stuck?

P.S: Please note that ow.init() is done only once and no specific timeout parameters are explicitly set.

- Thanks,
Puneet Ugru

miccoli, 2019-12-03:
> As can be seen in the above code, exception handling is done and after this, I would like to continue with the loop reading temperature value from other sensors.
>
> But, I'm not getting that desired [...]

There is no special API to call to resume processing. I have not enough information on what is really going on. On which line you think that the code is stuck?

In order to better understand what is going on you can invoke the interactive debugger like this:
```
import pdb
for sensor in self.sensors:
    try:
        sensorTemperature = sensor.temperature
    except Exception as error:
        pdb.set_trace()
        continue
```

In the exception clause the [pdb.set_trace](https://docs.python.org/2/library/pdb.html#pdb.set_trace) will start the debugger: you can interactively run the `step` or `next` commands to better understand where (and maybe why) your code is stuck, please see the [docs](https://docs.python.org/2/library/pdb.html#debugger-commands)

puneetugru, 2019-12-04:
Hi Stefano,

The code gets stuck in the below line,

`sensorTemperature = sensor.temperature`

Please note that this issue of loop getting stuck at the above line is intermittent when an exception is seen.

What I mean by my above statement is, for example, if the exception gets raised for multiple sensors and even if it happens multiple times, not always the read loop gets stuck.

The loop sometimes proceeds with reading the temperature data from the next available list of sensors and sometimes the loop doesn't proceed forward.

Hence, I cannot add the set_trace() at the Exception section of the code.

miccoli, 2019-12-04:
Unfortunately I have not enough info for reproducing the issue. The code fragment you provided looks good, so there is no hint on what is really going on.

Usually sensors do not disappear... may be you should first check if this issue is a physical problem of your 1-wire bus, more than an python ow software problem. I would suggest to add some debug print

```
for sensor in self.sensors:
    try:
        print "About to read", sensor
        sensorTemperature = sensor.temperature
    except Exception as error:
        #PRINT_LOG
        continue
```
so that you know on what sensor the program is stuck and you can check with the owshell programs (owdir, owread) if it is still reachable. 

A final comment:
ow python (libow swig bindings) is not actively maintained: if possible you should use a more modern python3 alternative, eg. [pyownet](https://pyownet.readthedocs.io/en/latest/) or the async [trio-owfs](https://github.com/python-trio/trio-owfs). BTW with pyownet you can set [timeouts](https://pyownet.readthedocs.io/en/latest/protocol.html#timeouts) and maybe it easier to debug your problem.

puneetugru, 2019-12-09:
Hi Stefano,

Thank you for your reply.

We're debugging this issue from various angles and I'll update this thread shortly with our observations.

puneetugru, 2019-12-09:
Hi Stefano,

I forgot to mention one thing about the code.

Before reading the sensor's data, we're trying to get a list of sensors, using the ow module.

To do that, first, we perform `ow.init()` and then we get a list of sensors.

After we get the list of sensors, to read data from them we go through them in a loop, as mentioned in the above comments.

But, before that loop, we've another `ow.init()` statement.

For now, I've commented out the second `ow.init()` and trying to reproduce the scenario, but somehow I forgot to mention this in the earlier comments.

My query is, can this double init cause some issues?

Before the second `ow.init() `call, should `ow.finish()` be done?

puneetugru, 2019-12-10:
Another observation is when this issue is seen on a particular interface, which is connected to multiple sensors, in the temperature_data_read_loop, some sensors are getting read repeatedly.

Let me try to explain this with an example,

Suppose there are 8 sensors connected to an interface, 1, 2, 3, 4, 5, 6, 7, 8 and we try to read the temperature from them one-by-one.

**Sensor 1 value is read.
Sensor 2 value is read.
Sensor 3 value is read.
Sensor 4 gives an exception - exUnknownSensor: '/4/temperature'

Sensor 5 value is read.**

Now, the loop should continue is as below,

**Sensor 6 value is read.
Sensor 7 value is read.
Sensor 8 value is read.**

But, it proceeds like below, 

**Sensor 1 value is read again.
Sensor 2 value is read again.
Sensor 3 value is read again.**

And then the read loop gets stuck.

Before we start the read loop, we get the list of sensors with below API call,

**ow.Sensor("/").sensorList()**

miccoli, 2019-12-12:
`ow.init` is meant to be called once in your program.

Unfortunately I have not enough info for reproducing your problem, so I cannot help. 

Let me state again that `ow` python (libow swig bindings) has to be considered legacy, so if possible should avoided in new programs.

puneetugru, 2019-12-17:
Hi Stefano,

Thank you for your response.

We've corrected the code to call ow.init only once in the program.

We will definitely try to integrate pyownet in our program.

andreycpcl, 2020-01-24:
> Hi,
> 
> I'm using the ow module of python in our project to read temperature values from different sensors connected to a Raspberry Pi.
> 
> Version details are as below,
> 
> /opt/owfs/bin/owserver version:
> 3.2p2
> libow version:
> 3.2p2
> 
> Python 2.7.9
> 
> The initialization of ow client module is done like below,
> 
> `ow.init('127.0.0.1:4304')`
> 
> After this, the list of sensors is stored using the below code,
> 
> `self.sensors = ow.Sensor("/").sensorList()`
> 
> After this, I read the temperature values of the sensors in a for loop.
> 
> **Also, the requirement is, if any sensor is not giving temperature value, I should be able to handle that case and continue reading the value from the next available sensor.**
> 
> ```
> for sensor in self.sensors:
>     try:
>         sensorTemperature = sensor.temperature
>     except Exception as error:
>         #PRINT_LOG
>         continue
> ```
> 
> For some sensors, below exception is seen when trying to read the temperature,
> 
> **exUnknownSensor: '/28.72297D080000/temperature'**
> 
> As can be seen in the above code, exception handling is done and after this, I would like to continue with the loop reading temperature value from other sensors.
> 
> But, I'm not getting that desired result, i.e., the read loop is stuck and the program is not coming out of the loop.
> 
> I would like to know whether some API needs to be called before the continue keyword so that the read loop doesn't get stuck?
> 
> * Thanks,
>   Puneet Ugru