owfs/owfs #50: How to pass options when starting owserver using Python ow module

puneetugru, 2019-12-03:
Hi,

I'm using the ow module of python in our project to read temperature values from different sensors connected to a Raspberry Pi.

Version details are as below,

/opt/owfs/bin/owserver version:
3.2p2
libow version:
3.2p2

Python 2.7.9

currently, the owserver is started with below code,

```
ow.opt('-c /etc/owfs.conf')
ow.init('127.0.0.1:4304')
```

But when I see the running process, the arguments don't have the config file name as part of them.

ps -eaf | grep owserver
root     27100 27086  0 08:48 ?        00:00:00 /opt/owfs/bin/owserver --w1 -p 4304 --foreground --error_level=0
root     27517 26320  0 09:00 pts/0    00:00:00 grep owserver

Also, I would like to provide a maximum timeout for reading temperature value from a sensor.

Please point me towards some example code for the same.

ianka, 2019-12-03:
You are mistaken. owserver isn't started by that python code. Instead, you advised the python owfs library to use the configuration file and (most likely) connect to your host adaptor directly.

Yes, it can do that —but you should use owserver instead, as it allows concurrent access—

Skip the ow.opt() line in your python program and edit the systemd startup file instead.

puneetugru, 2019-12-03:
My bad, owserver is running using systemd.
Our program is initializing the ow module to connect to the server.

Is there any way where we can tell the ow module to skip a faulty sensor by exception handling and proceed to next?

Please note that this question is also related to #49

ianka, 2019-12-03:
Stefano Miccoli is more firm on the python owfs bindings.

puneetugru, 2019-12-03:
Hi Jan,

Thank you for your responses.

How can this query and #49 be assigned to Stefano?

- Thanks,
Puneet

miccoli, 2019-12-03:
Closing, since as already mentioned, `owserver` should be started by systemd, not by python `ow`. Once this point is made clear this simply a duplicate of #49

closed 2019-12-03 by miccoli