owfs/owfs #51: MAX31826: Wrong temperature reported; fails to write to internal memory pages

andreycpcl, 2020-01-24:
It seems that for MAX31826 OWFS always returns the current temperature as -0.0625.
OWFS also fails to write to any internal memory page of MAX31826.

Here's the log for the temperature reading attempt:
```
ubuntu@ubuntu:~$ /usr/bin/owserver -c /etc/owfs.conf --debug --w1
DEBUG MODE
libow version:
        3.2p3
  DEBUG: ow_daemon.c:(170) main thread id = 281472919396368
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
   CALL: ow_parsename.c:(174) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
CONNECT: ow_w1_bind.c:(64) [Operation not permitted] Netlink (w1) bind (are you root?)
  DEBUG: ow_w1_monitor.c:(65) [Operation not permitted] Netlink problem -- are you root?
  DEBUG: ow_regex.c:(24) Reg Ex expression <^$> compiled to 0xffff85cd0358
  DEBUG: ow_regex.c:(24) Reg Ex expression <^all$> compiled to 0xffff85cd0398
  DEBUG: ow_regex.c:(24) Reg Ex expression <^scan$> compiled to 0xffff85cd03d8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^\*$> compiled to 0xffff85cd0418
  DEBUG: ow_regex.c:(24) Reg Ex expression <^[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}$> compiled to 0xffff85cd0458
  DEBUG: ow_regex.c:(24) Reg Ex expression <^-?[[:digit:]]+$> compiled to 0xffff85cd0498
]*$> compiled to 0xffff85cd04d8 expression <^ *([^ ]+)[
]*$> compiled to 0xffff85cd0518 expression <^ *([^ ]+) *: *([^ ]+)[
]*$> compiled to 0xffff85cd0558 expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(100) 0: 0->10 found <></dev/i2c-1><>
  DEBUG: ow_regex.c:(100) 1: 0->10 found <></dev/i2c-1><>
  DEBUG: ow_parse_address.c:(87) Text </dev/i2c-1>
  DEBUG: ow_parse_address.c:(142) First </dev/i2c-1>
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 18
  DEBUG: ow_ds2482.c:(516) ok
CONNECT: ow_ds2482.c:(426) i2c device at /dev/i2c-1 address 18 appears to be DS2482-x00
  DEBUG: ow_ds2482.c:(785) Channel select set error
CONNECT: ow_ds2482.c:(692) DS2482-100 (Single channel)
  DEBUG: ow_ds2482.c:(676) Cannot set to port configuration -- not a DS2483
  DEBUG: ow_net_server.c:(76) ServerAddr: [localhost] [4304]
  DEBUG: ow_tcp_read.c:(63) attempt 24 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 24 - 0 = 24
  DEBUG: from_client.c:(65) FromClient payload=38 size=65536 type=2 sg=0x10A offset=0
  DEBUG: from_client.c:(73) FromClient (no servermessage) payload=38 size=65536 type=2 controlflags=0x10A offset=0
  DEBUG: ow_tcp_read.c:(63) attempt 38 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 38 - 0 = 38
  DEBUG: handler.c:(152) START handler /uncached/3B.A74463000000/temperature
   CALL: data.c:(103) DataHandler: parse path=/uncached/3B.A74463000000/temperature
  DEBUG: ow_parseobject.c:(163) /uncached/3B.A74463000000/temperature
   CALL: ow_parsename.c:(174) path=[/uncached/3B.A74463000000/temperature]
  DEBUG: ow_regex.c:(24) Reg Ex expression <^bus\.([[:digit:]]+)/?> compiled to 0xffff85cd05b8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^settings/?> compiled to 0xffff85cd05f8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^statistics/?> compiled to 0xffff85cd0638
  DEBUG: ow_regex.c:(24) Reg Ex expression <^structure/?> compiled to 0xffff85cd0678
  DEBUG: ow_regex.c:(24) Reg Ex expression <^system/?> compiled to 0xffff85cd06b8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^interface/?> compiled to 0xffff85cd06f8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^text/?> compiled to 0xffff85cd0738
  DEBUG: ow_regex.c:(24) Reg Ex expression <^json/?> compiled to 0xffff85cd0778
  DEBUG: ow_regex.c:(24) Reg Ex expression <^uncached/?> compiled to 0xffff85cd07b8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^unaliased/?> compiled to 0xffff85cd07f8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^alarm?> compiled to 0xffff85cd0838
  DEBUG: ow_regex.c:(24) Reg Ex expression <^simultaneous/?> compiled to 0xffff85cd0878
  DEBUG: ow_regex.c:(24) Reg Ex expression <^thermostat/?> compiled to 0xffff85cd08b8
  DEBUG: ow_regex.c:(24) Reg Ex expression <^/bus\.[[:digit:]]+/?> compiled to 0xffff85cd08f8
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.> compiled to 0xffff85cd0938
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.all$> compiled to 0xffff85cd0978
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.byte$> compiled to 0xffff85cd09b8
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.[[:digit:]]+$> compiled to 0xffff85cd09f8
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.[[:alpha:]]$> compiled to 0xffff85cd0a38
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(24) Reg Ex expression <^([[:xdigit:]]{2})\.?([[:xdigit:]]{12})\.?([[:xdigit:]]{2}){0,1}$> compiled to 0xffff85cd0a80
  DEBUG: ow_regex.c:(100) 0: 0->15 found <><3B.A74463000000><>
[... 704 lines cut ...]
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_ds2482.c:(516) ok
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(176) CRC8 = 0
  DEBUG: ow_transaction.c:(97) compare = 1
  DEBUG: ow_write.c:(495) Write /uncached/3B.A74463000000/pages/page.0 Extension 0 Gives result -22
  DEBUG: ow_write.c:(112) Error writing to /uncached/3B.A74463000000/pages/page.0
  DEBUG: ow_parsename.c:(133) /uncached/3B.A74463000000/pages/page.0
  DEBUG: data.c:(193) DataHandler: FS_ParsedName_destroy done
  DEBUG: data.c:(207) DataHandler: cm.ret=-22
  DEBUG: to_client.c:(74) payload=0 size=0, ret=-22, sg=0x10A offset=0
  DEBUG: to_client.c:(84) No data
  DEBUG: data.c:(226) Finished with client request
  DEBUG: handler.c:(134) OWSERVER handler done
  DEBUG: ow_net_server.c:(259) Normal completion.
```

ianka, 2020-01-29:
As you are using the w1 kernel driver, please check the w1_therm kernel module isn't loaded. There may be a race condition between that module and owfs.

tattersa-n, 2021-06-18:
Are there any other suggestions? I see the same issue, and the w1_therm module is not loaded.

[log.txt](https://github.com/owfs/owfs/files/6674667/log.txt)