owfs/owfs #53: Issues with configuration file using w1 protocol, and other issues

snowdd1, 2020-01-31:
After having read #34 and #46 I have noted the following:
On my Raspberry Zero W, using Buster based Raspbian, I
* could not use a configuration file for neither owfs nor owserver: for both `server: w1` is ignored
* needed to specify `--foreground` as additional parameter (in addition to `--w1`) in the service file of owserver to avoid protocol errors

I hope that can help others struggling to use the 1wire kernel module.

snowdd1, 2020-01-31:
While updating an old Raspberry 1b+ to Buster, the same issues show up during `apt full-upgrade`:
```
Setting up owserver (3.2p3+dfsg1-2) ...
Job for owserver.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status owserver.service" and "journalctl -xe" for details.
invoke-rc.d: initscript owserver, action "restart" failed.
● owserver.service - Backend server for 1-wire control
   Loaded: loaded (/lib/systemd/system/owserver.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: protocol) since Fri 2020-01-31 20:57:51 CET; 331ms ago
     Docs: man:owserver(1)
  Process: 2481 ExecStart=/usr/bin/owserver -c /etc/owfs.conf (code=exited, status=0/SUCCESS)
 Main PID: 2481 (code=exited, status=0/SUCCESS)

Jan 31 20:57:51 knxd-raspberry systemd[1]: owserver.service: Service RestartSec=100ms expired, scheduling restart.
Jan 31 20:57:51 knxd-raspberry systemd[1]: owserver.service: Scheduled restart job, restart counter is at 1.
Jan 31 20:57:51 knxd-raspberry systemd[1]: Stopped Backend server for 1-wire control.
dpkg: error processing package owserver (--configure):
 installed owserver package post-installation script subprocess returned error exit status 1
```

I'll have to specify the `--forground` in /lib/systemd/system/owserver.service as well - even though that installation was not using the 1wire kernel module!

Seems to be specific to Buster (or the ow* version). Upgrade to Jessie was survived.

ianka, 2020-01-31:
If I understand this correctly, it's a Raspbian bug. You should note this to Raspbian packagers.