owfs/owfs #55: bug in split packet

honechko, 2020-03-09:
diff -ru owfs-3.2p2.orig/module/owlib/src/c/ow_bus_data.c owfs-3.2p2/module/owlib/src/c/ow_bus_data.c
--- owfs-3.2p2.orig/module/owlib/src/c/ow_bus_data.c    2016-02-04 22:09:53.000000000 +0200
+++ owfs-3.2p2/module/owlib/src/c/ow_bus_data.c 2020-03-09 15:49:31.073080159 +0200
@@ -104,6 +104,7 @@
                if (remain > 0) {
                        RETURN_BAD_IF_BAD( BUS_sendback_data_bitbang(data, resp, max_split_bytes, pn) );
                        RETURN_BAD_IF_BAD( BUS_sendback_data_bitbang(&data[max_split_bytes], resp ? (&resp[max_split_bytes]) : NULL, remain, pn) );
+                       return gbGOOD;
                }
        }

miccoli, 2020-03-09:
Thank you for sending your feedback. Could you please elaborate a little more on the bug and your proposed solution?

honechko, 2020-03-10:
Test program:
void do_packet(char *buf, int len) {
 if (len > 2) {
  do_packet(buf, 2);
  do_packet(buf + 2, len - 2);
 }
 printf("%.*s\n", len, buf);
}
do_packet("12345", 5);
prints:
12
34
5
345
12345
Patched test program:
void do_packet(char *buf, int len) {
 if (len > 2) {
  do_packet(buf, 2);
  do_packet(buf + 2, len - 2);
  return;
 }
 printf("%.*s\n", len, buf);
}
do_packet("12345", 5);
prints:
12
34
5
as expected:)

honechko, 2020-03-11:
https://github.com/owfs/owfs/issues/40 possibly caused by this bug. I encountered the same problem. When owserver reads/writes packet with more than 20 bytes, it sends "extra" timeslots/data which lead to crc error.

honechko, 2020-07-02:
Try write 17 bytes:
`owwrite --hex /uncached/1D.0110A5D4E800/pages/page.7 0102030405060708091011121314151617`
owserver debug:
`DEBUG MODE
libow version:
        3.2p3
  DEBUG: ow_daemon.c:(170) main thread id = 3072478656
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
   CALL: ow_parsename.c:(174) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
  DEBUG: ow_tcp_read.c:(63) attempt 1 bytes Time: 5.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 1 - 0 = 1
  DEBUG: ow_net_server.c:(76) ServerAddr: [localhost] [4304]
  DEBUG: ow_tcp_read.c:(63) attempt 24 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 24 - 0 = 24
  DEBUG: from_client.c:(65) FromClient payload=56 size=17 type=3 sg=0x10A offset=0
  DEBUG: from_client.c:(73) FromClient (no servermessage) payload=56 size=17 type=3 controlflags=0x10A offset=0
  DEBUG: ow_tcp_read.c:(63) attempt 56 bytes Time: 10.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 56 - 0 = 56
  DEBUG: handler.c:(152) START handler /uncached/1D.0110A5D4E800/pages/page.7
   CALL: data.c:(103) DataHandler: parse path=/uncached/1D.0110A5D4E800/pages/page.7
  DEBUG: ow_parseobject.c:(163) /uncached/1D.0110A5D4E800/pages/page.7
   CALL: ow_parsename.c:(174) path=[/uncached/1D.0110A5D4E800/pages/page.7]
  DEBUG: ow_regex.c:(24) Reg Ex expression <^bus\.([[:digit:]]+)/?> compiled to 0xb76dd800
  DEBUG: ow_regex.c:(24) Reg Ex expression <^settings/?> compiled to 0xb76dd7e0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^statistics/?> compiled to 0xb76dd7c0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^structure/?> compiled to 0xb76dd7a0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^system/?> compiled to 0xb76dd780
  DEBUG: ow_regex.c:(24) Reg Ex expression <^interface/?> compiled to 0xb76dd760
  DEBUG: ow_regex.c:(24) Reg Ex expression <^text/?> compiled to 0xb76dd740
  DEBUG: ow_regex.c:(24) Reg Ex expression <^json/?> compiled to 0xb76dd720
  DEBUG: ow_regex.c:(24) Reg Ex expression <^uncached/?> compiled to 0xb76dd700
  DEBUG: ow_regex.c:(24) Reg Ex expression <^unaliased/?> compiled to 0xb76dd6e0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^alarm?> compiled to 0xb76dd6c0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^simultaneous/?> compiled to 0xb76dd6a0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^thermostat/?> compiled to 0xb76dd680
  DEBUG: ow_regex.c:(24) Reg Ex expression <^/bus\.[[:digit:]]+/?> compiled to 0xb76dd660
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.> compiled to 0xb76dd640
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.all$> compiled to 0xb76dd620
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.byte$> compiled to 0xb76dd600
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.[[:digit:]]+$> compiled to 0xb76dd5e0
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.[[:alpha:]]$> compiled to 0xb76dd5c0
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(24) Reg Ex expression <^([[:xdigit:]]{2})\.?([[:xdigit:]]{12})\.?([[:xdigit:]]{2}){0,1}$> compiled to 0xb76dd840
  DEBUG: ow_regex.c:(100) 0: 0->15 found <><1D.0110A5D4E800><>
  DEBUG: ow_regex.c:(100) 1: 0->2 found <><1D><.0110A5D4E800>
  DEBUG: ow_regex.c:(100) 2: 3->15 found <1D.><0110A5D4E800><>
  DEBUG: ow_cache.c:(912) Looking for device 1D 01 10 A5 D4 E8 00 24
  DEBUG: ow_cache.c:(1070) Search in cache sn 1D 01 10 A5 D4 E8 00 24 pointer=0xb76dda44 index=0 size=4
  DEBUG: ow_cache.c:(1106) Value not found in cache
 DETAIL: ow_presence.c:(80) Checking presence of /uncached/1D.0110A5D4E800/pages/page.7
  DEBUG: ow_select.c:(69) Selecting a path (and device) path=/uncached/1D.0110A5D4E800/pages/page.7 SN=1D 01 10 A5 D4 E8 00 24 last path=00 00 00 00 00 00 00 00
  DEBUG: ow_select.c:(81) Clearing root branch
  DEBUG: ow_tcp_read.c:(63) attempt 1 bytes Time: 5.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 1 - 0 = 1
  DEBUG: ow_bus_data.c:(119) Splitting byte 0 of 2 = CC
  DEBUG: ow_bus_data.c:(119) Splitting byte 1 of 2 = 66
  DEBUG: ow_tcp_read.c:(63) attempt 16 bytes Time: 5.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 16 - 0 = 16
  DEBUG: ow_bus_data.c:(140) Consolidating byte 0 of 2 = CC
  DEBUG: ow_bus_data.c:(140) Consolidating byte 1 of 2 = 66
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_bus_data.c:(119) Splitting byte 0 of 1 = FF
  DEBUG: ow_tcp_read.c:(63) attempt 8 bytes Time: 5.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 8 - 0 = 8
  DEBUG: ow_bus_data.c:(140) Consolidating byte 0 of 1 = FF
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_select.c:(247) No DS2409 microlan hub found at this level
[... 1617 lines cut ...]
  DEBUG: ow_bus_data.c:(120) Splitting byte 0 of 4 = 5A
  DEBUG: ow_bus_data.c:(120) Splitting byte 1 of 4 = E0
  DEBUG: ow_bus_data.c:(120) Splitting byte 2 of 4 = 00
  DEBUG: ow_bus_data.c:(120) Splitting byte 3 of 4 = 11
  DEBUG: ow_tcp_read.c:(63) attempt 24 bytes Time: 5.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 24 - 0 = 24
  DEBUG: ow_tcp_read.c:(63) attempt 8 bytes Time: 5.000000 seconds
  DEBUG: ow_tcp_read.c:(113) read: 8 - 0 = 8
  DEBUG: ow_bus_data.c:(141) Consolidating byte 0 of 4 = 5A
  DEBUG: ow_bus_data.c:(141) Consolidating byte 1 of 4 = E0
  DEBUG: ow_bus_data.c:(141) Consolidating byte 2 of 4 = 00
  DEBUG: ow_bus_data.c:(141) Consolidating byte 3 of 4 = 11
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(207) end = 0
[... cut ...]
  DEBUG: ow_write.c:(110) Successful write to /uncached/1D.0110A5D4E800/pages/page.7
  DEBUG: ow_parsename.c:(133) /uncached/1D.0110A5D4E800/pages/page.7
  DEBUG: data.c:(193) DataHandler: FS_ParsedName_destroy done
  DEBUG: data.c:(207) DataHandler: cm.ret=0
  DEBUG: to_client.c:(74) payload=0 size=18, ret=0, sg=0x10A offset=0 
  DEBUG: to_client.c:(84) No data
  DEBUG: data.c:(226) Finished with client request
  DEBUG: handler.c:(134) OWSERVER handler done
  DEBUG: ow_net_server.c:(259) Normal completion.`
No errors (Successful write).

honechko, 2020-08-07:
Can I expect this patch to be included in next release? My project depends on it.

miccoli, 2020-08-07:
This project lacks a leader right now, so the release cycle is quite irregular.

@honechko to have your code included in the codebase, please issue a pull request, with a title mentioning this issue, something like "bug in split packet, fixes #55".

Docs on pull requests can be found [here](https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests).
Very briefly you should 

- fork this repo, 
- create a branch, 
- push your proposed changes in your forked repo, 
- open a pull request. 

The process seems complicated, but it in reality it is quite simple. Please let us know if you need assistance.

In the meantime I hope that we manage to schedule a new release.

honechko, 2020-08-07:
miccoli, thank you for detailed explanation. I have created new pull request https://github.com/owfs/owfs/pull/64

ianka, 2020-08-07:
Am 07.08.20 um 11:45 schrieb Stefano Miccoli:
> This project lacks a leader right now, so the release cycle is quite irregular.
>
> In the meantime I hope that we manage to schedule a new release.
>
I don't see what could keep us back.

Kind regards

	Jan

miccoli, 2020-08-07:
@ianka I agree, a new version could be released soon.

commit e93aec6e (2020-08-16): Merge pull request #64 from honechko/splitpkt

bug in split packet, fixes #55
closed 2020-08-16 by miccoli