owfs/owfs #56: DS18B20 times don't match datasheet

malcb, 2020-03-23:
I have a raspberry pi running owserver, i2c to DS2480 and then 1 wire to DS18B20.  The DS18B20 is 5V powered.  Timing the conversions gives:
`:~# time owread /uncached/28.AD5579A21603/temperature12
     20.3125
real    0m0.700s
user    0m0.002s
sys     0m0.019s
~# time owread /uncached/28.AD5579A21603/temperature9
          20
real    0m0.676s
user    0m0.002s
sys     0m0.018s
:~# time owread /uncached/28.AD5579A21603/latesttemp
          20
real    0m0.085s
user    0m0.011s
sys     0m0.010s
`
As expected latesttemp is very quick since there is no conversion.  But both 9bit and 12bit conversions take the same time, which they shouldn't do according the datasheet.  The 12bit can take 1s which is maximum so maybe 0.7s is reasonable.  However the 9bit should take 100ms so there should be some difference between the timings.

The returned temperatures seem reasonable.  Hence I think the most likely cause is that it is not correctly setting the resolution.  I've set the default resolution to 9 bit and checked that byte 4 in the scratchpad (config byte) has R1=R0=0 which is correct for 9 bit (100ms).  I've then timed a /temperature owread which should use the default, i.e. the programmed 9 bits.  Time is still 0.7s.

Any other suggestions to try?

ianka, 2020-03-23:
The default resolution only affects the power-up value of the sensor. It's overridden as soon you do the first non-simultaneous conversion.

For simultaneous conversions, you have to sleep the required time yourself.

I will setup a DS18B20 and check why the non-simultaneous 9-bit conversion takes 0.7s though.

malcb, 2020-03-26:
I've tracked the down issue to a fake DS18B20.  Maxim Semi (Dallas) say the batch code and date do not match.  I've tried a different part and that shows the correct behaviour of time increasing as resolution is increased.  Maxim say the datasheet figures are of course maximums (as it says) and typical times are more like 75ms, 150ms, 300ms and 600ms for the 9, 10, 11, and 12 bit conversions.  Here are my results with a different part.
~# owwrite /uncached/28.FF004FC21701/tempres 9
~# time owread /uncached/28.FF004FC21701/temperature
          22
real    0m0.224s
user    0m0.002s
sys     0m0.019s
~# owwrite /uncached/28.FF004FC21701/tempres 10
~# time owread /uncached/28.FF004FC21701/temperature
       21.75
real    0m0.418s
user    0m0.008s
sys     0m0.014s
~# owwrite /uncached/28.FF004FC21701/tempres 11
~# time owread /uncached/28.FF004FC21701/temperature
      21.625
real    0m0.568s
user    0m0.000s
sys     0m0.023s
~# owwrite /uncached/28.FF004FC21701/tempres 12
~# time owread /uncached/28.FF004FC21701/temperature
     21.6875
real    0m0.900s
user    0m0.000s
sys     0m0.024s

ianka, 2020-03-26:
Oh great. Thanks for posting this update. May I close this issue?

ianka, 2020-03-26:
Could it be the other chip said DS18S20, not DS18B20?

malcb, 2020-03-27:
Yes, this issue can be closed (which I'll do).  Marking is clearly 18B20 so not an S and I sent the photo of the part to Maxim so it's not just me that thinks it is a B.  Good question though.  Part, TO-18, was from ebay but then so were the other sensors (Stainless Steel encased) that work as they should.  Luck of the draw I guess.

closed 2020-03-27 by malcb