owfs/owfs #57: Problems with DS2482-800 in combination with DS2482-100

bernix50, 2020-04-09:
Hi

I use a DS2482-800 plus a DS2482-100 with a Raspberry Pi, Buster on a I2C bus. OWFS only finds the DS2482-800 on address 1C. **The DS2482-100 on address 18 could not be found.**

OWDIR:
root@BSR-PI1:~# owdir
/bus.7
/bus.6
/bus.5
/bus.4
/bus.3
/bus.2
/bus.1
/bus.0
/uncached
/settings
/system
/statistics
/structure

Abstract of debug protocol:
  DEBUG: ow_parse_address.c:(71) All <ALL>
CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/.
CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/..
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 18
  DEBUG: ow_ds2482.c:(516) ok
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 18 cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 19
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 19 cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1A
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1A cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1B
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1B cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1C
  DEBUG: ow_ds2482.c:(516) ok
CONNECT: ow_ds2482.c:(426) i2c device at /dev/i2c-1 address 1C appears to be DS2482-x00
CONNECT: ow_ds2482.c:(703) DS2482-800 (Eight channels)
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1D
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1D cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1E
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1E cannot be reset. Not a DS2482.
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 1F
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 1F cannot be reset. Not a DS2482.

If I plug out the DS2482-800 from my rapid prototyping port, the DS2482-100 is found by owserver:
CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/.
CONNECT: ow_ds2482.c:(382) [Is a directory] Could not open i2c device /dev/..
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 18
  DEBUG: ow_ds2482.c:(516) ok
CONNECT: ow_ds2482.c:(426) i2c device at /dev/i2c-1 address 18 appears to be DS2482-x00
  DEBUG: ow_ds2482.c:(785) Channel select set error
CONNECT: ow_ds2482.c:(692) DS2482-100 (Single channel)
  DEBUG: ow_ds2482.c:(676) Cannot set to port configuration -- not a DS2483
CONNECT: ow_ds2482.c:(396) Found an i2c device at /dev/i2c-1 address 19
CONNECT: ow_ds2482.c:(422) i2c device at /dev/i2c-1 address 19 cannot be reset. Not a DS2482.

Regards
Bernix

ianka, 2020-04-10:
> i2c device at /dev/i2c-1 address 18 cannot be reset. 
This typically happens if there are power supply problems on the DS2482. Try adding a 100nF bypass capacitor near to the supply pins of the chip.

bernix50, 2020-04-11:
Hi

I tested some capacitors 10µF ... 1nF, but I have still the same error.

ianka, 2020-04-11:
It must be a low-ESL ceramic or polyester cap with short leads.

Also check the power, it must be 3.3V if you hooked the I²C directly to the Raspberry Pi, without level translators. You get a 3.3V Onewire this way — not all Onewire slaves are compatible, check their datasheets. Don't forget you need two 1.5kΩ pullups form SDA and SCL to +3.3V.

bernix50, 2020-04-11:
I use a PCA9517ADP as level translator with 10k pullups acc. datasheet. 1 DS2482-800 Master and 1 DS2482-100 Master. Slaves are only DS18B20 for testing on the bus. 

I will buy a low-ESL capacitor, but that will take some time.

ianka, 2020-04-11:
> I will buy a low-ESL capacitor, but that will take some time.

Almost any 100nF ceramic cap with short leads will do. What about the pullups?

bernix50, 2020-04-12:
It does not work with Standard 100nF Cap an 1.5kOhm. Maybe the Chips are bad quality??

h-wissing, 2020-04-23:
..why do you use a level shifter at all?