owfs/owfs #58: Missing "temphigh" and "templow" for all my DS28EA00 sensors after Raspbian upgrade

kiwi64ajs, 2020-04-19:
Hi Guys,

Previously I was running version 2.8p15 of the various owfs packages on my RPi 2 running on Raspbian 7.11. However as its no longer supported I decided I should upgrade to the latest version of Raspbian.

I installed Raspbian 10.3 onto a new SD Card (and kept the old one intact) and installed the owfs packages which seems to be ok, once I got the services starting on boot - so far so good.

However I've run into some issues with my python scripts that use the "ow" package where I can't access the "temphigh" & "templow" fields of my DS28EA00 sensors that used to work fine for many years.

So I tried the various ow-shell commands like:

```
root@owfs-pi:~# owdir /42.B7D702000000/
/42.B7D702000000/PIO.BYTE
/42.B7D702000000/PIO.ALL
/42.B7D702000000/PIO.A
/42.B7D702000000/PIO.B
/42.B7D702000000/address
/42.B7D702000000/alias
/42.B7D702000000/crc8
/42.B7D702000000/family
/42.B7D702000000/fasttemp
/42.B7D702000000/id
/42.B7D702000000/latch.BYTE
/42.B7D702000000/latch.ALL
/42.B7D702000000/latch.A
/42.B7D702000000/latch.B
/42.B7D702000000/latesttemp
/42.B7D702000000/locator
/42.B7D702000000/power
/42.B7D702000000/r_address
/42.B7D702000000/r_id
/42.B7D702000000/r_locator
/42.B7D702000000/sensed.BYTE
/42.B7D702000000/sensed.ALL
/42.B7D702000000/sensed.A
/42.B7D702000000/sensed.B
/42.B7D702000000/temperature
/42.B7D702000000/temperature10
/42.B7D702000000/temperature11
/42.B7D702000000/temperature12
/42.B7D702000000/temperature9
/42.B7D702000000/tempres
/42.B7D702000000/type
```
Seeing that the temphigh and templow fields were missing in the listing, I figured I'd just try to  read them anyway using owread and they do return reasonable looking values
```
owread /42.B7D702000000/templow
           3

owread /42.B7D702000000/temphigh
          70
```

The temphigh and templow subfields are also missing on the owhttpd server pages for that device

[image]

So it looks like these particular subfields have disappeared from the list of supported or displayed fields for the DS28EA00 chip, but they can still be accessed if you know the path.

Any suggestions?

Regards
Alex Shepherd

ianka, 2020-04-19:
You did not tell us what version of owfs you are running now.

kiwi64ajs, 2020-04-19:
I'm running version 3.2p3 installed from apt packages using
```
apt install owfs
```

ianka, 2020-04-19:
I also wonder which python binding you use. There are several, and all but Stefano Miccoli's pyownet are considered outdated.

kiwi64ajs, 2020-04-19:
Yes, initially I wondered if it was a Python library support issue, I'm not 100% sure which Debian package the Python binding is in, but from within my Python scripts is loaded using:
```
import ow
```
But when I saw the fields were also missing at the owhttpd server and the owdir commandline tools I figured it was a core owfs code base issue and not just a Python binding issue

ianka, 2020-04-19:
As I found, the DS28EA00 field list has the visibility field for those properties set to VISIBLE_DS1825 instead of plain VISIBLE. This looks like a copypasta error to me, as that property is there because the DS1825, MAX31826, MAX31850, and MAX31851 chips all share the same code, but have slightly different features.

Paul wrote that line, but I think we can safely change it to plain old VISIBLE.

ianka, 2020-04-19:
[Fixed in commit 474f06deebdbc5f0221e6035248d3f41810017a2.](https://github.com/owfs/owfs/commit/474f06deebdbc5f0221e6035248d3f41810017a2)

Please either clone the source archive or extract the patch above and apply it to your v3.2p3 sources.

closed 2020-04-19 by ianka

kiwi64ajs, 2020-04-20:
I've downloaded and built this fix and confirmed its works on a Linux Ubuntu VM on my MacBook Pro and am testing building it on the RPi to make sure that works too.

kiwi64ajs, 2020-04-23:
I'm struggling to get this version to build on a RPi 4 running Raspbian 7.11, so I'm wondering if anyone has the list of packages that need to be installed via "apt install" to a fresh system, to allow ./configure and make to build all modules?

I seem to have some missing PHP dependancies
```
Making all in php
make[3]: Entering directory '/root/owfs/module/swig/php'
  CCLD     libowphp.la
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_swig_OW_get_newobject':
ow_wrap.c:(.text+0x3c): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0x68): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_swig_OW_alter_newobject':
ow_wrap.c:(.text+0xb4): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0xcc): undefined reference to `zend_is_true'
/usr/bin/ld: ow_wrap.c:(.text+0xdc): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `zm_startup_OW':
ow_wrap.c:(.text+0x128): undefined reference to `zend_get_constant_str'
/usr/bin/ld: ow_wrap.c:(.text+0x160): undefined reference to `zend_register_long_constant'
/usr/bin/ld: ow_wrap.c:(.text+0x530): undefined reference to `compiler_globals'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_version':
ow_wrap.c:(.text+0x5a4): undefined reference to `_emalloc'
/usr/bin/ld: ow_wrap.c:(.text+0x5f4): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_init':
ow_wrap.c:(.text+0x660): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0x6a4): undefined reference to `_convert_to_string'
/usr/bin/ld: ow_wrap.c:(.text+0x6ac): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_put':
ow_wrap.c:(.text+0x764): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0x7cc): undefined reference to `_convert_to_string'
/usr/bin/ld: ow_wrap.c:(.text+0x7d8): undefined reference to `_convert_to_string'
/usr/bin/ld: ow_wrap.c:(.text+0x7e0): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_get':
ow_wrap.c:(.text+0x87c): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0x8cc): undefined reference to `_emalloc'
/usr/bin/ld: ow_wrap.c:(.text+0x920): undefined reference to `_convert_to_string'
/usr/bin/ld: ow_wrap.c:(.text+0x928): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_finish':
ow_wrap.c:(.text+0x978): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_set_error_print':
ow_wrap.c:(.text+0x9e4): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0xa10): undefined reference to `zval_get_long_func'
/usr/bin/ld: ow_wrap.c:(.text+0xa18): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_get_error_print':
ow_wrap.c:(.text+0xa98): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_set_error_level':
ow_wrap.c:(.text+0xb04): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0xb30): undefined reference to `zval_get_long_func'
/usr/bin/ld: ow_wrap.c:(.text+0xb38): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_get_error_level':
ow_wrap.c:(.text+0xbb8): undefined reference to `zend_wrong_param_count'
/usr/bin/ld: .libs/ow_wrap.o: in function `_wrap_opt':
ow_wrap.c:(.text+0xc1c): undefined reference to `_zend_get_parameters_array_ex'
/usr/bin/ld: ow_wrap.c:(.text+0xc38): undefined reference to `_convert_to_string'
/usr/bin/ld: ow_wrap.c:(.text+0xc84): undefined reference to `_convert_to_string'
/usr/bin/ld: ow_wrap.c:(.text+0xc8c): undefined reference to `zend_wrong_param_count'
collect2: error: ld returned 1 exit status
make[3]: *** [Makefile:581: libowphp.la] Error 1
make[3]: Leaving directory '/root/owfs/module/swig/php'
make[2]: *** [Makefile:515: all-recursive] Error 1
make[2]: Leaving directory '/root/owfs/module/swig'
make[1]: *** [Makefile:525: all-recursive] Error 1
make[1]: Leaving directory '/root/owfs/module'
make: *** [Makefile:580: all-recursive] Error 1
```

I got it to compile and run on Ubuntu 18.04.04 ok but Raspbian is being a bit stubborn

Regards
Alex Shepherd

ianka, 2020-04-23:
`./configure --disable-owphp` then.