owfs/owfs #59: owfs file duplicates in /mnt/1wire Raspbian Buster all versions

h-wissing, 2020-04-23:
Hello,

I intend to use the abelectronics DS2482-100 interface board with owfs for managing DS18B20 temperature sensors. The following was tested on Pi 2 with Buster light, Pi3+ and Pi4 with Buster full 2020-02-13 ans subsequent upgrades.

I installed the owfs package according the tutorial given by abelectronics. 
https://www.abelectronics.co.uk/kb/article/3/owfs-with-i2c-support-on-raspberry-pi
and chanched owfs.conf accordingly.

system works, but virtual file system shows strange results 

:/mnt/1wire $ ls -l
drwxrwxrwx 1 root root 4096 Apr 23 11:46 28.0BA7052D1901
drwxrwxrwx 1 root root 4096 Apr 23 11:46 28.0BA7052D1901
drwxr-xr-x 1 root root 4096 Apr 13 21:58 alarm
drwxr-xr-x 1 root root 4096 Apr 13 21:58 bus.0
drwxr-xr-x 1 root root 4096 Apr 13 21:58 bus.1
drwxr-xr-x 1 root root 4096 Apr 13 21:58 settings
drwxrwxrwx 1 root root 4096 Apr 23 11:46 simultaneous
drwxr-xr-x 1 root root 4096 Apr 13 21:58 statistics
drwxr-xr-x 1 root root 4096 Apr 13 21:58 structure
drwxr-xr-x 1 root root 4096 Apr 13 21:58 system
drwxr-xr-x 1 root root 4096 Apr 13 21:58 uncached

the attached sensor is exposes in duplicate, also a 2nd bus, bus.1, is generated with the same content as bus.0. This is also true for the web interface.

With the help of abelectronics support
https://www.abelectronics.co.uk/forums/thread/303/owfs-duplicates
part of the problem was solved
by disabeling 1-wire support on the RaspberryPi via raspi-config and adding 
ds2482 in /etc/modules

after reboot, there were still duplicates which can be removed by the command

sudo owfs --w1

made permenatly by

by adding "owfs --w1" to /etc/rc.local.

This leads to the result, that duplicates disappear, but /bus.1 is still generated, without relevant content. Same happens in the web interface.

Surpisingly owdir never reports duplicates.

 owdir
/28.0BA7052D1901
/bus.0
/uncached
/settings
/system
/statistics
/structure
/simultaneous
/alarm

How can this different behavior of the package components be explained.

Is there a phyton3 package for owfs available.

Regards
h-wissing

ianka, 2020-04-23:
> How can this different behavior of the package components be explained.

So you start owfs --w1 in parallel to a running owserver? Don't do that, it creates problems like those you encountered. Use owserver with the --w1 option and advise all other programs to use owserver with the -s option (for owdir/owget/owread/owwrite `-s localhost` is the default).

And don't use the owfs tool. It has unfixable races when multiple programs access the same property.

> Is there a phyton3 package for owfs available.

Stefano Miccoli's _pyownet_

h-wissing, 2020-04-23:
I do not intetionally start any program. I just run installation and do a reboot.. that also happenes at the service of abelectronics, they also had no idea where to look

ianka, 2020-04-23:
It's irrelevant if you did start something intentionally or not. You have to check which owfs related processes are running and what options they use. There's a tool called `ps` that gives you an overview of running processes, and there's a tool `grep` that lets you filter lines by content
`$ ps ax|grep ow`

h-wissing, 2020-04-23:
tonight I made a system update / upgrade

with the duplicates 
drwxrwxrwx 1 root root 4096 Apr 24 00:19 28.0BA7052D1901
drwxrwxrwx 1 root root 4096 Apr 24 00:19 28.0BA7052D1901
drwxr-xr-x 1 root root 4096 Apr 24 00:16 alarm
drwxr-xr-x 1 root root 4096 Apr 24 00:16 bus.0
drwxr-xr-x 1 root root 4096 Apr 24 00:16 bus.1
drwxr-xr-x 1 root root 4096 Apr 24 00:16 settings
drwxrwxrwx 1 root root 4096 Apr 24 00:19 simultaneous
drwxr-xr-x 1 root root 4096 Apr 24 00:16 statistics
drwxr-xr-x 1 root root 4096 Apr 24 00:16 structure
drwxr-xr-x 1 root root 4096 Apr 24 00:16 system
drwxr-xr-x 1 root root 4096 Apr 24 00:16 uncached

result is:

ps ax|grep ow
  257 ?        Ssl    0:00 /usr/bin/owftpd --foreground -c /etc/owfs.conf
  272 ?        Ssl    0:00 /usr/bin/owserver -c /etc/owfs.conf
  323 ?        Ssl    0:00 /usr/lib/owfs/owfs -c /etc/owfs.conf -c /etc/owfs.conf --allow_other /run/owfs
  348 ?        Ssl    0:00 /usr/bin/owhttpd --foreground -c /etc/owfs.conf
  502 pts/0    S+     0:00 grep --color=auto ow

after running  sudo owfs --w1

tonight the first time this error message occurred: 

fuse: mountpoint is not empty
fuse: if you are sure this is safe, use the 'nonempty' mount option

no further action

according to abelectronics advice its made permanent as before by adding "owfs --w1" to /etc/rc.local before the last line "exit 0"

after reboot result was

/mnt/1wire $ ls -l
insgesamt 0  
drwxrwxrwx 1 root root 4096 Apr 24 00:28 28.0BA7052D1901
drwxr-xr-x 1 root root 4096 Apr 24 00:26 bus.0
drwxr-xr-x 1 root root 4096 Apr 24 00:26 bus.1
drwxr-xr-x 1 root root 4096 Apr 24 00:26 bus.2
drwxr-xr-x 1 root root 4096 Apr 24 00:26 settings
drwxr-xr-x 1 root root 4096 Apr 24 00:26 statistics
drwxr-xr-x 1 root root 4096 Apr 24 00:26 structure
drwxr-xr-x 1 root root 4096 Apr 24 00:26 system
drwxr-xr-x 1 root root 4096 Apr 24 00:26 uncached

ls -l took a long time, there are no dublicates but now there are three busses
with
/mnt/1wire/bus.0 $ ls -l
insgesamt 0
drwxrwxrwx 1 root root 4096 Apr 24 00:40 28.0BA7052D1901
drwxr-xr-x 1 root root 4096 Apr 24 00:26 alarm
drwxr-xr-x 1 root root 4096 Apr 24 00:26 bus.0
drwxr-xr-x 1 root root 4096 Apr 24 00:26 interface
drwxr-xr-x 1 root root 4096 Apr 24 00:26 settings
drwxrwxrwx 1 root root 4096 Apr 24 00:40 simultaneous
drwxr-xr-x 1 root root 4096 Apr 24 00:26 statistics
drwxr-xr-x 1 root root 4096 Apr 24 00:26 structure
drwxr-xr-x 1 root root 4096 Apr 24 00:26 system
drwxr-xr-x 1 root root 4096 Apr 24 00:26 uncached

and 

/mnt/1wire/bus.1 $ ls -l
insgesamt 0
drwxr-xr-x 1 root root 4096 Apr 24 00:26 interface

/mnt/1wire/bus.2 $ ls -l
insgesamt 0
drwxr-xr-x 1 root root 4096 Apr 24 00:26 interface

and the running ow processes are:

ps ax|grep ow
   31 ?        I      0:00 [kworker/0:1-events_power_efficient]
  122 ?        I      0:00 [kworker/1:2-events_freezable_power_]
  127 ?        I      0:00 [kworker/0:2-events_power_efficient]
  266 ?        Ssl    0:00 /usr/bin/owserver -c /etc/owfs.conf
  268 ?        Ssl    0:00 /usr/bin/owftpd --foreground -c /etc/owfs.conf
  348 ?        Ssl    0:00 /usr/bin/owhttpd --foreground -c /etc/owfs.conf
  413 ?        Ssl    0:00 /usr/lib/owfs/owfs -c /etc/owfs.conf --w1
  495 pts/0    S+     0:00 grep --color=auto ow

now I'm completely lost

ianka, 2020-04-23:
> 323 ? Ssl 0:00 `/usr/lib/owfs/owfs -c /etc/owfs.conf `

This line means owfs is already running with the configuration taken form /etc/owfs.conf. The other programs, too. Please post the contents of that file.

> after running sudo owfs --w1

Don't do that.

> according to abelectronics advice its made permanent as before by adding "owfs --w1" to /etc/rc.local before the last line "exit 0"

This is wrong advice. Don't do that.

h-wissing, 2020-04-23:
here we go

# Sample configuration file for the OWFS suite for Debian GNU/Linux.
#
#
# This is the main OWFS configuration file. You should read the
# owfs.conf(5) manual page in order to understand the options listed
# here.

######################## SOURCES ########################
#
# With this setup, any client (but owserver) uses owserver on the
# local machine...
! server: server = localhost:4304
#
# ...and owserver uses the real hardware, by default fake devices
# This part must be changed on real installation
#server: FAKE = DS18S20,DS2405
#
# USB device: DS9490
#server: usb = all
#
server: device = /dev/i2c-1

# Serial port: DS9097
#server: device = /dev/ttyS1
#
# owserver tcp address
#server: server = 192.168.10.1:3131
#
# random simulated device
#server: FAKE = DS18S20,DS2405
#
######################### OWFS ##########################
#
mountpoint = /mnt/1wire
allow_other
#
####################### OWHTTPD #########################

http: port = 2121

####################### OWFTPD ##########################

ftp: port = 2120

####################### OWSERVER ########################

server: port = localhost:4304

ianka, 2020-04-23:
Putting that in _code quotes_ had been nice.

`server: device = /dev/i2c-1`

Well, and here we have the culprit. You advice owserver to access the DS2482-100 directly by userland I²C programming, but there's also the w1 ds2482 kernel module, which does the same thing in parallel. This leads to a mishmash of I²C commands to the chip and in result to all kinds of funny errors.

Remove or comment out that line and add a line

`server: w1`

instead. Remove that wrong line from /etc/rc.local, too, and reboot.

h-wissing, 2020-04-23:
thanks

somewhere I've read, that with Buster w1 kernel module should have a problem with I2C

ianka, 2020-04-23:
I'm pretty sure that _problem with I2C_ is what people have that ask the hardware supplier and get wrong advice from them.

h-wissing, 2020-04-23:
now with owdir I see two busses and no sensor

owdir
/bus.1
/bus.0
/uncached
/settings
/system
/statistics
/structure

also no sensor in the file system 

/mnt/1wire $ ls -l
insgesamt 0
drwxr-xr-x 1 root root 4096 Apr 24 01:49 bus.0
drwxr-xr-x 1 root root 4096 Apr 24 01:49 bus.1
drwxr-xr-x 1 root root 4096 Apr 24 01:49 settings
drwxr-xr-x 1 root root 4096 Apr 24 01:49 statistics
drwxr-xr-x 1 root root 4096 Apr 24 01:49 structure
drwxr-xr-x 1 root root 4096 Apr 24 01:49 system
drwxr-xr-x 1 root root 4096 Apr 24 01:49 uncached

ps ax|grep ow
    5 ?        I      0:00 [kworker/0:0-events_power_efficient]
  127 ?        I      0:00 [kworker/0:2-events_power_efficient]
  254 ?        Ssl    0:00 /usr/bin/owserver -c /etc/owfs.conf
  272 ?        Ssl    0:00 /usr/bin/owftpd --foreground -c /etc/owfs.conf
  324 ?        Ssl    0:00 /usr/lib/owfs/owfs -c /etc/owfs.conf -c /etc/owfs.conf --allow_other /run/owfs
  346 ?        Ssl    0:00 /usr/bin/owhttpd --foreground -c /etc/owfs.conf
  480 pts/0    S+     0:00 grep --color=auto ow

h-wissing, 2020-04-23:
..probably i2c driver has to be loaded manually

ianka, 2020-04-24:
What does `lsmod|grep ds2482` say?

h-wissing, 2020-04-24:
no reply

ianka, 2020-04-24:
Then the module isn't loaded. Try `modprobe ds2482` as root and check if you see devices a few seconds later.

To make that happen automatically, you have to use a dtb overlay on the Raspberry. There are some people who made such a thing but __I__ would have to play with that a day or so to make it work. I recommend to put that modprobe line into /etc/rc.local instead.

h-wissing, 2020-04-24:
now its there

lsmod|grep ds2482
ds2482                 16384  0
wire                   45056  2 ds2482,w1_gpio

shall I add this to /etc/rc.local and reboot?

ianka, 2020-04-24:
Yes, add `modprobe ds2482` to /etc/rc.local. That should work.

h-wissing, 2020-04-24:
still two busses, no sensor

owdir
/bus.1
/bus.0
/uncached
/settings
/system
/statistics
/structure

same in virtual filesystem

driver loaded

lsmod|grep ds2482
ds2482                 16384  0
wire                   45056  2 ds2482,w1_gpio

h-wissing, 2020-04-24:
there seems to be a bug in the Kernel driver

ianka, 2020-04-24:
Hmm. I can explain the two buses though. One is the bitbang hostadapter on GPIO4, which is controlled through the w1_gpio module. To disable that one, you have to remove the `dtoverlay=w1-gpio` line from your /boot/config.txt (or wherever the boot partition is mounted). Maybe that one is also responsible for the missing chip on the DS2482.

Also, please always use `$ owdir /uncached` to trigger a new scan of the bus. Otherwise you have to wait a minute or so for the continous scan the w1 subsystem does automatically.

h-wissing, 2020-04-24:
I'll try that tomorrow. 
I've soldered  on chip carrier PCBs 2 additional DS2482-100 and 1 DS2482-800 and set them up on a breadboard. With the initially working system - according to advice - I find all the the sensors, but the busses merely as subfolders to bus 0. owdir was OK. I intend to open a separate issue for that, when this simple setting is running. 
Gute Nacht in den Harz

h-wissing, 2020-04-24:
hello
I commented out dtoverlay=w1-gpio  in /boot/config.txt.

 $ owdir
/bus.0
/uncached
/settings
/system
/statistics
/structure

..still no sensor detected.

Am I right, to get a working system, I have to go back to the hardware manufacturers advice?
Does it make sense to open a issue for Raspbian, not loading I2C devices as 1-wire bus masters?

ianka, 2020-04-24:
> Am I right, to get a working system, I have to go back to the hardware manufacturers advice?

No, because that one included a mishmash of w1 and userland I²C. Please follow __my__ and only __my__ advice. Otherwise, I cannot help you.

It seems the ds2482 kernel module is indeed broken in your distribution. So you can't use the w1 subsystem in combination with a DS2482 chip. You have to stick to userland I²C. Do the following:

- remove `ds2482` in `/etc/modules`
- create/edit `/etc/modprobe.d/blacklist.conf`and add a line `blacklist ds2482` (sometimes the common blacklist file has a slightly different name)
- remove the `modprobe ds2482` line from `/etc/rc.local`
- remove the `server: w1` line from `/etc/owfs.conf`
- re-add the `server: device=/dev/i2c-1` line in `/etc/owfs.conf`

Reboot. This should give you a single bus for the owserver, and one bus per connected DS2482 within the owserver bus. Check with `$ owdir /uncached` and `$ owdir /uncached/bus.0`

robertocannella, 2022-05-10:
I'm having similar issues.  My setup is a bit different.  I know it's not Buster, but I think adding this may be helpful to others.

```
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
```

 I have the same abelectronics DS2482-100 installed on my RP4.  I also have an additional ds2482-800 attached at another address i2c(configured through jumpers A0,A1,A1).    I'm not sure how I'm getting duplicates, but after some poking around, I decided to load the driver manually through etc/rc.local on boot (without relying on owfs.conf).

```
# OWFS Subsystem

modprobe ds2482
sh -c "echo ds2482 0x18 > /sys/bus/i2c/devices/i2c-1/new_device "   <-- abelectronics 1-wire Plus (DS2482-100)
sh -c "echo ds2482 0x1a > /sys/bus/i2c/devices/i2c-1/new_device "   <-- soldered DS2482-800
modprobe wi
``` 
This seemed to have resolved the duplicate issue.  Here is the output from owdir:
```
cbi@raspberrypi:~ $ owdir
/28.91C0FF030000
/28.79B1FF030000
/28.7FA5FF030000
/bus.9
/bus.8
/bus.7
/bus.6
/bus.5
/bus.4
/bus.3
/bus.2
/bus.1
/bus.0
/uncached
/settings
/system
/statistics
/structure
```

One strange occurrence is that the 1-wire plus hat is actually a /bus.1, not at /bus.0.   This is where I ultimately found my solution: 

[link](https://www.faschingbauer.me/trainings/material/soup/linux/hardware/w1/topic.html#wiring-attach-onewire-device-to-ds2482-800)