owfs/owfs #60: Does owserver support a kernel with w1-gpio, but not w1-therm?

w4tsn, 2020-04-28:
I'm using a Raspberry Pi 3B+ with Fedora IoT, which uses Linux Kernel 5.5.x. This system only ships the w1-gpio module, but not w1_therm. I've connected a D18 Temperature Sensor on GPIO Pin 4 and I'm seeing the sensor under it's sysfs path. On Raspbian with w1_therm however the sysfs contains a w1_slave file which outputs the sensor readings. On Fedora IoT however there is only the generic `/sys/bus/w1/devices/<sensorid>/rw`. Is owfs able to get this device in theory or is the missing w1_therm module the problem here?

Also, on top of that, the owserver runs in a docker container with `/sys` mounted to use the `--w1` functionality. On top of that the server is started with `--usb=ALL and --device=<path-to-serial-device` to leverage multiple connected bus masters.

owserver is able to read devices from the serial bus master, but not from the one connected with gpio4.

ianka, 2020-04-28:
It's a bit complicated, because there are many many options how you may setup and use your onewire devices with Linux. The kernel w1 subsystem splits into three parts: host adapter drivers, slave device drivers, and a userspace connector. The OWFS suite consists of host adapter drivers, slave device drivers, and a kernelspace connector.

You don't need the w1 kernel modules at all to run the OWFS suite. The OWFS suite also does not make use of the slave device drivers the kernel offers, as w1_therm.

> owserver is able to read devices from the serial bus master, but not from the one connected with gpio4.

That is because it was configured to control the serial masters, which the kernel do not support at all, from userspace. But it was not configured to use the kernel's w1 subsystem for the GPIO master, which the OWFS suite cannot control from userspace.

> The server is started with --usb=ALL and --device=<path-to-serial-device>

And that misses --w1.

Make sure to blacklist the ds2490 module if you use both --usb and --w1 at the same time. Otherwise the kernel and the OWFS suite try to access the host adapter concurrently which leads to all kinds of funny bugs.

w4tsn, 2020-04-29:
Thanks for the explanation and the tip with the blacklisting, that cleared up a lot of things.

My Post was a little ambiguous, sorry. The server is indeed started with `--w1`, that's why I'm so confused why the sensor on gpio4 is not visible through owserver while I can see it on the host machines sysfs path.

So to wrap it up, for `--usb` to work in owserver I have to have access to all the respective usb / usb-bus devices, right? For serial port based chips with `--device` I need access to `/dev/serial` and for `--w1` I have to enable the gpio overlay on gpio4 on the host system end owserver needs access to `/sys/bus/w1` and `/sys/devices`. That's how far I'm now. Did I forget something or get it wrong? E.g. I wasn't aware about the conflicts with the host OS' modules.

After reading the kernel docs it appeals to me that it could actually be sufficient to only use `--w1` and let the kernel handle gpio, usb and serial bus master alltogether. Is that correct? In my scenario there may be all three present at the same device, therefore I want to have the general interface of owserver for all of them.

ianka, 2020-04-29:
> The server is indeed started with --w1, that's why I'm so confused why the sensor on gpio4 is not visible through owserver 

The w1 userspace connector protocol has been changed at least one time. Which OWFS version do you use? Does your kernel even feature the userspace connector used by the w1 subsystem? Please post the output of

```
$ owserver --version
$ zgrep -i connector /proc/config.gz
$ zgrep -i w1 /proc/config.gz
```

> So to wrap it up, for --usb to work in owserver I have to have access to all the respective usb / usb-bus devices, right? 

Access to /sys/bus/usb is needed. You may use an udev rule as
```
SUBSYSTEMS=="usb", ATTRS{idVendor}=="04fa", ATTRS{idProduct}=="2490", NAME="bus/usb/$env{BUSNUM}/$env{DEVNUM}", GROUP="users", MODE="0660"
```
to limit user access to DS9490 devices specifically.

> For serial port based chips with --device I need access to /dev/serial

to whatever device node you use in the owserver call.

> and for --w1 […] owserver needs access to /sys/bus/w1 and /sys/devices.

Wrong. Owserver accesses the kernel's w1 subsystem through the _connector_ driver, not through sysfs.

> After reading the kernel docs it appeals to me that it could actually be sufficient to only use --w1 and let the kernel handle gpio, usb and serial bus master alltogether.

The kernel w1 subsystem features no host adapter drivers for serial host adapters.

w4tsn, 2020-04-29:
Output of the above commands:

```bash
$ owserver --version
owserver version:
	3.2p3
libow version:
	3.2p3
$ zgrep -i connector /proc/config.gz
gzip: /proc/config.gz: No such file or directory
$ zgrep -i w1 /proc/config.gz
gzip: /proc/config.gz: No such file or directory
```

The output is the same on the host as well as within the container. Apparently Fedora IoT does not have `/proc/config.gz` enabled. I'll search for the kernel configs elsewhere.

What do I need to have access to on the host in order to use the kernels connector driver for the `--w1` functionality? I'm wondering if the containerization of owserver keeps it from accessing the necessary bits and peaces of the host system. I've started the container in privileged mode to test if it's a permission issue to no avail.

ianka, 2020-04-29:
What do I need to have access to on the host in order to use the kernels connector driver for the --w1 functionality?

CONFIG_CONNECTOR=y/m
CONFIG_W1=y/m
CONFIG_W1_CON=y
CONFIG_W1_MASTER_GPIO=y/m

> I'm wondering if the containerization of owserver keeps it from accessing the necessary bits and pieces of the host system

Maybe. I don't have experience with that

w4tsn, 2020-05-15:
To give you an update, I'm still investigating what's preventing owfs to access one-wire devices over GPIO bus master in my setup.

I was not yet able to verify that the kernel configs are correct, but the dtoverlay is indeed loaded and the w1_gpio module is present (wire and w1_gpio are the only loaded kernel mods).

The difference to raspbian is, that w1_therm is not loaded, but this shouldn't be a problem, because owfs is not using the sysfs interface as you already mentioned.

Right now I suspect the container not being able to access the w1-gpio devices provided by the kernel.

I'd keep this issue open until the situation is resolved so that others with a similar situation may wind up here ;)

ianka, 2020-05-15:
The connector functionality depends on netlink sockets. Maybe checking other netlink-based tools helps you checking if that's the problem.

agates, 2024-02-04:
I was able to get this working running with `--cap-add NET_ADMIN` under podman.  Not ideal, of course -- would be interested in a workaround.

ianka, 2024-02-06:
You have to help me here: What devices aren't showing? The onewire slaves? This may be due to not the correct W1 host adapter driver in the kernel not loaded.

agates, 2024-02-06:
> You have to help me here: What devices aren't showing? The onewire slaves? This may be due to the correct W1 host adapter driver in the kernel is not loaded.

Yes, the onewire slaves.  They show up on the host but not in the container, so the driver is loaded correctly.

I'm using https://github.com/pstolarz/w1-gpio-cl, for what it's worth.  Not sure if that matters.

ianka, 2024-02-08:
I see. No idea about the particular problem with the w1 driver, sorry.

As a workaround, I recommend to run an `owserver --w1` in the host and let the client programs inside the container connect to that host owserver via TCP with the `-s <ip>[ :port]` option.