owfs/owfs #61: simultaneous feature in ds2409 branches doesn't seem to be implemented

sophana, 2020-05-02:
Hi

I've tried the simultaneous temperature read feature with ds2409 based switches.

I've tried to write 1 in each branch (1F.xxxxxxxx/aux/simultaneous/temperature) (tried both cached uncached)
then read the temperatures.
If I read temperature cached, it goes fast, but the temperature is always the same (no conversion is done)
if I read temperature uncached, it goes slow, but correct temperature is read.

It seems that writing 1 to simultaneous/temperature has no effect in ds2409 branches.

Looking at ow_2409.c there is no mention to eimultaneous feature. I'd like to help resolving the issue, but have no idea how (and where) simultaneous feature is implemented.

Thanks

ianka, 2020-05-02:
You should trigger `/simultaneous/temperature` but then read `/uncached/<id>/latesttemp`. [This is documented in the manpage.](https://github.com/owfs/owfs-doc/wiki/DS18B20)

Using `/uncached` makes sure you access the chip and not the cache, and using `latesttemp` makes sure you are only reading the result of the latest conversion, and not triggering a new one.

sophana, 2020-05-02:
Hi I debugged some more and modified the first post.
It seems there is a bug in branch selection

ianka, 2020-05-02:
I'm not the author of that code piece, but the last one who worked on it due to the implementation of `latesttemp`. Please try `latesttemp` first. The decision mechanism whether reading `temperature` should trigger a new conversion or not is rather complicated.

sophana, 2020-05-02:
what is latesttemp? a branch?

ianka, 2020-05-02:
`latesttemp` is a property in the individual slave device directory, as `temperature`. Please read the manpage of the chip I linked above.

sophana, 2020-05-02:
ok I will try, but I'm almost sure that the convert command is not sent.
Thanks

sophana, 2020-05-10:
I've compiled last commit, and it still doesn't work.
owfs looks incredibly complex to debug.
The problem seems that the conversion command is sent to the wrong branch (1 level too high).
```  
DEBUG: ow_write.c:(437) Write a non-array element /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/simultaneous/temperature
  DEBUG: ow_simultaneous.c:(156) TEST if bus powered
  DEBUG: ow_select.c:(70) Selecting a path (and device) path=/uncached/1F.C4DC03000000/main/1F.44D003000000/aux/simultaneous/temperature SN=1F 44 D0 03 00 00 00 01 last path=1F 6A E2 03 00 00 00 64
  DEBUG: ow_select.c:(95) Clearing all branches to level 1
  DEBUG: ow_ds9490.c:(562) DS9490 RESET. changed 0, flex: 1
  DEBUG: ow_ds9490.c:(613) DS9490_Reset: OK
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_ds9490.c:(562) DS9490 RESET. changed 0, flex: 1
  DEBUG: ow_ds9490.c:(613) DS9490_Reset: OK
  DEBUG: ow_select.c:(202) Selecting subbranch 1F C4 DC 03 00 00 00 0D
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(97) compare = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_ds9490.c:(562) DS9490 RESET. changed 0, flex: 1
  DEBUG: ow_ds9490.c:(613) DS9490_Reset: OK
  DEBUG: ow_select.c:(202) Selecting subbranch 1F C4 DC 03 00 00 00 0D
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(97) compare = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_transaction.c:(90) select = 0
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_cache.c:(510) Adding for conversion time for 1F 44 D0 03 00 00 00 01
  DEBUG: ow_cache.c:(516) 1F 44 D0 03 00 00 00 01
  DEBUG: ow_cache.c:(520) Simultaneous add type=S_T
  DEBUG: ow_cache.c:(635) Add to cache sn 1F 44 D0 03 00 00 00 01 pointer=0xb6f77348 index=0 size=0
  DEBUG: ow_select.c:(70) Selecting a path (and device) path=/uncached/1F.C4DC03000000/main/1F.44D003000000/aux/simultaneous/temperature SN=1F 44 D0 03 00 00 00 01 last path=1F C4 DC 03 00 00 00 0D
  DEBUG: ow_select.c:(102) Reselecting branch at level 1
  DEBUG: ow_ds9490.c:(562) DS9490 RESET. changed 0, flex: 1
  DEBUG: ow_ds9490.c:(613) DS9490_Reset: OK
  DEBUG: ow_select.c:(202) Selecting subbranch 1F C4 DC 03 00 00 00 0D
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(97) compare = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_transaction.c:(90) select = 0
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_write.c:(495) Write /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/simultaneous/temperature Extension 0 Gives result 0
  DEBUG: ow_write.c:(110) Successful write to /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/simultaneous/temperature
  DEBUG: ow_parsename.c:(133) /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/simultaneous/temperature
   write[0] 1 bytes to 0
   unique: 278, success, outsize: 24
unique: 280, opcode: RELEASE (18), nodeid: 8, insize: 64, pid: 0
release[0] flags: 0x20001
```
reading latesttemp gives another selection which look beter
```
  DEBUG: ow_read.c:(72) /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/10.C3088A010800/latesttemp
  DEBUG: ow_read.c:(204) /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/10.C3088A010800/latesttemp
  DEBUG: ow_read.c:(238) About to read </uncached/1F.C4DC03000000/main/1F.44D003000000/aux/10.C3088A010800/latesttemp> extension=0 size=4096 offset=0
  DEBUG: ow_read.c:(333) file_length=12 offset=0 size=12
  DEBUG: ow_select.c:(70) Selecting a path (and device) path=/uncached/1F.C4DC03000000/main/1F.44D003000000/aux/10.C3088A010800/latesttemp SN=10 C3 08 8A 01 08 00 1C last path=1F C4 DC 03 00 00 00 0D
  DEBUG: ow_select.c:(95) Clearing all branches to level 2                                                                                                                                                         
  DEBUG: ow_ds9490.c:(562) DS9490 RESET. changed 0, flex: 1
  DEBUG: ow_ds9490.c:(613) DS9490_Reset: OK
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
[... 28 lines cut ...]
  DEBUG: ow_ds9490.c:(613) DS9490_Reset: OK
  DEBUG: ow_select.c:(202) Selecting subbranch 1F C4 DC 03 00 00 00 0D
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(97) compare = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_select.c:(202) Selecting subbranch 1F 44 D0 03 00 00 00 C6
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(97) compare = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_select.c:(226) Selecting device 10 C3 08 8A 01 08 00 1C
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_transaction.c:(90) select = 0
  DEBUG: ow_transaction.c:(112) send = 0
  DEBUG: ow_transaction.c:(130) readin = 0
  DEBUG: ow_transaction.c:(176) CRC8 = 0
  DEBUG: ow_transaction.c:(207) end = 0
  DEBUG: ow_read.c:(620) Read /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/10.C3088A010800/latesttemp Extension 0 Gives result 0
  DEBUG: ow_cache.c:(384) Adding data for /uncached/1F.C4DC03000000/main/1F.44D003000000/aux/10.C3088A010800/latesttemp
  DEBUG: ow_cache.c:(421) 10 C3 08 8A 01 08 00 1C size=8
  DEBUG: ow_cache.c:(635) Add to cache sn 10 C3 08 8A 01 08 00 1C pointer=0xb6f5d2f8 index=0 size=8
  DEBUG: ow_read.c:(253) return=0
```

sophana, 2020-05-10:
Without gdb, this seem very difficult to debug.
Unfortunately gdb does a segfault at the very beginning on my arm platform.
I will have to try to debug this on a pc.

ianka, 2020-05-10:
Thanks for the followup, and yes, you are right, it's incredibly complex.