owfs/owfs #63: Eeprom write error

bahm-lgs, 2020-07-08:
Hi,

I have discovered an issue where writing a certain amount (>=4) of bytes to a DS28EC20P + Eeprom occasionally clears the whole Eeprom page to 0xFF.

To reproduce the issue, I have started only owserver 3.2p3 which accesses the 1wire bus via w1-gpio kernel 4.19.0 driver.

After setting the first page to all zeroes, I continuously write the dword 0x11223344 to page 0, offset 0 and check the contents of the whole page with a read afterwards:

`owwrite --hex --start=0 /43.F0CFFB000000/pages/page.0 0000000000000000000000000000000000000000000000000000000000000000; count=0; while owwrite --e.0 11223344; ret=$?; readback=``owread --hex --start=0 --size=32 /43.F0CFFB000000/pages/page.0`` ; count=$[$count+1]; [ $readback == "112233440000000000000000000; do echo "$count: $readback"; done; echo $readback`

After a few or even a few 1000 cycles the whole page has been reset to 0xFF:

`1: 1122334400000000000000000000000000000000000000000000000000000000
2: 1122334400000000000000000000000000000000000000000000000000000000
3: 1122334400000000000000000000000000000000000000000000000000000000
4: 1122334400000000000000000000000000000000000000000000000000000000
5: 1122334400000000000000000000000000000000000000000000000000000000
6: 1122334400000000000000000000000000000000000000000000000000000000
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF`

Now the funny fact, when starting owserver with `--error_print=2 --error_level=9` the issue does not occur anymore.

Any hints  how I can further debug this?

Regards
Martin

PS: yes this happens on different systems with the same setup, so its not a faulty part.

miccoli, 2020-07-09:
> Now the funny fact, when starting owserver with `--error_print=2 --error_level=9` the issue does not occur anymore.

Maybe a timing issue? I often noticed that when you enable the very verbose debug output owserver is *much much* slower.

If you inset a `sleep 1` in your tight loop above, does the  problem persist?

(I'm not able to properly understand your one-liner shell example: is somewhere  missing a `]`? If you rewrite your example as a code block inside [triple back quotes](https://docs.github.com/en/github/writing-on-github/basic-writing-and-formatting-syntax#quoting-code) with proper indentation it should be easier to grasp.)

bahm-lgs, 2020-07-10:
Sry for the malformed example, this was just copy&paste from my shell. Something got lost and the formatting was really crappy anyways. Please check again now, I fixed it. I'll try again with a "sleep 1" between write and readback, to check if I still can reproduce it.

miccoli, 2020-07-10:
I cannot replicate your setup, unfortunately... (I hope that some other developer could look into this.) I would just add another silly question, just to be 100% sure, although I think you already checked this.

Can you positively confirm that the `0xFF` value is on the Eeprom? I mean can you rule out a transient read error? That is: subsequent `owread` calls will give the same value, even after a power cycle? What about the other pages?

bahm-lgs, 2020-07-10:
> Can you positively confirm that the `0xFF` value is on the Eeprom? I mean can you rule out a transient read error? That is: subsequent `owread` calls will give the same value, even after a power cycle? What about the other pages?

Yes, it is definitely in the Eeprom. All other pages are still ok after the error happened.

honechko, 2020-07-14:
As I know, write to eeprom occurs in two steps 1) flush eeprom to all FF 2) write actual data.
Try turn on owserver debug 'owserver --debug' and make 'diff -ru' between "successfull write" part of log and "error" part.

bahm-lgs, 2020-07-14:
> As I know, write to eeprom occurs in two steps 1) flush eeprom to all FF 2) write actual data.
> Try turn on owserver debug 'owserver --debug' and make 'diff -ru' between "successfull write" part of log and "error" part.

Hi, this is exactly what I tried, but as stated above the error does not happen when debugging is turned on. Seems to be a timing issue.

honechko, 2020-07-14:
According to DS28EC20P+ datasheet, there is no separate commands for "flush" and "write" eeprom, there is only "copy scratchpad" command, so I guess error occurs internally in device (it successfully flushes eeprom, but does not write scratchpad contents to eeprom). From datasheet: "The duration of the device’s internal data transfer is tPROG (10ms) during which the 1-Wire bus must be idle or actively pulled high." Do you have other devices on bus or other clients connected to owserver?

bahm-lgs, 2020-07-14:
> According to DS28EC20P+ datasheet, there is no separate commands for "flush" and "write" eeprom, there is only "copy scratchpad" command, so I guess error occurs internally in device (it [...]

Yes, I have one other device connected, a DS18B20U+ temperature sensor. But for these tests I made sure no other client is accessing owserver. The programming time I could visualize with the scope, due to a small voltage drop on the bus during programming. In my setup the programming takes actually around 8ms and with the scope I verified the following read did not occur during the programming time.

miccoli, 2020-07-16:
Nevertheless, the fact that with verbose debug the error does not occur points to a timing problem. I'm unsure if this is a OWFS bug or not... I lack both the expertise and equipment to check.

However I will flag this issue as a "bug", hoping that someone is av

honechko, 2020-08-07:
bahm-lgs, can you provide you schematic? Is your w1 bus 5V or 3.3V? What is the value of pull up resistor?
For some operations (such as write eeprom) devices require certain level of current.

bahm-lgs, 2020-08-17:
> bahm-lgs, can you provide you schematic? Is your w1 bus 5V or 3.3V? What is the value of pull up resistor?
> For some operations (such as write eeprom) devices require certain level of current.

Hi, my W1 bus is 3.3V, pull-up is 1K, I also tried with 500 Ohm, no difference in behavior. In the schematic there is nothing special otherwise, just the two devices DS28EC20P+ and DS18B20U+ connected to the same bus.

honechko, 2020-08-26:
bahm-lgs, is your device powered directly from 3.3V line (third wire) or from 1wire line (parasitic power)? In last case there is voltage drop (0.4..0.6V) on internal diode between 1W and VCC, so  supply voltage may be insufficient for programming eeprom.

bahm-lgs, 2020-08-26:
> bahm-lgs, is your device powered directly from 3.3V line (third wire) or from 1wire line (parasitic power)? In last case there is voltage drop (0.4..0.6V) on internal diode between 1W and VCC, so supply voltage may be insufficient for programming eeprom.

The DS28EC20 is powered via 1wire, the device does not have a dedicated power supply pin. The 1wire pullup voltage is within the specified range of 3.3V+/-5%. The pullup resistor is specified as 0.3-2k ohms and we use 1k. Even reducing to 500 ohms did not change anything regarding failure probability.

After I changed the error_level to 9, the failure never occured anymore. I am fine with that, since the debug output is going nowhere and the system load does not increase notably. Unfortunately I don't have the time to further debug this. If nobody else can reproduce it, we might as well close this issue.

miccoli, 2020-08-28:
I would suggest to keep this issue open: the fact that the issue disappears depending on the debug level is worrying...