owfs/owfs #75: python3-ow module/bindings not built

Kihltech, 2020-09-29:
The default Python version for some modern Linux distributions (i.e Ubuntu 20.04) is Python3.

Currently, there is no python3-ow module/bindings built for Ubuntu 20.04. 

What is the current status on this topic?

ianka, 2020-09-30:
The python-bindings which come with owfs are obsolete. Use *owserver* and Stefano Miccoli's *pyownet* instead.

miccoli, 2020-09-30:
The "official" bindings are not maintained. There was some activity to make them python2/3 compatible, see PR #32, but the work is now stalled.

Unless you need support for some legacy code, my suggestion is to switch to a more modern implementation: [trio-owfs](https://pypi.org/project/trio-owfs/) or [pyownet](https://pyownet.readthedocs.io/en/latest/)

stromnet, 2020-10-01:
Headsup: the FreeBSD port will also disable python bindings "soon" (whenever I get around to submit the patch). Reason is that Python2.7 is EOL and scheduled for removal, and any ports depending on py2.7 will be removed too unless fixed before 2020-12-31.

https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=249337

Kihltech, 2020-10-01:
> The "official" bindings are not maintained. There was some activity to make them python2/3 compatible, see PR #32, but the work is now stalled.
>
> Unless you need support for some legacy code, my [...]

From what I can tell after a quick look, both those packages might help me! Thanks a lot!