owfs/owfs #76: DS9097U-009 failing to connect

silvaphoenix, 2020-09-29:
Hi,
I have replaced the motherboard on a control system that has been working for some years.
On the new board the DS9097U-009 is not connecting. Clearly all the board hardware is new but hard drive and software are unchanged.
The serial port parameters are:
Note the default baud rate is 115200 that I changed to 9600.

pwrp@LinITX:/opt/owfs/bin$ setserial -av /dev/ttyS0
/dev/ttyS0, Line 0, UART: 16550A, Port: 0x03f8, IRQ: 4
        Baud_base: 9600, close_delay: 50, divisor: 0
        closing_wait: 3000
        Flags: spd_normal skip_test

Starting owserver in debug mode gives the following

pwrp@LinITX:/opt/owfs/bin$ owserver -d /dev/ttyS0 --debug 
DEBUG MODE
libow version:
        3.1p5
  DEBUG: ow_daemon.c:(170) main thread id = 140345131562944
  DEBUG: ow_inotify.c:(80) No configuration files to monitor
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
   CALL: ow_parsename.c:(104) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
  DEBUG: ow_ds9097U.c:(471) Attempt 0 of 3 to initialize the DS9097U

At this point nothing else happens.
Opening another terminal to run owdir gives no result

pwrp@LinITX:/opt/owfs/bin$ owdir
pwrp@LinITX:/opt/owfs/bin$ 

I would appreciate any help with this problem
Thanks Stephen

silvaphoenix, 2020-09-30:
Hi,
Starting from a machine reboot I get a lot more output. Here is an extract (the new board has 2 off RS 232 connections so I tried the second)

DEBUG: ow_serial_open.c:(174) Cannot power cycle a closed serial port
  DEBUG: ow_ds9097U.c:(565) Send the initial reset to the bus master.
  DEBUG: ow_tcp_read.c:(63) attempt 1 bytes Time: 5.000000 seconds
CONNECT: ow_tcp_read.c:(109) TIMEOUT after 0 bytes
  DEBUG: ow_ds9097U.c:(660) Failed first attempt at resetting baud rate of bus master /dev/ttyS1
  DEBUG: ow_tcp_read.c:(63) attempt 1 bytes Time: 5.000000 seconds
CONNECT: ow_tcp_read.c:(109) TIMEOUT after 0 bytes
  DEBUG: ow_ds9097U.c:(665) Failed second attempt at resetting baud rate of bus master /dev/ttyS1

There seems to be an issue with the serial port configuration?
Any ideas?
Stephen
Stephen

silvaphoenix, 2020-09-30:
Hi,
I've borrowed an RS 232 to USB converter and success all work ok through ttyUSB0.
So definitely a serial configuration problem.
Any advice on steps to resolve?
Stephen

ianka, 2020-09-30:
Check the permissions of /dev/ttyS0. Maybe the user who runs owserver needs to be in a certain group (often "dialout").

silvaphoenix, 2020-09-30:
Thanks ianka, the user is in the tty and dialout groups. The permissions are the same for ttyUSB0 that works

ianka, 2020-10-01:
What kind of board is this? Does the serial ports work *at all*? I ask this because there's even x86 hardware out there which has a bogus interrupt routing for the serial ports so you can't receive anything.

silvaphoenix, 2020-10-01:
Hi Jan,
Thanks for looking at this.
I have no other devices to try so don't know if they work at all.
The board is a Gigabyte GA-J3455N-D3H.
I note the DS9097U needs a minimum RS232 signal voltage of +/-5v. I'm trying to find if the board has this or 3.3v.
From setserial the UART is 16550A. Could this be the problem?

ianka, 2020-10-01:
That board has DE9 sockets for the RS232 ports, so there cannot be a problem with a reversed or unmatching pin header connector. Good to know. Check the voltage of pin 2 against pin 5. It must be -5V to -12V, otherwise the DS9097U will not function correctly.

The 16550A is a bog standard PC UART. There should be no problems. But it all depends on the interrupt routing. The BIOS may say its IRQ4 or IRQ3, but in reality, it's routed to IRQ12 or so. I had this problem with some embedded x86 boards already. On the other hand, Gigabyte is a good brand and I doubt they sell green bananas.

silvaphoenix, 2020-10-02:
Voltages are pin 2 (RXD) 0.09, pin 3 (TXD) - 10.38, pin 4 (DTR) - 10.38 and pin 7 (RTS) -10.38
So the problem is that pin 2 is 0. It's the same on both ports.
I've been reading up on serial ports and found that pins 2, 3 and 5 are fixed by the hardware. As there should be a -ve voltage on pin 2 either the board is faulty or perhaps there is a jumper on the board that needs setting