owfs/owfs #77: [systemd] Job for owserver.service failed because the service did not take the steps required by its unit configuration.

kalvdans, 2020-09-30:
After upgrading from Ubuntu 18.04 to 20.04 I started to see errors when restarting owserver.service. They appear approximately every second time without clues from the logs.
```
$ sudo systemctl stop owserver.service; sudo systemctl start owserver.service
Job for owserver.service failed because the service did not take the steps required by its unit configuration.
See "systemctl status owserver.service" and "journalctl -xe" for details.

$ systemctl status owserver.service
● owserver.service - Backend server for 1-wire control
     Loaded: loaded (/lib/systemd/system/owserver.service; disabled; vendor preset: enabled)
     Active: active (running) since Wed 2020-09-30 12:40:53 UTC; 7s ago
       Docs: man:owserver(1)
   Main PID: 13861 (owserver)
      Tasks: 2 (limit: 18965)
     Memory: 1.5M
     CGroup: /system.slice/owserver.service
             └─13861 /usr/bin/owserver -c /etc/owfs.conf

Sep 30 12:40:53 arvid-m12 systemd[1]: Starting Backend server for 1-wire control...
Sep 30 12:40:53 arvid-m12 OWFS[13861]: DEFAULT: ow_daemon.c:(144) Entered background mode, quitting.
Sep 30 12:40:53 arvid-m12 OWFS[13861]: DEFAULT: ow_usb_msg.c:(214) Opened USB DS9490 bus master at 1:3.
Sep 30 12:40:53 arvid-m12 OWFS[13861]: DEFAULT: ow_usb_cycle.c:(128) Set DS9490 1:3 unique id to 81 2C B9 3C 00 00 00 87
Sep 30 12:40:53 arvid-m12 systemd[1]: Started Backend server for 1-wire control.
```

Versions:
```
$ dpkg --list | grep "  ow"
ii  owfs-common                          3.2p3+dfsg1-5build1               all          common files used by any of the OWFS programs
ii  owserver                             3.2p3+dfsg1-5build1               amd64        Backend server for 1-Wire control
```

I've tried both with and without socket activation enabled, with the same intermittent results. I suspect it has to do with sd_notify() mechanism. Be aware that Ubuntu have some patches on top of the releases, the error might be in the packaging, but reporting here first.

ianka, 2020-09-30:
*Owserver* turns itself into a demon by default. If you don't want that, you have to start it with the *--foreground* option. The author of the unit file should be aware of that. You should show us the content of */etc/owfs.conf* and the unit file.

kalvdans, 2020-09-30:
Thanks ianka for your insight!

This is the contents of /lib/systemd/system/owserver.service
```
[Unit]
Description=Backend server for 1-wire control
Documentation=man:owserver(1)

[Service]
Type=notify
NotifyAccess=all
ExecStart=/usr/bin/owserver -c /etc/owfs.conf
Restart=on-failure
#User=Debian-ow
#Group=Debian-ow

[Install]
WantedBy=multi-user.target
Also=owserver.socket
```
Contents of /lib/systemd/system/owserver.socket
```
[Unit]
Description=Listening socket for owserver
Documentation=man:owserver(1)

[Socket]
ListenStream=4304

[Install]
WantedBy=sockets.target
```
Contents of /etc/owfs.conf
``` 
! server: server = 127.0.0.1:4304
server: usb = all
alias = /etc/owfs.aliases
http: port = 2121
ftp: port = 2120
server: port = 127.0.0.1:4304
```
systemd version 
```
systemd 245 (245.4-4ubuntu3.2)
+PAM +AUDIT +SELINUX +IMA +APPARMOR +SMACK +SYSVINIT +UTMP +LIBCRYPTSETUP +GCRYPT +GNUTLS +ACL +XZ +LZ4 +SECCOMP +BLKID +ELFUTILS +KMOD +IDN2 -IDN +PCRE2 default-hierarchy=hybrid
```

kalvdans, 2020-09-30:
Appending `--foregruond` to the ExecStart command seems to work fine 100% of the times. I think that is the fix! I will post an issue to the Ubuntu team and close this one.

kalvdans, 2020-09-30:
Actually reopening since the [owserver.service template in this repo](https://github.com/owfs/owfs/blob/master/src/scripts/systemd/owserver.service.in#L8) also don't have --foreground.

kalvdans, 2020-09-30:
Ubuntu bug opened at https://bugs.launchpad.net/ubuntu/+source/owfs/+bug/1897966

miccoli, 2020-09-30:
Please see also #46

kalvdans, 2020-09-30:
Yes, closing as a duplicate of #46. Sorry for doing a sloppy search for existing bugs before posting!

closed 2020-09-30 by kalvdans