owfs/owfs #84: Fedora-33 and owfs-3.2p4

ilya-ilyin, 2021-01-19:
OWFS has stopped working after FC33 upgrade. I use either owfs-3.2p4 or owfs-3.1p4, but it gives the same result shown below, i.e. it cannot recognize three USB sensors as before. What has gone missing here?

  DEBUG: ow_cache.c:(295) Flipping cache tree (purging timed-out data)
  DEBUG: ow_cache.c:(313) flip cache. tdestroy() will be called.
  DEBUG: ow_api.c:(74) OWLIB started with <--usb>
  DEBUG: ow_inotify.c:(80) No configuration files to monitor
  CONNECT: ow_dnssd.c:(75) DL_open [libdns_sd.so] success
  DEBUG: ow_dnssd.c:(85) Linked in Bonjour function DNSServiceRefSockFD
  DEBUG: ow_dnssd.c:(86) Linked in Bonjour function DNSServiceProcessResult
  DEBUG: ow_dnssd.c:(87) Linked in Bonjour function DNSServiceRefDeallocate
  DEBUG: ow_dnssd.c:(88) Linked in Bonjour function DNSServiceResolve
  DEBUG: ow_dnssd.c:(89) Linked in Bonjour function DNSServiceBrowse
  DEBUG: ow_dnssd.c:(90) Linked in Bonjour function DNSServiceRegister
  DEBUG: ow_dnssd.c:(91) Linked in Bonjour function DNSServiceReconfirmRecord
  DEBUG: ow_dnssd.c:(92) Linked in Bonjour function DNSServiceCreateConnection
  DEBUG: ow_dnssd.c:(93) Linked in Bonjour function DNSServiceEnumerateDomains
  CALL: ow_parsename.c:(104) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
]*$> compiled to 0x7f3e5cde2740 expression <^ *([^ ]+)[ 
]*$> compiled to 0x7f3e5cde2700 expression <^ *([^ ]+) *: *([^ ]+)[ 
]*$> compiled to 0x7f3e5cde26c0 expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[ 
  DEBUG: ow_usb_cycle.c:(169) Compare (add,bus) (2,3) with (-1,-1) handle (nil)

ianka, 2021-01-19:
How do you start owfs? Please post the systemd unit file and the configuration file. Also, as this is about USB, please show us what kernel modules are loaded, and what DS9490 udev rules you use.

ilya-ilyin, 2021-01-20:
Now it has started to work again after these kernel modules were blacklisted:

nano /etc/modprobe.d/blacklist.conf
blacklist ds9490r
blacklist ds2490
blacklist wire

since there were loaded a day before:
lsmod
ds2490             16384  0
wire                   53248  2 ds2490,w1_therm

No other rules are applied or systemd used. The only OWFS invocation I use is: OW_init("--usb") from owfs-capi.
I guess DS* and WIRE modules were not loaded in the older Fedora distributions which explains why it was working for seven years with no problem.

closed 2021-01-20 by ilya-ilyin

ianka, 2021-01-20:
You could as well use the kernel modules and configure OWFS to use --w1 instead of --usb.

ilya-ilyin, 2021-01-21:
I enabled these blacklisted kernel modules again, but --w1 would not recognize any existing USB sensors, manpage owfs says --w1 is still under development, is that the case?

ianka, 2021-01-21:
No, --w1 works just fine. But you maybe need to load the _cn_ unified kernelspace<->userspace driver module as well. Some distribution build that driver as a module, and it's not autoloaded.

ilya-ilyin, 2021-01-21:
cn module is buildin according to modinfo and it cannot help to recognise any USB. I saw numerous similar posts in the OFWS mailing lists.

ianka, 2021-01-21:
It's most likely a problem with the ds2490 kernel driver then. That one used to work as fine as the DS2482 kernel driver I use. Maybe something broke with an update.

ilya-ilyin, 2021-01-22:
I added ds2482 module 
wire                   53248  3 ds2482,ds2490,w1_therm
ds2490              16384  0
w1_therm           36864  0

and "owfs --w1 -m dd/" still shows this contents
drwxr-xr-x   1 root  root  4096 Jan 21 19:14 bus.0
drwxr-xr-x   1 root  root  4096 Jan 21 19:14 bus.1
drwxr-xr-x   1 root  root  4096 Jan 21 19:14 settings
drwxr-xr-x   1 root  root  4096 Jan 21 19:14 statistics
drwxr-xr-x   1 root  root  4096 Jan 21 19:14 structure
drwxr-xr-x   1 root  root  4096 Jan 21 19:14 system
drwxr-xr-x   1 root  root  4096 Jan 21 19:14 uncached

instead of this for "owfs -uall -m dd/" and blacklisted modules
drwxrwxrwx   1 root  root  4096 Jan 21 19:27 12.8871B7000000
drwxrwxrwx   1 root  root  4096 Jan 21 19:27 28.5EFC3B000000
drwxrwxrwx   1 root  root  4096 Jan 21 19:27 28.FCED3B000000
drwxrwxrwx   1 root  root  4096 Jan 21 19:27 81.EDBE21000000
drwxr-xr-x   1 root  root  4096 Jan 21 19:27 alarm  etc.

I use now repository package owfs-libs-3.2p3-6.fc33.x86_64 instead of github/owfs

ilya-ilyin, 2021-01-22:
... and these two modules look quite OK to me:

[root@solaris pepsi]# modinfo ds2490
filename:       /lib/modules/5.10.7-200.fc33.x86_64/extra/drivers/w1/masters/ds2490.ko.xz
license:        GPL
description:    DS2490 USB <-> W1 bus master driver (DS9490*)
author:         Evgeniy Polyakov <zbr@ioremap.net>
alias:          usb:v04FAp2490d*dc*dsc*dp*ic*isc*ip*in*
depends:        wire
retpoline:      Y
intree:         Y
name:           ds2490
vermagic:       5.10.7-200.fc33.x86_64 SMP mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        54:40:C4:49:C3:9F:C1:35:51:96:20:1D:DF:BE:B9:2D:10:F5:99:5E
sig_hashalgo:   sha256

root@solaris pepsi]# modinfo ds2482
filename:       /lib/modules/5.10.7-200.fc33.x86_64/extra/drivers/w1/masters/ds2482.ko.xz
license:        GPL
description:    DS2482 driver
author:         Ben Gardner <bgardner@wabtec.com>
alias:          i2c:ds2482
depends:        wire
retpoline:      Y
intree:         Y
name:           ds2482
vermagic:       5.10.7-200.fc33.x86_64 SMP mod_unload 
sig_id:         PKCS#7
signer:         Fedora kernel signing key
sig_key:        54:40:C4:49:C3:9F:C1:35:51:96:20:1D:DF:BE:B9:2D:10:F5:99:5E
sig_hashalgo:   sha256