owfs/owfs #86: owdir/owserver hangs after reconnecting device

tengel, 2021-03-13:
After disconnecting and reconnecting one wire devices, `owserver` and `owdir` hang and do not update the device list anymore. 

I am using a Raspberry Pi with `w1` module and run `owdir` to see the list of connected devices. After disconnecting and reconnecting a device several or many times, `owdir` hangs. It returns after about 30 seconds, but does not show connected devices anymore. Recovery is only possible after reboot. Restarting `owserver` does not help. Same behaviour was observed with `owfs` and `ls` and with owcapi `OW_get()`.

Raspberry Pi, owfs-3.2p4, kernel 4.19.113-v7 (Buildroot 2020.05.2)

Steps to reproduce:
- `owserver --pid_file /run/owserver.pid --w1 --port localhost:4304 --foreground --error_level=9 &> owserver.out`
- `watch -n1 owdir`
- physically disconnect and connect one wire device many times
- after several reconnects `owdir` takes 30 seconds to return and does not show
  connected device.

output from `owserver`: [owserver.out.txt](https://github.com/owfs/owfs/files/6135169/owserver.out.txt)

ianka, 2021-03-13:
What host adapter do you use? This is a well known bug of the DS2482-800. Only power-cycling the chip helps.

tengel, 2021-03-13:
The 1-Wire device is directly connected to a GPIO using the `w1-gpio` kernel module.

tengel, 2021-04-17:
This seems to be a kernel issue. I removed the lines
```
mutex_unlock(&dev->mutex);
request_module("w1-family-0x%02X", rn->family);
mutex_lock(&dev->mutex);
```
in file `drivers/w1/w1.c`, function `int w1_attach_slave_device(struct w1_master *dev, struct w1_reg_num *rn)`, 
line 755. Since then, I did not observe any problems anymore.

ianka, 2021-04-17:
Ok, thanks for the investigation. Any further pointers why you think those lines are a problem?
To escalate it to kernel developers, we need some more info.

tengel, 2021-04-25:
This is the stack trace after enabling lock debugging options:

[   73.076622] ------------[ cut here ]------------
[   73.076654] WARNING: CPU: 0 PID: 117 at kernel/locking/lockdep.c:3665 lock_release+0x288/0x4cc
[   73.076667] DEBUG_LOCKS_WARN_ON(depth <= 0)
[   73.076671] Modules linked in: ipv6 brcmfmac brcmutil bcm2835_v4l2(C) bcm2835_codec(C) v4l2_common cfg80211 v4l2_mem2mem videobuf2_vmalloc evdev joydev bcm2835_mmal_vchiq(C) rfkill videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon videobuf2_common ads7846 videodev vc_sm_cma(C) media fixed uio_pdrv_genirq uio w1_ds2431 w1_gpio wire cn hwmon fb_ili9340(C) fbtft(C) syscopyarea sysfillrect sysimgblt fb_sys_fops spi_bcm2835
[   73.076809] CPU: 0 PID: 117 Comm: w1_bus_master1 Tainted: G         C        4.19.113-v7 #2
[   73.076821] Hardware name: BCM2835
[   73.076857] [<801121bc>] (unwind_backtrace) from [<8010d508>] (show_stack+0x20/0x24)
[   73.076883] [<8010d508>] (show_stack) from [<8089caec>] (dump_stack+0xd8/0x11c)
[   73.076910] [<8089caec>] (dump_stack) from [<80121108>] (__warn.part.3+0xcc/0xe8)
[   73.076934] [<80121108>] (__warn.part.3) from [<8012119c>] (warn_slowpath_fmt+0x78/0x94)
[   73.076958] [<8012119c>] (warn_slowpath_fmt) from [<8018288c>] (lock_release+0x288/0x4cc)
[   73.076984] [<8018288c>] (lock_release) from [<808b4804>] (__mutex_unlock_slowpath+0x58/0x350)
[   73.077007] [<808b4804>] (__mutex_unlock_slowpath) from [<808b4b18>] (mutex_unlock+0x1c/0x20)
[   73.077056] [<808b4b18>] (mutex_unlock) from [<7f04eb34>] (w1_attach_slave_device+0x10c/0x464 [wire])
[   73.077123] [<7f04eb34>] (w1_attach_slave_device [wire]) from [<7f04f018>] (w1_slave_found+0xcc/0xd4 [wire])
[   73.077187] [<7f04f018>] (w1_slave_found [wire]) from [<7f0507c4>] (w1_found_send_slave+0x24/0x38 [wire])
[   73.077251] [<7f0507c4>] (w1_found_send_slave [wire]) from [<7f04f87c>] (w1_search+0x264/0x28c [wire])
[   73.077314] [<7f04f87c>] (w1_search [wire]) from [<7f051dac>] (w1_search_devices+0x50/0x54 [wire])
[   73.077378] [<7f051dac>] (w1_search_devices [wire]) from [<7f04f91c>] (w1_search_process_cb+0x78/0x120 [wire])
[   73.077442] [<7f04f91c>] (w1_search_process_cb [wire]) from [<7f051180>] (w1_process_cb+0x214/0x3b0 [wire])
[   73.077505] [<7f051180>] (w1_process_cb [wire]) from [<7f04fa1c>] (w1_process_callbacks+0x58/0x7c [wire])
[   73.077568] [<7f04fa1c>] (w1_process_callbacks [wire]) from [<7f04fb30>] (w1_process+0xf0/0x150 [wire])
[   73.077613] [<7f04fb30>] (w1_process [wire]) from [<801444fc>] (kthread+0x174/0x178)
[   73.077636] [<801444fc>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[   73.077651] Exception stack(0xb7299fb0 to 0xb7299ff8)
[   73.077669] 9fa0:                                     00000000 00000000 00000000 00000000
[   73.077688] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[   73.077706] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[   73.077722] ---[ end trace fe2f738c402c434a ]---

ianka, 2021-04-26:
Thanks again.

rgillan, 2021-04-27:
> What host adapter do you use? This is a well known bug of the DS2482-800. Only power-cycling the chip helps.

ianka, can you point us to anything on this?  We've got a DS2482-800 on board and are seeing this happen (have been chasing our tails a little this past couple of days to try and isolate the cause).  Presents us a little bit of an issue as we're switching 28e17's on and off the bus. 

Armbian Ubuntu 5.10.21-sunxi owfs-v3.2p4

rgillan, 2021-04-27:
Bit more info.  Using owserver with w1 option, as we have the w1_ds28e17 module loaded.  When owserver goes into slowdown, owdir eventually responds but only sees the devices on other master buses, not the one having the disconnect/reconnections going on.
The ds28e17 device is switched on and off the bus with a ds2406, and when in this slowdown/failed state it can't see either of those two chips on the bus.  They are still working though it seems, as we can access the remote i2c connected sensors, and they are still visible in /sys/bus/w1/devices

rgillan, 2021-04-29:
We have patched the linux-mainline module and rebuilt Armbian (removing the three lines identified above), and pleased to say that the same issue with the ds2482 master appears to have been resolved.  We are yet to do extensive testing but switching a ds28e17 on and off the bus dozens of times has not locked up the device, which prior would guarantee hang after the second or third disconnection (I hope this isn't too premature a post)

ianka, 2021-04-29:
I'm very happy to hear that!

I always though the problem with that DS2482-800 was inherent to the chip, because power-cycling the chip alone helps. You don't have to power-cycle the computer. It seemed to me the problem is triggered by a short on any of the eight buses. Now I start to think the problem is that this request_module() may ask for device family 0x00, which does of course not exist.

rgillan, 2021-04-29:
We have had a support case with Maxim going the past week on this, and they were suspect that the master was mid communications when the bus got interrupted, and had us check all sorts of things like i2c voltage/state at the time of it hanging, but that all looked ok.  They also surmised it may have needed a bunch more SCL clock cycles to clear out the state, but it appears that this isn't needed.  We are going to stress test it over the next week and will update here with the results.  Saves us a board turn which I am very happy about ;) Appreciate all your efforts on the w1 system.

rgillan, 2021-05-03:
Been rock solid all weekend.  Have our topology code switching the ds28e17's on and off the bus and connecting/disconnecting downstream devices.  Think the removal of those three lines have fixed it (big thanks)

ianka, 2021-05-03:
I will discuss the problem with the w1 kernel subsystem maintainer Evgeniy Polyakov then. I may come back to you.

mschlenstedt, 2022-11-12:
Any news on that? I have the same problem with w1 kernel module with 5.10.92-v7+ #1514 SMP Mon Jan 17 17:36:39 GMT 2022 armv7l GNU/Linux