owfs/owfs #87: Owget "memory" vs "pages/page.ALL" vs "pages/page.0"

kbfifi, 2021-04-10:
I scanned the internet and read all kinds of documents. I'm sorry if this the wrong forum to ask, but I guess you are experts that might know.

I've written a DS1992 (1kbit memory, 128 bytes = 4 pages of 32 bytes) slave emulator using the "OneWireSlave" library from "youen/OneWireArduinoSlave" inspired by the example FakeDS18B20.

It's working and I'd like to use it as a wired alternative for my NRF24L01 arduino projects. However I don't understand the low level protocol byte difference between the following calls. Note that F0 is the command for read memory followed by TA1 and TA2 forming the the address offset:
owget /08.000000000002/pages/page.ALL => eventually results in command F0 60 00
owget /08.000000000002/pages/page.3 => eventually results in command F0 60 00
Both command bytes are identical but to my understanding the first call shoudl somehow return all 128 bytes and the 2nd call should return page 3 with 32 bytes

Something similar is happening here:
owget /08.000000000002/pages/page.0 => eventually results in command F0 00 00
owget /08.000000000002/memory => eventually results in command F0 00 00
First call should return page 0 with 32 bytes
Second call should return all memory, meaning 128 bytes (page 0-3)

 For now  I simply return 32 bytes for all calls, but that doesn't make sense.

Do you have any clue how the DS1992 would discriminate these calls?

ianka, 2021-04-10:
To my understanding, you should always return **all** bytes, starting from the address selected in the command. The master simply stops to send bitframes as soon it has read enough. Either after one byte, 23 bytes, 32 bytes or 128 bytes.

kbfifi, 2021-04-10:
You made my day! Thanks for the info. I never knew the master asks for bits even for receiving data. I always assumed the F0 00 00 command was the last master action. I fixed the code and now it always sends till the last byte. My scope confirms that only the required bytes appear on the line. Thanks!

kbfifi, 2021-04-11:
Although a big step forward, I facing instability espcially with the pages/pages.ALL command. It succeeds and fails. Looks like a timing issue of some kind. I setup a debug session with owserver. Here's a log with an owget pages/page.0 (this is always seems to work) followed by owget pages/page.aLL (fails immediate here)

Hope you can give me some pointers.

` DEBUG: ow_regex.c:(201) 0: 10->12 found <pages/page><.0><>
   CALL: data.c:(144) Read message
  DEBUG: read.c:(55) ReadHandler start
  DEBUG: read.c:(61) ReadHandler: From Client sm->payload=30 sm->size=65536 sm->offset=0
  DEBUG: read.c:(79) ReadHandler: call FS_read_postparse on /08.000000000002/pages/page.0
  DEBUG: ow_read.c:(72) /08.000000000002/pages/page.0
  DEBUG: ow_read.c:(204) /08.000000000002/pages/page.0
  DEBUG: ow_read.c:(238) About to read </08.000000000002/pages/page.0> extension=0 size=32 offset=0
  DEBUG: ow_read.c:(333) file_length=32 offset=0 size=32
  DEBUG: ow_read.c:(435) Read a separate element /08.000000000002/pages/page.0
  DEBUG: ow_select.c:(70) Selecting a path (and device) path=/08.000000000002/pages/page.0 SN=08 00 00 00 00 00 02 4D last path=00 00 00 00 00 00 00 00
  DEBUG: ow_select.c:(84) Continuing root branch
  DEBUG: ow_ds9490.c:(562) DS9490 RESET. changed 0, flex: 1
  DEBUG: ow_ds9490.c:(613) DS9490_Reset: OK
  DEBUG: ow_select.c:(226) Selecting device 08 00 00 00 00 00 02 4D
  DEBUG: ow_transaction.c:(113) send = 0
  DEBUG: ow_transaction.c:(208) end = 0
  DEBUG: ow_transaction.c:(91) select = 0
  DEBUG: ow_transaction.c:(113) send = 0
  DEBUG: ow_transaction.c:(131) readin = 0
  DEBUG: ow_transaction.c:(208) end = 0
  DEBUG: ow_read.c:(620) Read /08.000000000002/pages/page.0 Extension 0 Gives result 0
  DEBUG: ow_cache.c:(374) Adding data for /08.000000000002/pages/page.0
  DEBUG: ow_read.c:(253) return=0
OWQ OneWireQuery structure of /08.000000000002/pages/page.0
    OneWireQuery size=32 offset=0, extension=0
Byte buffer OneWireQuery buffer, length=32
--000: 3E 3E 50 61 67 65 20 30 3C 3C 2E 3B 3C 3D 3E 3F
--016: 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F
   <>>Page 0<<.;<=>?@ABCDEFGHIJKLMNO>
    Cleanup = 0006    OneWireQuery I=32 U=32 F=1.58101E-322 Y=32 D=Thu Jan  1 01:00:32 1970

--- OneWireQuery done
  DEBUG: ow_read.c:(263) After read is performed (bytes or error 32)
OWQ OneWireQuery structure of /08.000000000002/pages/page.0
    OneWireQuery size=32 offset=0, extension=0
Byte buffer OneWireQuery buffer, length=32
--000: 3E 3E 50 61 67 65 20 30 3C 3C 2E 3B 3C 3D 3E 3F
--016: 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F
   <>>Page 0<<.;<=>?@ABCDEFGHIJKLMNO>
    Cleanup = 0006    OneWireQuery I=32 U=32 F=1.58101E-322 Y=32 D=Thu Jan  1 01:00:32 1970

--- OneWireQuery done
  DEBUG: ow_read.c:(226) /08.000000000002/pages/page.0 returns 32
  DEBUG: ow_read.c:(103) /08.000000000002/pages/page.0 return 32
  DEBUG: read.c:(81) ReadHandler: FS_read_postparse read on /08.000000000002/pages/page.0 return = 32
OWQ OneWireQuery structure of /08.000000000002/pages/page.0
    OneWireQuery size=32 offset=0, extension=0
Byte buffer OneWireQuery buffer, length=32
--000: 3E 3E 50 61 67 65 20 30 3C 3C 2E 3B 3C 3D 3E 3F
--016: 40 41 42 43 44 45 46 47 48 49 4A 4B 4C 4D 4E 4F
   <>>Page 0<<.;<=>?@ABCDEFGHIJKLMNO>
    Cleanup = 0006    OneWireQuery I=32 U=32 F=1.58101E-322 Y=32 D=Thu Jan  1 01:00:32 1970

--- OneWireQuery done
  DEBUG: read.c:(89) ReadHandler: FS_read_postparse ok size=32
  DEBUG: read.c:(100) ReadHandler: To Client cm->payload=32 cm->size=32 cm->offset=0
Byte buffer Data returned to client, length=32
-- NULL buffer
  DEBUG: data.c:(146) Read message done value=0x7f88000b20
  DEBUG: ow_parsename.c:(63) /08.000000000002/pages/page.0
  DEBUG: data.c:(193) DataHandler: FS_ParsedName_destroy done
  DEBUG: data.c:(207) DataHandler: cm.ret=32
  DEBUG: to_client.c:(76) payload=32 size=32, ret=32, sg=0x10A offset=0
  DEBUG: data.c:(226) Finished with client request
  DEBUG: handler.c:(135) OWSERVER handler done
[... 174 lines cut ...]
  DEBUG: ow_read.c:(103) /08.000000000002/pages/page.ALL return -22
  DEBUG: read.c:(81) ReadHandler: FS_read_postparse read on /08.000000000002/pages/page.ALL return = -22
OWQ OneWireQuery structure of /08.000000000002/pages/page.ALL
    OneWireQuery size=128 offset=0, extension=-1
Byte buffer OneWireQuery buffer, length=128
--000: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
--016: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
--032: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
--048: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
--063: == abridged ==
   <................................................................>
    Cleanup = 0016    OneWireQuery I=-2147475808 U=2147491488 F=2.70554E-312 Y=-2147475808 D=
--- OneWireQuery done
  DEBUG: read.c:(86) ReadHandler: FS_read_postparse error -22
  DEBUG: read.c:(100) ReadHandler: To Client cm->payload=0 cm->size=0 cm->offset=0
  DEBUG: data.c:(146) Read message done value=(nil)
  DEBUG: ow_parsename.c:(63) /08.000000000002/pages/page.ALL
  DEBUG: data.c:(193) DataHandler: FS_ParsedName_destroy done
  DEBUG: data.c:(207) DataHandler: cm.ret=-22
  DEBUG: to_client.c:(76) payload=0 size=0, ret=-22, sg=0x10A offset=0
  DEBUG: to_client.c:(85) No data
  DEBUG: data.c:(226) Finished with client request
  DEBUG: handler.c:(135) OWSERVER handler done
  DEBUG: ow_net_server.c:(259) Normal completion.
`

ianka, 2021-04-11:
I'm sorry, I see nothing from this mess. I recommend you to check against the DS1992 datasheet if your code does exactly what a DS1992 would do. The base protocol is explained on the last eight pages of each onewire slave datasheet. You should read that one first.

kbfifi, 2021-04-11:
@ianka thanks for your quick response. I'm familiair with those pages and even tried to follow the call tree from owget till deep down in owserver. It's not that it doesn't work but it seems timing or synchronisation is off. btw haven't seen any description on how pages.page.ALL works. Looks like owserver requests page.0 - page.3 sequentially. Is there some kind of "done" signal owserver expects?

ianka, 2021-04-11:
The log says

> DEBUG: ow_select.c:(226) Selecting device 08 00 00 00 00 00 02 4D
> DEBUG: loop.c:(109) Taking too long, send a keep-alive pulse

To me, that reads as if the slave doesn't return to its idle state on bus reset if the previous command isn't completed — you changed the implementation to 128 bytes, but read only 32 bytes in that first attempt of reading page.0. 
A bus reset is authoritative, it may happen at any time. Commands don't need to complete.

Also, please always read /uncached/<id>/<node> as otherwise the owfs cache may kick in and distord your results.