owfs/owfs #88: Programming a DS2406

rgillan, 2021-05-14:
Hi all,
Sorry if it's a bit off topic but haven't found an answer via the usual search methods.  Trying to use owserver to write the eprom memory on a DS2406.  We have a circuit that can pulse the 1-wire line to the device to +12v, but reading the data sheet it says there's a sequence of commands to be sent from the master but not sure where to start.  We're using a DS2482-800 master via w1.  Any suggested approaches very much appreciated.

Cheers and thanks
Rob

rgillan, 2021-05-14:
some more info.  running:

./owwrite --hex --dir /12.DF07D5000000/pages/page.0 3030303030303031

causes owserver to fail and restart

May 14 07:49:25 nanopineo systemd[1]: owserver.service: Main process exited, code=killed, status=11/SEGV
May 14 07:49:25 nanopineo systemd[1]: owserver.service: Failed with result 'signal'.
May 14 07:49:26 nanopineo systemd[1]: owserver.service: Scheduled restart job, restart counter is at 26.
May 14 07:49:26 nanopineo systemd[1]: Stopped Backend server for 1-wire control.
May 14 07:49:26 nanopineo systemd[1]: Starting Backend server for 1-wire control...

Have a bunch of logs if that helps

Armbian Ubuntu 5.10.21-sunxi owfs-v3.2p4

rgillan, 2021-05-14:
Sorry for the flood.  Could be a different issue.  Currently have a DS2482-800 being managed via w1.  There is a DS2408 on bus 5 and DS18B20 on bus2.  There's the DS2406 on bus 7.  /sys/bus/w1/devices shows the correct device to bus master association in the symlinks.
./owdir to each device (e.g. --dir /12.DF07D5000000) works as expected.  If we however ./owdir --dir /bus.7, it actually shows the device on bus 2.  It seems that the device to bus association is messed up.  If we then run

./owwrite --hex --size 8 --dir /12.DF07D5000000/pages/page.0 3030303030303031

has a bunch of debug logs:

May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: ow_tcp_read.c:(63) attempt 24 bytes Time: 10.000000 seconds
May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: ow_tcp_read.c:(113) read: 24 - 0 = 24
May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: from_client.c:(65) FromClient payload=38 size=8 type=3 sg=0x10A offset=0
May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: from_client.c:(73) FromClient (no servermessage) payload=38 size=8 type=3 controlflags=0x10A offset=0
May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: ow_tcp_read.c:(63) attempt 38 bytes Time: 10.000000 seconds
May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: ow_tcp_read.c:(113) read: 38 - 0 = 38
May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: handler.c:(152) START handler /12.DF07D5000000/pages/page.0
May 14 08:05:57 nanopineo OWFS[12048]:    CALL: data.c:(103) DataHandler: parse path=/12.DF07D5000000/pages/page.0
May 14 08:05:57 nanopineo owserver[12048]: =2 (W1_CMD_SEARCH) len=0
May 14 08:05:57 nanopineo owserver[12048]: NULL data
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=52 type=3 (NLMSG_DONE) flags=0 seq=6|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=6|3 ack=393220 len=16 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=4 id=6
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=0
May 14 08:05:57 nanopineo owserver[12048]: NULL data
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=52 type=3 (NLMSG_DONE) flags=0 seq=6|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=6|3 ack=393220 len=16 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=4 id=6
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=0
May 14 08:05:57 nanopineo owserver[12048]: NULL data
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=52 type=3 (NLMSG_DONE) flags=0 seq=6|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=6|3 ack=393219 len=16 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=4 id=6
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=0
May 14 08:05:57 nanopineo owserver[12048]: NULL data
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=52 type=3 (NLMSG_DONE) flags=0 seq=6|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=6|3 ack=393219 len=16 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=4 id=6
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=0
May 14 08:05:57 nanopineo owserver[12048]: NULL data
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=52 type=3 (NLMSG_DONE) flags=5 seq=5|3 pid=12048
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=5|3 ack=327683 len=16 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=4 id=5
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=0
May 14 08:05:57 nanopineo owserver[12048]: NULL data
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=60 type=3 (NLMSG_DONE) flags=0 seq=5|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=5|3 ack=327684 len=24 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=12 id=5
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=8
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=60 type=3 (NLMSG_DONE) flags=0 seq=5|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (NLMSGHDR: len=52 type=3 (CN_W1_IDXNLMSG_DONE) seq=5|3 ack=327684 len=24 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=12 id=5
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=8
May 14 08:05:57 nanopineo owserver[12048]: ) flags=0 seq=5|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=5|3 ack=327683 len=16 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=4 id=5
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_CMD: cmd=2 (W1_CMD_SEARCH) len=0
May 14 08:05:57 nanopineo owserver[12048]: NULL data
May 14 08:05:57 nanopineo owserver[12048]: NLMSGHDR: len=52 type=3 (NLMSG_DONE) flags=0 seq=5|3 pid=0
May 14 08:05:57 nanopineo owserver[12048]: CN_MSG: idx/val=3/1 (CN_W1_IDX) seq=5|3 ack=327683 len=16 flags=0
May 14 08:05:57 nanopineo owserver[12048]: W1_NETLINK_MSG: type=4 (W1_MASTER_CMD) len=4 id=5
May 14 08:05:57 nanopineo OWFS[12048]:   DEBUG: ow_parseobject.c:(163) /12.DF07D5000000/pages/page.0
[... cut ...]
May 14 08:05:58 nanopineo OWFS[12253]:   DEBUG: ow_w1_addremove.c:(80) Request master be added: w1_bus_master2.
May 14 08:05:58 nanopineo OWFS[12253]:   DEBUG: ow_add_inflight.c:(25) Request master be added: w1_bus_master2
May 14 08:05:58 nanopineo OWFS[12253]:   DEBUG: ow_w1_addremove.c:(54) Setup structure for w1_bus_master1
May 14 08:05:58 nanopineo OWFS[12253]:   DEBUG: ow_w1_addremove.c:(80) Request master be added: w1_bus_master1.
May 14 08:05:58 nanopineo OWFS[12253]:   DEBUG: ow_add_inflight.c:(25) Request master be added: w1_bus_master1

rgillan, 2021-05-14:
The device is connected to bus 7, although we see

root@nanopineo:/opt/owfs/bin# ./owdir --dir /uncached/bus.7
/uncached/bus.7/interface/
/uncached/bus.7/28.AADD143D1401/
/uncached/bus.7/simultaneous/
/uncached/bus.7/alarm/
root@nanopineo:/opt/owfs/bin# ./owdir --dir /uncached/bus.5
/uncached/bus.5/interface/
root@nanopineo:/opt/owfs/bin# ./owdir --dir /uncached/bus.2
/uncached/bus.2/interface/
/uncached/bus.2/12.DF07D5000000/
/uncached/bus.2/alarm/

not sure if that helps or has any impact

ianka, 2021-05-14:
How do you control the +12V supply? I think you have to isolate the DS2482-800 1W pin as soon +12V is applied to the bus. As shown in the DS2406 datasheet. 

I'm not even sure if EPROM programming works if you have any non-EPROM devices on the bus.

rgillan, 2021-05-14:
Had a thought.  Could this be because the w1 2406 driver doesn't support the extended memory, and we're using owsserver via w1?  As this is for a programming rig, we don't need the w1_ds28e17 module that drove us to w1 anyway.  Am going to try and do the same with owserver directly talking to the ds2482-800 master device via i2c.

rgillan, 2021-05-14:
so there's an external circuit that we pulse with a gpio, which connects the +12v to the ds2406 and disconnects the master.  It's the only device on the bus at the time.  Will have the test with owserver->i2c->ds2406 set up next week and I'll circle back here when I have some more info.  Thanks again for your prompt response, it's really helpful

ianka, 2021-05-14:
The programming pulse has to be controlled by the onewire driver. I'm also not aware that any of the available host adapter drivers supports this.

rgillan, 2021-05-23:
We've got a DS2480B on the way and hopefully will make this possible.  Cheers

closed 2021-05-23 by rgillan