owfs/owfs #89: Add support for PHP8

utkarsh2102, 2021-05-27:
Hello,

Thanks for your work on owfs! 🎉 

However, whilst building against PHP8, the build failed with the following logs:
https://launchpadlibrarian.net/539581686/buildlog_ubuntu-impish-amd64.owfs_3.2p4+dfsg1-4build2_BUILDING.txt.gz

Can we please add support for PHP8? TIA!

williamdes, 2021-05-27:
Relates to: https://github.com/swig/swig/issues/1929
https://github.com/swig/swig/commit/56d74355735f3661406d69d04d89d1bdb4ca96f9#diff-f65c9bb1f02bb677912be45b343abf8beddc6150f53069b72e4699a8a45db7a4

Swig added PHP8 support: https://github.com/swig/swig/pull/1978

Seems similar: https://github.com/MapServer/MapServer/issues/6201

Maybe @ojwb would have some advice on this one, because it looks like the php 8 compatible code has not been released at Swig

miccoli, 2021-05-28:
Thank you for reporting and links to relevant swig documentations. See also #83.