owfs/owfs #90: IPv6 support not found by configure

bib1963, 2021-06-09:
In the dim and distant past, 2008-06-24 12:06:47 +0100, a patch was made to help support Solaris.

This contained a test for IPv6 support by using '#include "include/stdinc.h"]],'.

Strange for 2 reasons, 1, quoting rather than having the header contained within chevrons, suggests this was part of the owfs distribution, yet it is not there.

Secondly, Linux at least, does not contain a stdinc.h file and I'm not sure if Solaris did or does either.

Changing the 2 lines to include "<netinet.in.h>" allows configure to include IPv6 support.

miccoli, 2021-06-10:
Thanks for reporting: could you please add more detail? Are referring to this code block in `configure.ac`?

https://github.com/owfs/owfs/blob/b85b33a6978e158ae2654ad3354c950ba2575a6c/configure.ac#L1488-L1527

Please feel free to submit a PR, but at the moment support for OWFS is very low... at least from my side. I have a long list of things to sort out. Somehow this project has reached a complexity which makes it very hard to add new features.

bib1963, 2021-06-11:
Just read your 2 lines at the end. This is needed to get IPv6 working on owserver and the utilities, rather than just the owfs client - as opposed to the owfs release...

miccoli, 2021-06-12:
> Just read your 2 lines at the end. This is needed to get IPv6 working on owserver and the utilities, rather than just the owfs client - as opposed to the owfs release...

Yes, the problem indeed is that with the current monolithic build system, one has to verify that chanes to the single configure file do not cause problems in all the modules. Therefore also very simple changes require great work for testing all modules.

bib1963, 2021-06-12:
On Saturday, 12 June 2021 18:02:12 BST Stefano Miccoli wrote:
> Yes, the problem indeed is that with the current monolithic build system, one has to verify that chanes to the single configure file do not cause problems in all the modules. Therefore also very [...]

Very Zen like reply...