owfs/owfs #91: Windows Defender claims trojan detected in v3.2p4 release .zip

glenn-horton-smith, 2021-06-17:
Upon downloading owfs-3.2p4.zip from the github release page, Windows Defender immediately claims "Virus detected." Specifically, it is the file owfs-3.2p4/module/ownet/VisualBasic/csocketmaster/OWNet_VB.exe that is flagged inside the .zip file.
Strangely, none of the virus scanners at VirusTotal think this zip file dangerous, but MANY think that the .exe file it contains IS dangerous: 

.zip file: https://www.virustotal.com/gui/url/1cd3e16d60a2053dca164bcc7e47907c07256437014af0c485c7b323ef3a86a3/detection

.exe file: https://www.virustotal.com/gui/file/ee2db3c9af9b638b883635314df9ddd80cb174378b9daf5a9e532c21152c6067/detection

Similar issues have been reported with Windows Defender on other software --- e.g., https://github.com/bazelbuild/bazelisk/issues/168 --- but the fact that 30 out of 70 virus checkers on VirusTotal flag it is disturbing, including well known virus checkers such as Avast, AVG, BItDefender, Comodo, MalwareBytes, McAfee, and Symantec, among others.

Windows Defender describes it as follows:
```
Trojan:Win32/Ymacco.AAEE
Details:This program is dangerous and executes commands from an attacker.
https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=Trojan:Win32/Ymacco.AAEE&ThreatID=2147757298
```

I presume/hope these are false positives, similar to https://github.com/bazelbuild/bazelisk/issues/168.

It would be nice if this issue could be mitigated for owfs.

miccoli, 2021-06-17:
Thanks for reporting... the file you are mentioning is old stuff that was not touched for more than 15 years... and although git uses *only* a SHA-1 checksum it is very unlikely that malicious code has been injected in a way that fools git...

Nevertheless I think that this .exe can safely removed from the next owfs release.

miccoli, 2021-08-19:
Just to clarify: the problematic file [`OWNet_VB.exe`](https://www.virustotal.com/gui/file/ee2db3c9af9b638b883635314df9ddd80cb174378b9daf5a9e532c21152c6067/detection) is not in the source *distribution* `owfs-3.2p4.tar.gz` but in the source *tree* `v3.2p4.zip`, so technically it is not is not part of the owfs release, but just another instance of the deep stratification of code (partially unused/unmaintained) that forms this old and venerable project ;-).

For this reason I remove the v3.2p5 tag.