owfs/owfs #93: DS9490 bus master not found

silvaphoenix, 2021-08-18:
Hi,
I'm looking for some help setting up a DS9490 bus master in owserver.
I've installed owserver in Kubuntu 18.04 from the repository, version 3.1p5.
Running with the command owserver -u --debug gives the error message
<LIBUSB_ERROR_ACCESS> Could not open the USB bus master. Is there a problem with permissions?
DEFAULT: ow_usb_msg.c:(188) <LIBUSB_ERROR_ACCESS> Could not open the USB bus master. Is there a problem with permissions?
  DEBUG: ow_ds9490.c:(297) Cannot open USB device 5:1
CONNECT: ow_ds9490.c:(312) No USB DS9490 bus master found
DEFAULT: owlib.c:(208) Cannot open USB bus master
DEFAULT: owlib.c:(52) No valid 1-wire buses found
  DEBUG: ow_exit.c:(17) Exit code = 1
Running as root
 DEBUG: ow_usb_msg.c:(200) <LIBUSB_ERROR_NOT_FOUND> Could not release kernel module
CONNECT: ow_usb_msg.c:(207) <LIBUSB_ERROR_BUSY> Failed to set configuration on USB DS9490 bus master at 1:5
  DEBUG: ow_usb_msg.c:(233) Did not successfully open DS9490 1:5 -- permission problem?
  DEBUG: ow_ds9490.c:(297) Cannot open USB device 5:1
CONNECT: ow_ds9490.c:(312) No USB DS9490 bus master found
DEFAULT: owlib.c:(208) Cannot open USB bus master
CONNECT: ow_usb_msg.c:(246) <LIBUSB_ERROR_NOT_FOUND> Release interface (USB) failed
  DEBUG: ow_usb_msg.c:(250) <LIBUSB_ERROR_OTHER> Linux kernel driver reattach problem
Segmentation fault

The device is listed 
Bus 001 Device 005: ID 04fa:2490 Dallas Semiconductor DS1490F 2-in-1 Fob, 1-Wire adapter

Please can anyone help resolve this issue.
Thanks, Stephen

miccoli, 2021-08-19:
> I've installed owserver in Kubuntu 18.04 from the repository, version 3.1p5.
> Running with the command owserver -u --debug gives the error message
> <LIBUSB_ERROR_ACCESS> Could not open the USB bus master. Is there a problem with permissions?

So, if I've got it right, you are running `owserver` bundled with the Kubuntu distribution. It seems you are facing a permission problem.

Things to check:

- is there another `owserver` process running? Just in case stop/kill the process.
```
sudo systemctl stop owserver.service
sudo killall owserver
```
- try starting `owserver` as root.
``` 
sudo owserver -u --debug
``` 

If this works, next you should try to configure/enable/start owserver as a `systemd` service.

silvaphoenix, 2021-08-19:
Hi Stefano,
Thanks for responding.
I had already done all things you suggest but no luck.
I listed running as root in the post. 
I removed and re-installed the software in case there was something missing but nothing reported.
I'm now thinking the DS9490 might be faulty.
Any other ideas?
Stephen

miccoli, 2021-08-19:
oops, sorry, I didn't read that you have already tried running as root (I reformatted your post in markdown to make it more legible.)

I have very limited experience with USB adapters. Can you run `owserver -u --debug` with no USB adapter, just to verify that the `<LIBUSB_ERROR_XXX>`  disappear? My point is that if you still get `<LIBUSB_ERROR_XXX>` with no DS9490 inserted, then it is not the adapter to blame.

On raspbian if I run `owserver -u` with no adapters I obtain the following "clean" log

```
DEBUG MODE
libow version:
	3.2p3
  DEBUG: ow_daemon.c:(170) main thread id = 3069164624
  DEBUG: ow_inotify.c:(80) No configuration files to monitor
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
   CALL: ow_parsename.c:(174) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
  DEBUG: ow_regex.c:(24) Reg Ex expression <^$> compiled to 0xb6ec560c
  DEBUG: ow_regex.c:(24) Reg Ex expression <^all$> compiled to 0xb6ec562c
  DEBUG: ow_regex.c:(24) Reg Ex expression <^scan$> compiled to 0xb6ec564c
  DEBUG: ow_regex.c:(24) Reg Ex expression <^\*$> compiled to 0xb6ec566c
  DEBUG: ow_regex.c:(24) Reg Ex expression <^[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}$> compiled to 0xb6ec568c
  DEBUG: ow_regex.c:(24) Reg Ex expression <^-?[[:digit:]]+$> compiled to 0xb6ec56ac
]*$> compiled to 0xb6ec56ccg Ex expression <^ *([^ ]+)[ 
]*$> compiled to 0xb6ec56ecg Ex expression <^ *([^ ]+) *: *([^ ]+)[ 
]*$> compiled to 0xb6ec570cg Ex expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[ 
CONNECT: ow_ds9490.c:(312) No USB DS9490 bus master found
DEFAULT: owlib.c:(208) Cannot open USB bus master
DEFAULT: owlib.c:(52) No valid 1-wire buses found
  DEBUG: ow_exit.c:(17) Exit code = 1
   CALL: ow_lib_close.c:(21) Starting Library cleanup
   CALL: ow_lib_stop.c:(22) Clear Cache
  DEBUG: ow_cache.c:(295) Flipping cache tree (purging timed-out data)
  DEBUG: ow_cache.c:(313) flip cache. tdestroy() will be called.
  DEBUG: ow_cache.c:(295) Flipping cache tree (purging timed-out data)
  DEBUG: ow_cache.c:(313) flip cache. tdestroy() will be called.
   CALL: ow_lib_stop.c:(24) Closing input devices
   CALL: ow_lib_stop.c:(26) Closing output devices
   CALL: ow_lib_close.c:(42) Finished Library cleanup
  DEBUG: ow_lib_close.c:(50) Libraries closed
```

silvaphoenix, 2021-08-19:
I did the run with no adapter and get the clean log with no<LIBUSB_ERROR_XXX>
The log is a little different as I am running version 3.1p5 but in principle the same. Perhaps this is the problem.
Before getting the DS9490 I was running a DS9097U serial adapter through a USB adapter. It worked but not reliably.
The installation of 3.2p4 failed in Ubuntu 18.04. I think there were some undefined dependences missing.