owfs/owfs #96: Problems with DS9490 bus master

silvaphoenix, 2021-09-03:
I'm still getting a problem with the DS9490 bus master.
The udev rule Jan advised seemed to solve the problem for a few days. However the DS9490 stopped reading the sensors and now will not work again. It is a libusb error problem.
 DEBUG: ow_usb_msg.c:(200) <LIBUSB_ERROR_NOT_FOUND> Could not release kernel module
It does find the DS9490
DEFAULT: ow_usb_msg.c:(214) Opened USB DS9490 bus master at 1:6.
However debug output finishes with
DEFAULT: ow_usb_cycle.c:(119) Set DS9490 1:6 unique id 0x00 (no devices at all)
  DEBUG: ow_net_server.c:(76) ServerAddr: [0.0.0.0] [4304]
At this point owserver is still running until killed.

Any clues as to where to go next would be much appreciated
Stephen

miccoli, 2021-09-03:
`LIBUSB_ERROR_NOT_FOUND` is expected when there is no kernel module claiming the DS9490 bus master.

Which is the output of `owdir` command?

What hapenns when you unplug/plug the DS9490 bus master or the 1-wire devices?

silvaphoenix, 2021-09-04:
Thanks for looking at this Sefano,
owdir gives
/bus.0
/uncached
/settings
/system
/statistics
/structure
With the DS9490 plugged in but the 1-wire devices unplugged the error is there but the bus master is eventually found and owdir gives
/81.EA3E30000000
/bus.0
/uncached
/settings
/system
/statistics
/structure
I guess the problem is in the 1-wire bus. I have 18 DS18S20 temperature sensors.
The supply voltage to the wire is 5.02 the return data voltage is only 3.71. Presumably one or more of the sensors has failed.
Would you agree?
Stephen

miccoli, 2021-09-04:
> I guess the problem is in the 1-wire bus. I have 18 DS18S20 temperature sensors.
> The supply voltage to the wire is 5.02 the return data voltage is only 3.71. Presumably one or more of the sensors has failed.
> Would you agree?

Sounds reasonable, since when a DS9490 is up and running the onboard slave with family code 81 should always be visible. Unfortunately I'm not very good at the HW part of 1-wire... @ianka should be able to give you better advice.

silvaphoenix, 2021-09-07:
To keep you in the picture Stefano I plugged a spare DS18S20 into the DS9490 and all worked giving the 28 and 81 family codes proving the fault lay in the wire.
Hindsight says I should have checked this earlier. The first sensor I considered was an outside one. I found corroded connections in the supposedly waterproof connectors. All worked after disconnecting it so problem solved.
I appreciated you taking the time to consider the problem.
Thanks 
Stephen

miccoli, 2021-09-07:
Glad you solved your problem.

closed 2021-09-07 by miccoli