owfs/owfs #103: 1-wire devices for sale

honechko, 2022-08-17:
Hello. I am from Ukraine. I develop 1-wire slave devices on AVR microcontrollers and have 3 ready products. Can you let me advertise them here?

honechko, 2022-08-18:
# 1-wire pulse counter

[image]

Price: $14 (with soldered screw connector)

Device is compatible with OWFS, but its logic of operation differs from DS2423.
Device periodically pools inputs A,B and increments two 32bit counters
according to state changes. To distinguish high and low states of input signal,
device compares ADC result with two configurable thresholds. There is
possibility to switch on internal pull-up resistor and/or power supply for
sensor for very short time before polling input. For example it can flash IR
LED, measure reflected light with phototransistor, compare with thresholds and
detect "presence". Device consumes uA of current and can operate autonomously
on CR2032 battery for few years. Device can be used in home automation for
electricity, water, heat, gas accounting, it requires only sensor and no extra
components, such as comparators or pull-up resistors. Configuration is
performed by writing to device's memory, there is also web-interface.

# 1-wire port expander

[image]

Price: $13 (with soldered screw connector)

Device replicates all possibilities of DS2450 and in this part compatible with
OWFS, but has some extensions that can be supported in OWFS by additional code.
Extensions:
* fifth GPIO with PWM capability;
* states of GPIO extended to: input, weak pull-up, output low, output high;
* all 5 GPIO are capable to read input logic signal;
* device can continuously watch for certain logic state on all GPIO and set alarm flags;
* 1-wire slave to SPI-master bridge.
Device can be widely used in home automation for sensors read, relay control,
alarm systems, communication with microcontrollers over 1-wire. There is
web-interface for control over all functions of device except SPI.

# 1-wire bus master

[image]

Price: $16 (with soldered screw connector, without DCDC),
+$1 for B0505S (not soldered due to risk of shipping damage)

Device is based on DS2480B and have full support in OWFS. 1-wire part
halvanically isolated from interface (UART) side. Device can work with
3.3V and 5V TTL logic and can be connected to PC via any USB-UART adapter
or can be directly connected to Raspberry Pi pins or microcontroller pins.
1-wire part can be powered from external 5V power supply or from host power
via B0505S or B0305S.

You can order this devices by sending me email to: ow (at) honey.com.ua
Shipping from Ukraine is aproximately $7 for small packet.

miccoli, 2022-08-19:
For a glitch, the comment I wrote was not published... but only the help wanted label was added. Please forgive me if I add a late reply now.

I do not think that GitHub issues are an effective/appropriate place for advertising. Maybe you should try on the [owfs-developers mailing list](https://sourceforge.net/projects/owfs/lists/owfs-developers).