owfs/owfs #107: Enabling Aux Power on LinkUSB

hhinma, 2023-05-16:


miccoli, 2023-05-18:
FS_w_LINK_aux is available at `/bus.0/interface/settings/link/auxctrl`, see
https://github.com/owfs/owfs/blob/62cb355d99913948241013d4867fcdff31124001/module/owlib/src/c/ow_interface.c#L188

I suppose you can interact with this settings from the shell with
```
$ owrite /bus.0/interface/settings/link/auxctrl 1
$ owrite /bus.0/interface/settings/link/auxctrl 0
```

(disclaimer: I have no idea which is the pourpose of this settings: I've just inspected the source code).

stromnet, 2023-05-18:
It was me who implemented that ages ago! Iirc, I used it to control a P-FET that supplied power to the 1-Wire bus, this aided in some bus hangs that I had issues with back then, de-powering the bus via aux-ctrl allowed me to recover the bus remotely.

I believe writing 1/0 will just toggle the output on or off. So as long as your integration can write arbitrary owfs paths I guess it should work? No idea about the HA integration though.

hhinma, 2023-05-19:
Thank you Johan!   after posting the request to the list I had managed to track down your code via the various posted discussions about what to name the new value in the filesystem.   
The only configurable items in the HA integration at present are which precision values to use for temperature sensors there, so it seems like a new little sub-project for me.  Next on my list is to understand the owserver protocol.