owfs/owfs #108: owfs installation proplem

pin32, 2023-09-30:
hi, I'm trying to install owfs for reading 18b20 sensors via a ds9490r, the system is on rpi4.
the problem I encounter is the debug output:

pin32@raspberrypi:~ $ sudo owserver -u --debug
DEBUG MODE
libow version:
3.2p4
   DEBUG: ow_daemon.c:(170) main thread id = 548272477312
   DEBUG: ow_inotify.c:(80) No configuration files to monitor
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
    CALL: ow_parsename.c:(174) path=[]
   DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
   DEBUG: ow_regex.c:(24) Reg Ex expression <^$> compiled to 0x7fa81451d0
   DEBUG: ow_regex.c:(24) Reg Ex expression <^all$> compiled to 0x7fa8145210
   DEBUG: ow_regex.c:(24) Reg Ex expression <^scan$> compiled to 0x7fa8145250
   DEBUG: ow_regex.c:(24) Reg Ex expression <^\*$> compiled to 0x7fa8145290
   DEBUG: ow_regex.c:(24) Reg Ex expression <^[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{ 1,3}\.[[:digit:]]{1,3}$> compiled to 0x7fa81452d0
   DEBUG: ow_regex.c:(24) Reg Ex expression <^-?[[:digit:]]+$> compiled to 0x7fa8145310
]*$> compiled to 0x7fa8145350Ex expression <^ *([^ ]+)[
]*$> compiled to 0x7fa8145390Ex expression <^ *([^ ]+) *: *([^ ]+)[
]*$> compiled to 0x7fa81453d0Ex expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[
   DEBUG: ow_usb_cycle.c:(169) Compare (add,bus) (3,1) with (-1,-1) handle (nil)

   DEBUG: ow_usb_msg.c:(200) <LIBUSB_ERROR_NOT_FOUND> Could not release kernel module
CONNECT: ow_usb_msg.c:(207) <LIBUSB_ERROR_BUSY> Failed to set configuration on USB DS9490 bus master at 1:3
   DEBUG: ow_usb_msg.c:(233) Did not successfully open DS9490 1:3 -- permission problem?
   DEBUG: ow_ds9490.c:(297) Cannot open USB device 3:1
CONNECT: ow_ds9490.c:(312) No USB DS9490 bus master found
DEFAULT: owlib.c:(208) Cannot open USB bus master
owserver: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
Aborted

but despite the no access indication I can still see the sensors and get the readings correctly and constantly.
I attach the answers to the commands and the configurations I have in use:
pin32@raspberrypi:~ $ sudo systemctl status owserver.service
● owserver.service - Backend server for 1-wire control
      Loaded: loaded (/lib/systemd/system/owserver.service; enabled; vendor preset: enabled)
      Active: active (running) since Sat 2023-09-30 12:00:21 BST; 2min 3s ago
TriggeredBy: ● owserver.socket
        Docs: man:owserver(1)
    Main PID: 474 (owserver)
       Tasks: 3 (limit: 3933)
         CPU: 128ms
      CGroup: /system.slice/owserver.service
              └─474 /usr/bin/owserver --foreground -c /etc/owfs.conf

Sep 30 12:00:20 raspberrypi systemd[1]: Starting Backend server for 1-wire control...
Sep 30 12:00:20 raspberrypi owserver[474]: DEFAULT: ow_opt.c:(780) systemd mode -- ignore foreground request
Sep 30 12:00:21 raspberrypi owserver[474]: DEFAULT: ow_opt.c:(670) systemd mode -- ignore 'p' option
Sep 30 12:00:21 raspberrypi owserver[474]: DEFAULT: ow_usb_msg.c:(214) Opened USB DS9490 bus master at 1:3.
Sep 30 12:00:21 raspberrypi owserver[474]: DEFAULT: ow_usb_cycle.c:(128) Set DS9490 1:3 unique id to 81 C3 5A 30 00 00 00 0C
Sep 30 12:00:21 raspberrypi systemd[1]: Started Backend server for 1-wire control.

pin32@raspberrypi:~ $ ls /mnt/1wire
28.371176E0013C 28.447695F0013C 28.B66995F0013C 81.C35A30000000 alarm bus.1 simultaneous structure uncached
28.447695F0013C 28.534276E0013C 28.DCEF81E3E13C 81.C35A30000000 bus.0 settings statistics system

owfs.conf:
# Sample configuration file for the OWFS suite for Debian GNU/Linux.
#
#
# This is the main OWFS configuration file. You should read the
# owfs.conf(5) manual page in order to understand the options listed
# here.

######################## SOURCES ##########################
#
# With this setup, any client (but owserver) uses owserver on the
# localmachine...
! server: server = localhost:4304
#
# ...and watcher uses the real hardware, by default fake devices
# This part must be changed on real installation
#server: FAKE = DS18S20,DS2405
#
# USB device: DS9490
server: usb = all
#
# Serial port: DS9097
#server: device = /dev/ttyS1
#
# owserver tcp address
#server: server = 192.168.10.1:3131
#
# random simulated device
#server: FAKE = DS18S20,DS2405
#
########################## OWFS ########################## ##
#
mountpoint = /mnt/1wire
allow_other
#
####################### OWHTTPD #########################

http: port = 2121

######################## OWFTPD ##########################

ftp: port = 2120

######################## OWSERVER ##########################

server: port = localhost:4304

fuse.conf:
# /etc/fuse.conf - Configuration file for Filesystem in Userspace (FUSE)

# Set the maximum number of FUSE mounts allowed to non-root users.
# The default is 1000.
mount_max = 1000

# Allow non-root users to specify the allow_other or allow_root mount options.
user_allow_other

pin32, 2023-09-30:
rule udev:
#SUBSYSTEM!="usb_device", ACTION!="add", GOTO="owfs_rules_end"

# DS2490 1-Wire adapter (funziona ma da problemi di permessi su debug) 
#SYSFS{idVendor}=="04fa", SYSFS{idProduct}=="2490", MODE="660", GROUP="owfs"

#LABEL="owfs_rules_end"
# ultriore prova ....
SUBSYSTEMS=="usb", ATTRS{idVendor}=="04fa", ATTRS{idProduct}=="2490", NAME="bus/usb/$env{BUSNUM}/$env{DEVNUM}", GROUP="users", MODE="0660"

pin32, 2023-09-30:
screen shot

[image]

pin32, 2023-09-30:
[image]

closed 2023-09-30 by pin32

pin32, 2023-09-30:
in practice I get errors from debugging permissions and non-access to the busmaster ds9490r, checking the status of the service everything seems in order, in the "1wire" folder I have double the sensors compared to reality (12 sensors and two busmasters are displayed but they are duplicates ).
I use the system with "openhab" and I can also get the sensor data reliably.
I can't understand why debugging gives me problems and the sensors are read twice.
thank you for the help you want to give me.

ianka, 2023-09-30:
To me it seems the kernel has the ds2490 module loaded. Check this with `lsmod`.

The sensors are listed twice because there is a race between the kernel W1 subsystem and owserver controlling the DS9490 via USB directly. For the same reason the access is not reliable.

If that ds2490 module is loaded, the kernel controls the hardware, too. You can't access it with owfs' -u driver option then but have to use the --w1 option that makes owfs use the kernel's W1 subsystem instead. In the configuration file, the relevant option is `server: usb = all` which you have to replace with `server: w1`

As an alternative, you have to blacklist the ds2490 kernel module or the w1 kernel module.

pin32, 2023-10-01:
Thanks for your help.
I checked with lsmod and it seems there are no modules with the name "ds2490". If I remember correctly I put this form in the blacklist, I'll go and check it. I'll edit the "owfs.conf" file and report the results, thanks

ianka, 2023-10-01:
If you have blacklisted the `ds2490` module, the kernel will ignore the USB adapter and `--w1` and `server: w1` will not work as intended. Those are alternatives. You have to do one or the other, not both.

If you have openhab running with direct access to USB, and start an owserver with direct access to USB you run into the same problem. It's two programs directly controlling the hardware and they race with each other then. Don't do that. If you want to use both openhab and owserver at the same time, you either have to make openhab use owserver instead of controlling the USB hardware directly, or make both openhab and owserver use the kernel W1 subsystem.

pin32, 2023-10-01:
with the modification to the .conf file I obtain this data:

pin32@raspberrypi:~ $ sudo owserver -u --debug
DEBUG MODE
libow version:
	3.2p4
  DEBUG: ow_daemon.c:(170) main thread id = 548629398656
  DEBUG: ow_inotify.c:(80) No configuration files to monitor
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
   CALL: ow_parsename.c:(174) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
  DEBUG: ow_regex.c:(24) Reg Ex expression <^$> compiled to 0x7fbd5a81d0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^all$> compiled to 0x7fbd5a8210
  DEBUG: ow_regex.c:(24) Reg Ex expression <^scan$> compiled to 0x7fbd5a8250
  DEBUG: ow_regex.c:(24) Reg Ex expression <^\*$> compiled to 0x7fbd5a8290
  DEBUG: ow_regex.c:(24) Reg Ex expression <^[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}$> compiled to 0x7fbd5a82d0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^-?[[:digit:]]+$> compiled to 0x7fbd5a8310
]*$> compiled to 0x7fbd5a8350Ex expression <^ *([^ ]+)[ 
]*$> compiled to 0x7fbd5a8390Ex expression <^ *([^ ]+) *: *([^ ]+)[ 
]*$> compiled to 0x7fbd5a83d0Ex expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[ 
  DEBUG: ow_usb_cycle.c:(169) Compare (add,bus) (6,1) with (-1,-1) handle (nil)

  DEBUG: ow_usb_msg.c:(200) <LIBUSB_ERROR_NOT_FOUND> Could not release kernel module
DEFAULT: ow_usb_msg.c:(214) Opened USB DS9490 bus master at 1:6.
   CALL: ow_parsename.c:(174) path=[]
   CALL: ow_parsename.c:(174) path=[/uncached/bus.0]
  DEBUG: ow_regex.c:(24) Reg Ex expression <^bus\.([[:digit:]]+)/?> compiled to 0x7fbd5a8430
  DEBUG: ow_regex.c:(24) Reg Ex expression <^settings/?> compiled to 0x7fbd5a8470
  DEBUG: ow_regex.c:(24) Reg Ex expression <^statistics/?> compiled to 0x7fbd5a84b0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^structure/?> compiled to 0x7fbd5a84f0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^system/?> compiled to 0x7fbd5a8530
  DEBUG: ow_regex.c:(24) Reg Ex expression <^interface/?> compiled to 0x7fbd5a8570
  DEBUG: ow_regex.c:(24) Reg Ex expression <^text/?> compiled to 0x7fbd5a85b0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^json/?> compiled to 0x7fbd5a85f0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^uncached/?> compiled to 0x7fbd5a8630
  DEBUG: ow_regex.c:(24) Reg Ex expression <^unaliased/?> compiled to 0x7fbd5a8670
  DEBUG: ow_regex.c:(24) Reg Ex expression <^alarm?> compiled to 0x7fbd5a86b0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^simultaneous/?> compiled to 0x7fbd5a86f0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^thermostat/?> compiled to 0x7fbd5a8730
  DEBUG: ow_regex.c:(24) Reg Ex expression <^/bus\.[[:digit:]]+/?> compiled to 0x7fbd5a8770
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.> compiled to 0x7fbd5a87b0
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.all$> compiled to 0x7fbd5a87f0
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.byte$> compiled to 0x7fbd5a8830
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.[[:digit:]]+$> compiled to 0x7fbd5a8870
  DEBUG: ow_regex.c:(24) Reg Ex expression <\.[[:alpha:]]$> compiled to 0x7fbd5a88b0
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(100) 0: 0->5 found <><bus.0><>
  DEBUG: ow_regex.c:(100) 1: 4->5 found <bus.><0><>
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_dir.c:(69) path=/uncached/bus.0
   CALL: ow_dir.c:(104) path=/uncached/bus.0
   CALL: ow_parsename.c:(174) path=[/uncached/bus.0/interface]
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(100) 0: 0->5 found <><bus.0><>
  DEBUG: ow_regex.c:(100) 1: 4->5 found <bus.><0><>
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(24) Reg Ex expression <^([[:xdigit:]]{2})\.?([[:xdigit:]]{12})\.?([[:xdigit:]]{2}){0,1}$> compiled to 0x7fbd5a88f8
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_usb_cycle.c:(62) Callback on /uncached/bus.0/interface
  DEBUG: ow_parsename.c:(133) /uncached/bus.0/interface
  DEBUG: ow_search.c:(31) Start of directory path=/uncached/bus.0 device=00 00 00 00 00 00 00 00
  DEBUG: ow_ds9490.c:(674) Index 0
  DEBUG: ow_select.c:(69) Selecting a path (and device) path=/uncached/bus.0 SN=00 00 00 00 00 00 00 00 last path=00 00 00 00 00 00 00 00
  DEBUG: ow_select.c:(81) Clearing root branch
  DEBUG: ow_ds9490.c:(562) DS9490 RESET. changed 15, flex: 1
   DATA: ow_ds9490.c:(1008) set flexible speed
[... 223 lines cut ...]
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_usb_cycle.c:(62) Callback on /uncached/bus.0/simultaneous
  DEBUG: ow_parsename.c:(133) /uncached/bus.0/simultaneous
   CALL: ow_parsename.c:(174) path=[/uncached/bus.0/alarm]
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(100) 0: 0->5 found <><bus.0><>
  DEBUG: ow_regex.c:(100) 1: 4->5 found <bus.><0><>
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_regex.c:(53) Not found
  DEBUG: ow_usb_cycle.c:(62) Callback on /uncached/bus.0/alarm
  DEBUG: ow_parsename.c:(133) /uncached/bus.0/alarm
  DEBUG: ow_dir.c:(199) ret=0
  DEBUG: ow_usb_cycle.c:(92) Finished FS_dir
  DEBUG: ow_parsename.c:(133) /uncached/bus.0
DEFAULT: ow_usb_cycle.c:(128) Set DS9490 1:6 unique id to 81 C3 5A 30 00 00 00 0C
  DEBUG: ow_net_server.c:(76) ServerAddr: [0.0.0.0] [4304]
CONNECT: ow_net_server.c:(112) [Address already in use] Bind problem []
CONNECT: ow_net_server.c:(123) No good listen network sockets []
CONNECT: ow_net_server.c:(158) [Address already in use] Default port not successful. Try an ephemeral port
  DEBUG: ow_net_server.c:(76) ServerAddr: [0.0.0.0] [0]

pin32, 2023-10-01:
repeating the debug command after some time:

pin32@raspberrypi:~ $ sudo owserver -u --debug
DEBUG MODE
libow version:
	3.2p4
  DEBUG: ow_daemon.c:(170) main thread id = 547611985024
  DEBUG: ow_inotify.c:(80) No configuration files to monitor
CONNECT: ow_dnssd.c:(81) Zeroconf/Bonjour is disabled since dnssd library isn't found
   CALL: ow_parsename.c:(174) path=[]
  DEBUG: owlib.c:(77) Global temp limit 0C to 100C (for fake and mock adapters)
  DEBUG: ow_regex.c:(24) Reg Ex expression <^$> compiled to 0x7f80b601d0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^all$> compiled to 0x7f80b60210
  DEBUG: ow_regex.c:(24) Reg Ex expression <^scan$> compiled to 0x7f80b60250
  DEBUG: ow_regex.c:(24) Reg Ex expression <^\*$> compiled to 0x7f80b60290
  DEBUG: ow_regex.c:(24) Reg Ex expression <^[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}\.[[:digit:]]{1,3}$> compiled to 0x7f80b602d0
  DEBUG: ow_regex.c:(24) Reg Ex expression <^-?[[:digit:]]+$> compiled to 0x7f80b60310
]*$> compiled to 0x7f80b60350Ex expression <^ *([^ ]+)[ 
]*$> compiled to 0x7f80b60390Ex expression <^ *([^ ]+) *: *([^ ]+)[ 
]*$> compiled to 0x7f80b603d0Ex expression <^ *([^ ]+) *: *([^ ]+) *: *([^ ]+)[ 
  DEBUG: ow_usb_cycle.c:(169) Compare (add,bus) (6,1) with (-1,-1) handle (nil)

  DEBUG: ow_usb_msg.c:(200) <LIBUSB_ERROR_NOT_FOUND> Could not release kernel module
CONNECT: ow_usb_msg.c:(207) <LIBUSB_ERROR_BUSY> Failed to set configuration on USB DS9490 bus master at 1:6
  DEBUG: ow_usb_msg.c:(233) Did not successfully open DS9490 1:6 -- permission problem?
  DEBUG: ow_ds9490.c:(297) Cannot open USB device 6:1
CONNECT: ow_ds9490.c:(312) No USB DS9490 bus master found
DEFAULT: owlib.c:(208) Cannot open USB bus master
owserver: ../../libusb/os/threads_posix.h:46: usbi_mutex_lock: Assertion `pthread_mutex_lock(mutex) == 0' failed.
Aborted

pin32, 2023-10-01:
the sensors are no longer detected:

pin32@raspberrypi:/mnt/1wire $ ls
bus.0  bus.1  settings  statistics  structure  system  uncached

pin32, 2023-10-01:
in blacklist:

blacklist ds2490
#blacklist ds9490r
blacklist wire

ianka, 2023-10-01:
Please check whether openhab is configured to control the USB device directly or whether it runs another owserver that is configured to use the USB device.

ianka, 2023-10-01:
If you want to run multiple programs accessing the same onewire hardware, what I recommend to do is not to blacklist the kernel modules but use them, and configure all the programs so they use the kernel driver for their access rather than manipulating the device directly.

pin32, 2023-10-01:
I don't think that openhab creates its own server for accessing the hardware resource, in fact it is specified that an already functioning server installed on the host is assumed.

pin32, 2023-10-01:
As regards the issues indicated in the post, I'm not very clear how to proceed. if I understand correctly I could also blacklist the ds9490r module (which is now commented) in order to exclude access by the kernel and probably solve the problem.
unfortunately I am not very expert and things that you take for granted are new to me, I would be grateful if you add more details on how to practically act on the system, thank you very much!

ianka, 2023-10-02:
If openhab assumes there is an owserver running, it likely also assumes that one uses the /etc/owfs.conf configuration file that you posted.

In that configuration file, you specified `server: usb = all`, which means that this automatically started owserver accesses the USB host adapter directly. That's okay, you can do it that way, but you must blacklist the `ds2490` kernel module then. (There is no ds9490r kernel module. That DS9490R hardware has the ds2490 chip inside. I don't know why this bogus line is in your module blacklist.)

If you do it that way, you can't start another owserver with the -u option. Or with that particular configuration file. Because then you had two owservers both accessing the USB host adapter directly. You have a race on the USB host adapter then. You have to start the second owserver so that it uses the first. Use the -s option for your debugging purposes. Most likely `-s localhost` or `-s localhost:4304` does the trick.

But I'm unsure what you want to achieve. If you wanted to debug, you had to check the log file of the first owserver which is being run by the system instead of starting a second one.

pin32, 2023-10-04:
hi, I don't know exactly how openhab acquires information about the sensors, but I think it simply fishes them out of the /mnt/1wire folder without doing anything else, after all it's just a binding. I did a test: having openhab on the container I tried not to start it and I got the same result both in terms of status and debug, and repeated readings, so it shouldn't be openhab that is attempting to access the USB device.
as you well pointed out, there are two systems that compete in communicating with the hardware device, but at least one of them does not have permissions (from debugging), but in any case I see two sensor readings.
Unfortunately I can't understand how to investigate the solution to the problem, I don't see anything abnormal in the active processes. I would like to point out that my intentions are not to start more servers for reading the sensors. If you can, tell me what I can do to investigate the problem, taking into account that I don't have a lot of experience with the terminal and Linux. Thank you

ianka, 2023-10-04:
What you have to understand is that all those daemons `owfs`, `owserver`, `owhttpd`, `owftpd` use the very same `owlib` library and that way they all have all device options built in. So if you run more than one of those, you have to make sure that they don't access the same hardware.

Check which of those daemons is running.

External software may also link that `owlib`. It gains the same functionality. The other option is to run the `owserver` program, for the hardware access and then access it through a TCP socket. Most of the language bindings do it that way and `openhab` does the same, as I saw [here.](https://www.openhab.org/addons/bindings/onewire/)

What I suspect is that there is more than one of the daemons `owfs`, `owserver`, `owhttpd`, `owftpd`  running and they all access the USB hardware. If you have a populated `/mnt/1wire` directory, that means there is the literal `owfs` daemon running. Disable that one. You don't need it for openhab, as openhab accesses the onewire through the `owserver` daemon.