owfs/owfs #109: owserver with w1 GPIO driver / different temperature

ThomasCr, 2023-11-15:
Hi,

I discovered some strange behavior with owserver with w1 backend. First of all, the sensor works when I query the kernel driver:

```
thomas@rpitemp:~ $ cat /sys/devices/w1_bus_master1/28-0000034a5a5d/temperature
19375
thomas@rpitemp:~ $ cat /sys/devices/w1_bus_master1/28-0000034a5a5d/w1_slave
37 01 4b 46 7f ff 09 10 26 : crc=26 YES
37 01 4b 46 7f ff 09 10 26 t=19437
```

But when I query with owserver/owfs/owhttp I get different value:

```
thomas@rpitemp:~ $ echo `cat /mnt/1wire/28.5D5A4A030000/temperature`
85
thomas@rpitemp:~ $ echo `cat /mnt/1wire/settings/units/temperature_scale`
C
```

My config is quite simple:
```

######################## SOURCES ########################
#
# With this setup, any client (but owserver) uses owserver on the
# local machine...
! server: server = localhost:4304
#
# ...and owserver uses the real hardware, by default fake devices
# This part must be changed on real installation
#server: FAKE = DS18S20,DS2405
#
# USB device: DS9490
#server: usb = all
#
# Serial port: DS9097
#server: device = /dev/ttyS1
#
# owserver tcp address
#server: server = 192.168.10.1:3131
#
# random simulated device
#server: FAKE = DS18S20,DS2405
#
server: w1
######################### OWFS ##########################
#
mountpoint = /mnt/1wire
allow_other
#
####################### OWHTTPD #########################

http: port = 2121

####################### OWFTPD ##########################

ftp: port = 2120

####################### OWSERVER ########################

#server: port = localhost:4304
server: port = 4304

# Set Unit global
Celsius

```

Some system info:
```
thomas@rpitemp:~ $ lsmod | grep w1
w1_therm               28672  0
w1_gpio                16384  0
wire                   49152  2 w1_gpio,w1_therm
thomas@rpitemp:~ $ modinfo w1_gpio
filename:       /lib/modules/6.1.61-v8+/kernel/drivers/w1/masters/w1-gpio.ko.xz
license:        GPL
author:         Ville Syrjala <syrjala@sci.fi>
description:    GPIO w1 bus master driver
srcversion:     89D33780A3861BE22B688C6
alias:          of:N*T*Cw1-gpioC*
alias:          of:N*T*Cw1-gpio
depends:        wire
intree:         Y
name:           w1_gpio
vermagic:       6.1.61-v8+ SMP preempt mod_unload modversions aarch64
thomas@rpitemp:~ $ uname -r
6.1.61-v8+
thomas@rpitemp:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
thomas@rpitemp:~ $ dpkg -l | grep owserver
ii  owserver                             3.2p4+dfsg1-4.1+b1               arm64        Backend server for 1-Wire control
```

Any idea whats wrong?

ThomasCr, 2023-11-15:
It looks like I am not alone.

https://community.home-assistant.io/t/1-wire-temps-85-from-owserver-home-assistant-but-correct-reading-from-sys-bus-w1-domoticz-same-pi/238325/2

ThomasCr, 2023-11-16:
Okay, I want to give back my newest finding:
After searching for a solution I found [this post](https://sourceforge.net/p/owfs/mailman/owfs-developers/thread/e90fdabf-7882-11f6-a7fa-b1d74829db0b@gmx.de/): 

> > Could I ask for further clarification on this please?  I have used
> > dtoverlay=w1-gpio,gpiopin=4
> > and installed the kernel module using
> > sudo modprobe w1-gpio
> > and the DS18B20 works fine in three wire mode with a 4.7k pullup on DQ
> > 
> 1.5k please. PLEEEEAAAAAASE! (and to 3.3V, not 5V)
> 
> Why? Because when you use parasitic power on the DS18B20, a 4.7k pullup
> to 3.3V will only deliver 3.3V/4.7k = 0.7mA to the chip. That's not
> enough during conversion. You will get power-on-resets (aka 85°C readings).
> 

so I also replaced my restistor from 4.7k to a 1.5k and yes... I works.

But what I find not clear, why the kernel driver could also with the higher resistor the right temperature?

ianka, 2023-11-17:
85°C is the power-on reset value of the DS18B20.

So, you read 85°C with OWFS because the device did a power-on reset before. It did that power-on reset before because the kernel driver queried it, depleted its tiny internal capacitor that acts as its power supply, and you haven't given it enough time to recharge through the pullup before your query.

The kernel driver filters out those 85°C results as bogus readings. It shows you the previous result then. OWFS does the same but gives up earlier with retries. As 85°C is also a valid temperature.

ThomasCr, 2023-11-23:
if have still problems with owfs and different readings between owfs and the kernel driver:

```
thomas@rpitemp:~ $ cat /sys/devices/w1_bus_master1/28-0000034a76ba/temperature
25437
thomas@rpitemp:~ $ cat /sys/devices/w1_bus_master1/28-0000034a76ba/w1_slave
97 01 4b 46 7f ff 09 10 1c : crc=1c YES
97 01 4b 46 7f ff 09 10 1c t=25437
thomas@rpitemp:~ $ echo `owread -C /28.BA764A030000/temperature`
127.938
```

is there anything else I can do?

miccoli, 2023-11-23:
Sorry for the dumb question, but is the 25.437 °C reading correct?

127.938 °C is the full scale reading of the [DS18B20](https://www.analog.com/media/en/technical-documentation/data-sheets/DS18B20.pdf) sensor at 12 bit resolution:
```
>>> round((2**11 - 1) * 2**-4, 3)
127.938
```
Did you try to read the sensor at lower resolutions, i.e `/28.BA764A030000/temperature9`, `/28.BA764A030000/temperature10`, `/28.BA764A030000/temperature11`?

ThomasCr, 2023-11-24:
Hi,

ah, interesting - didnt know that the temperature9..12 readings are different timings - thought it are history readings 🤦‍♂️
But no, the readings are the same.

up | directory
-- | --
address | 28BA764A03000038
alias | Wohnzimmer Oben
crc8 | 38
errata | errata
family | 28
fasttemp | 127.5
id | BA764A030000
latesttemp | 127.5
locator | FFFFFFFFFFFFFFFF
power | NO  (0)
r_address | 380000034A76BA28
r_id | 0000034A76BA
r_locator | FFFFFFFFFFFFFFFF
scratchpad | F8074B461FFF01106F
temperature | 127.938
temperature10 | 127.75
temperature11 | 127.875
temperature12 | 127.938
temperature9 | 127.5
temphigh | 75
templow | 70
tempres | 12
type | DS18B20

The High Temperature reading can be true (didnt check it exaclty) ..
The current reading is:
```
thomas@rpitemp:~ $ cat /sys/devices/w1_bus_master1/28-0000034a76ba/w1_slave
8f 01 4b 46 7f ff 01 10 14 : crc=14 YES
8f 01 4b 46 7f ff 01 10 14 t=24937
```
And a Alexa Echo which is close to that sensor (but close to the floor) is 21,6 °C - and the 1-wire sensor is close to the heater and at the ceiling. The upper level (where we talking about) has new isolation... so it can match...