owfs/owfs #110: Error installing owhttpd and owserver  on Debian 12.4 (Bookworm)

msinn, 2023-12-15:
When installing owhttpd and owserver, apt gives me two errors **Could not execute systemctl** (one for httpd and one for owserver):

```
Vormals nicht ausgewähltes Paket owhttpd wird gewählt.
(Lese Datenbank ... 44851 Dateien und Verzeichnisse sind derzeit installiert.)
Vorbereitung zum Entpacken von .../owhttpd_3.2p4+dfsg1-4.1+b1_amd64.deb ...
Entpacken von owhttpd (3.2p4+dfsg1-4.1+b1) ...
Vormals nicht ausgewähltes Paket owserver wird gewählt.
Vorbereitung zum Entpacken von .../owserver_3.2p4+dfsg1-4.1+b1_amd64.deb ...
Entpacken von owserver (3.2p4+dfsg1-4.1+b1) ...
owserver (3.2p4+dfsg1-4.1+b1) wird eingerichtet ...
Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.
owhttpd (3.2p4+dfsg1-4.1+b1) wird eingerichtet ...
Could not execute systemctl:  at /usr/bin/deb-systemd-invoke line 145.
Trigger für man-db (2.11.2-2) werden verarbeitet ...
```

ianka, 2023-12-15:
We do not provide those debian packages nor the install scripts. You have to contact the package maintainer **Vincent Danjean** on this.

See e.g. [the package description.](https://packages.debian.org/buster/owfs)

vdanjean, 2025-08-26:
I do not look at this tracker very often. Please, open a Debian bug report if you observe problems with the Debian packaging.
Not being able to call systemctl is strange unless you choose to avoid systemd on your system.
If this is the case, I would welcome patches to fix the issue (or, at least, a more precise trace as plain init files are still provided)
In any case, this issue can be closed here (and opened in the Debian BTS if needed)