owfs/owfs #111: Search algorithm with target setup

koen-php, 2024-02-02:
I would like to use the uncached owfs to quickly search for devices of a specific family code, ignoring all other possibly connected devices.

The search algorithm describes how to achieve this in the chapter TARGET SETUP.

> The 'TARGET SETUP' operation is a way to preset the search state to first find a particular family type. Each 1-Wire device has a one byte family code embedded within the ROM number (see Figure 1). This family code allows the 1-Wire master to know what operations this device is capable of. If there are multiple devices on the 1-Wire it is common practice to target a search to only the family of devices that are of interest. To target a particular family, set the desired family code byte into the first byte of the ROM_NO register and fill the rest of the ROM_NO register with zeros. Then set the LastDiscrepancy to 64 (40 hex) and both LastDeviceFlag and LastFamilyDiscrepancy to 0. When the search algorithm is next performed the first device of the desired family type is discovered and placed in the ROM_NO register. Note that if no devices of the desired family are currently on the 1-Wire, then another type will be found, so the family code in the resulting ROM_NO must be verified after the search. 

I was wondering if anyone had already tried this with OWFS?
By default the uncached browsing discovers all possible devices, but this results in too slow results for my case. 

Thanks in advance for any feedback!

ianka, 2024-02-04:
Not an answer, but which family is it specifically? The iButton keys?

koen-php, 2024-02-05:
Yes, family code 0x01 for unique identifiers.

ianka, 2024-02-06:
If you only have a single iButton attached to the bus, or a sub-bus of the DS2409, so you can identify the lock it is put it, you can use the `simultaneous/single_ds2400` node below the bus directory. That one issues a special 0x0f READ ROM command that only the iButtons respond to, none of the other chips.

It doesn't work with more than one iButton per bus of course, and it doesn't work with the DS9490 host adapters as they have a built-in iButton.

koen-php, 2024-02-06:
Actually I want to (quickly) scan which devices of family code 0x01 are connected on a a bus where there are also 0x28 temperature sensors connected. The 'uncached' folder uses the SEARCH ROM algorithm, but it becomes too slow when there are multiple devices connected. If the TARGET SETUP is used, the algorithm ignores all other family codes and it works just as fast as if there were only IDs of 0x01 connected...

ianka, 2024-02-08:
I understand that, but it's not implemented in OWFS so far. I'm not sure whether the DS9490 host adapters support it, and to support it with the popular w1 kernel driver, it had to be implemented in that one.