owfs/owfs #116: owserver.service: Start request repeated too quickly

Piotr-23, 2024-08-01:
Hi, how can I get rid of this:

piotr@piotr:~$ sudo service owserver status
[sudo] password for piotr:
× owserver.service - Backend server for 1-wire control
     Loaded: loaded (/usr/lib/systemd/system/owserver.service; enabled; preset: enabled)
     Active: failed (Result: exit-code) since Thu 2024-08-01 22:02:23 CEST; 47s ago
TriggeredBy: × owserver.socket
       Docs: man:owserver(1)
   Main PID: 1096 (code=exited, status=1/FAILURE)
        CPU: 13ms

Aug 01 22:02:23 piotr systemd[1]: owserver.service: Scheduled restart job, restart counter is at 5.
Aug 01 22:02:23 piotr systemd[1]: owserver.service: Start request repeated too quickly.
Aug 01 22:02:23 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 01 22:02:23 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.

?

I've been using owfs as a weather station since many years... and on re-installing on Ubuntu 24, I apparently mis-configured.... please help | Piotr

miccoli, 2024-08-02:
This seems a downstream issue, so it is hard to help you from our side.

The system messages simply state that `owserver` is not starting properly, but without any clue to the reason.

Does `sudo journalctl -u owserver.service` gives you more detailed info on the causes of failure?

Piotr-23, 2024-08-02:
Thank you!

The message looks differently now after re-installing owserver and tweaking config files:

sudo systemctl status owserver
× owserver.service - Backend server for 1-wire control
     Loaded: loaded (/usr/lib/systemd/system/owserver.service; enabled; preset: enabled)
    Drop-In: /etc/systemd/system/owserver.service.d
             └─override.conf
     Active: failed (Result: exit-code) since Fri 2024-08-02 11:06:41 CEST; 49min ago
       Docs: man:owserver(1)
    Process: 2532 ExecStart=/usr/bin/owserver -c /etc/owfs.conf --foreground (code=exited, status=1/FAILURE)
   Main PID: 2532 (code=exited, status=1/FAILURE)
        CPU: 10ms

Aug 02 11:06:41 piotr systemd[1]: owserver.service: Scheduled restart job, restart counter is at 5.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Start request repeated too quickly.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 02 11:06:41 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.

sudo journalctl -u owserver.service gives

-- Boot 7b55d526d63447c282f712532af3c235 --
Aug 02 11:06:41 piotr systemd[1]: Starting owserver.service - Backend server for 1-wire control...
Aug 02 11:06:41 piotr owserver[2511]: DEFAULT: owlib.c:(208) Cannot open USB bus master
Aug 02 11:06:41 piotr owserver[2511]: DEFAULT: owlib.c:(52) No valid 1-wire buses found
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 02 11:06:41 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Scheduled restart job, restart counter is at 1.
Aug 02 11:06:41 piotr systemd[1]: Starting owserver.service - Backend server for 1-wire control...
Aug 02 11:06:41 piotr owserver[2515]: DEFAULT: owlib.c:(208) Cannot open USB bus master
Aug 02 11:06:41 piotr owserver[2515]: DEFAULT: owlib.c:(52) No valid 1-wire buses found
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 02 11:06:41 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Scheduled restart job, restart counter is at 2.
Aug 02 11:06:41 piotr systemd[1]: Starting owserver.service - Backend server for 1-wire control...
Aug 02 11:06:41 piotr owserver[2521]: DEFAULT: owlib.c:(208) Cannot open USB bus master
Aug 02 11:06:41 piotr owserver[2521]: DEFAULT: owlib.c:(52) No valid 1-wire buses found
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 02 11:06:41 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Scheduled restart job, restart counter is at 3.
Aug 02 11:06:41 piotr systemd[1]: Starting owserver.service - Backend server for 1-wire control...
Aug 02 11:06:41 piotr owserver[2526]: DEFAULT: owlib.c:(208) Cannot open USB bus master
Aug 02 11:06:41 piotr owserver[2526]: DEFAULT: owlib.c:(52) No valid 1-wire buses found
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 02 11:06:41 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Scheduled restart job, restart counter is at 4.
Aug 02 11:06:41 piotr systemd[1]: Starting owserver.service - Backend server for 1-wire control...
Aug 02 11:06:41 piotr owserver[2532]: DEFAULT: owlib.c:(208) Cannot open USB bus master
Aug 02 11:06:41 piotr owserver[2532]: DEFAULT: owlib.c:(52) No valid 1-wire buses found
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Main process exited, code=exited, status=1/FAILURE
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 02 11:06:41 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Scheduled restart job, restart counter is at 5.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Start request repeated too quickly.
Aug 02 11:06:41 piotr systemd[1]: owserver.service: Failed with result 'exit-code'.
Aug 02 11:06:41 piotr systemd[1]: Failed to start owserver.service - Backend server for 1-wire control.
lines 4367-4463/4463 (END)

miccoli, 2024-08-02:
So `DEFAULT: owlib.c:(208) Cannot open USB bus master` means that `owserver` is not able to access the 1-wire bus adapter on USB (permission problems?).

Piotr-23, 2024-08-02:
Thanks! How to check that and fix? | P

miccoli, 2024-08-21:
Closing because downstream issue (USB permissions, owserver uid)

closed 2024-08-21 by miccoli