owfs/owfs #117: wrong values not detected

boellner, 2024-08-18:
Hello,
It happens that tempeature vales of every of my DS18B20 read out a wrong value that is not detected by crc. 
To better reproduce the error I made a test setup where I connected a lot of sensors to a long line.
Half of the displayed values are in the measurement range but show random false values that are not filtered out. What can be the issue and how to detect bad values?

[image]

ianka, 2024-08-19:
Are those sensors by chance all parasite powered?

boellner, 2024-08-19:
Some sensors are parasitic powered, some not. The above screenshot is a worst case setup to show you the problem. In my "normal" setup I get this kind of errors occasionally.
I can live with network dropouts etc.
but bad receiving values in a valid range that are obviously false like in the picture above without being detected by the bus at all is a no-go for me. I have to rely on the values Thanks

ianka, 2024-08-22:
The DS18B20 are calibrated but they aren't safe against brownouts. You have to avoid brownouts. Power the sensors.

miccoli, 2024-08-22:
I agree with @ianka : if an inaccurate value gets on the DS18B20 scratchpad memory, there is no safeguard inside the one-wire protocol or OWFS to detect the anomaly.

You can work on the HW side (stress test sensors to detect faulty ones or check the "Powering the DS18B20" section in the [data sheet](https://www.analog.com/media/en/technical-documentation/data-sheets/ds18b20.pdf)). Sometimes HW problems can be mitigated by reading temperature values at lower resolution (9bit instead of 12bit), because temperature conversion times are much shorter.

On the SW side you can implemented anomaly detection algorithms.

boellner, 2024-08-23:
ok, thank you all for the explanation

closed 2024-08-23 by boellner