owfs/owfs #119: input/output error

promy, 2025-03-13:
I'm using 'cat /mnt/1wire/28.C3E5AE030000/temperature' to get te value from a temp sensor for a perl script.
This is working perfectly untill i upgraded my homeassistant version to 2025.3
Homeassistant is polling the 1wire via owserver.
It seems that when the 'cat' command is trying to read the value when owserver is reading it, it fails. Logs show a "input/output error".

Is this to be expected? and more importantly, can it be fixed (or should i just program a retry in the script?

ianka, 2025-03-13:
This could be a problem with FUSE. In general, we recommend not to use the FUSE binding any more because it has those unsolvable concurrency issues but instead use a language binding (for perl in your case) or, as a last resort, call the owread/owdir/owget/owwrite client tools that belong to owserver instead. So instead of

    $ cat /mnt/1wire/28.C3E5AE030000/temperature

do

    $ owread /28.C3E5AE030000/temperature

There are manpages for those programs.

promy, 2025-04-06:
not really a fix, but i reduced the accuracy of the other script (homeassistant) to "reduce" the problem. Sometimes it seems to remain, even with  owread...