owfs/owfs #123: How 1-wire puts a heavy load on the processor?

szczukot, 2025-09-14:
I'm using 1-wire on my RPi 5 on HomeAssitant.
It uses around 7-8% of the CPU constantly.
Is this the fault of the base 1-wire itself? Very big complicated loop?
Can I add something to configuration, "something," to disable it so it doesn't load the CPU so much?
It's not a 3D game that needs to load the CPU like that.

arnout, 2025-09-17:
There may be some process that triggers rescans of the one-wire bus. The CPU load of that rescan is taken by the owfs process, but it is not actually responsible for triggering it. With 20 devices on the bus, a scan can take quite a bit of load.

szczukot, 2025-09-17:
Does it really have to be that CPU-hungry?
8% of the RPi 5's total usage?
3D games would require less CPU than reading a few bytes over RS232 :)
As you can see, the entire HA, hundreds of different sensors, automation, etc., consumes several times less CPU than this single OWFS.

miccoli, 2025-09-18:
The basic information is missing: which is the process that is responsible of the green line in the graph above? Is it `owserver` which is the currently supported way of accessing the owfs system, or is it the legacy `owfs`, which is strongly discouraged?

I would for sure label the current usage anomalous, but there is no way of telling the cause from the supplied info.

szczukot, 2025-09-18:
I see in logs something like that:

s6-rc: info: service owserver-init: starting
[10:57:01] INFO: Generated owfs.config file:
[10:57:01] INFO: # owserver configuration 
## device(s)
### serial
server: device = /dev/ttyUSB0
### serial
server: device = /dev/ttyUSB1
!server: server = 4304
server: port = 4304
http: port = 8099
## temperature scale
Celsius
s6-rc: info: service owserver-init successfully started
s6-rc: info: service owserver: starting
s6-rc: info: service owserver successfully started
s6-rc: info: service discovery: starting
s6-rc: info: service owhttpd: starting
s6-rc: info: service owhttpd successfully started
[10:57:01] INFO: Starting owserver...

szczukot, 2025-09-18:
I don't really understand what's going on there.
Why does the CPU show high usage, but the processes are practically zero all the time?
Maybe the container isn't configured correctly (does it have some options enabled or something).
And why are so many processes launching while reading data – one process per sensor (isn't it running in a loop somehow?).

https://github.com/user-attachments/assets/612f74d8-8695-4c94-839a-822904e8ddc5

miccoli, 2025-09-18:
@szczukot please, have a look at <https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax> to properly format your posts... in particular use use triple backticks for fencing blocks you copy verbatim (see my edit).

Back to your issue, there is a clear problem in the setup of your system: there should be only a single `owserver` process running at any instant. The purpose of `owserver` is exactly to serialise multiple concurrent clients trying to access the 1-wire network. The explosion of `owserver` processes that we see at second 6 of your recording is insane.

I suspect that there is a problem in the setup of your HomeAssistant system, but probably this is not the correct place for asking questions on HomeAssistant configuration.